nkolban/esp32-snippets #373: SampleHIDKeyboard.cpp not sending any keys?

JuanPotato, 2018-01-22:
Hi, I've setup esp-idf and created a project for SampleHIDKeyboard.cpp, but once I make it and connect to my phone, I see the program runs with make monitor, but nothing is ever sent to my phone. Can anyone else test this and tell me if it is working for them and so what settings they had to enable.

Logs: https://www.hastebin.com/raw/larufeboko
I only added some logging in the task function to see that it ran

chegewara, 2018-01-22:
What do you do to receive keyboard strokes on your phone? For example i run notes to see message, thats why there is 5 seconds delay that gives me a time to open notes.

JuanPotato, 2018-01-22:
I open notes just like you mentioned, I've also tried this with my computer
to no avail.

chegewara, 2018-01-22:
Ok, im checking right now. I'll give you answer shortly.

chegewara, 2018-01-22:
Just compilled latest library with latest esp-idf and on my phone it seems to works fine. I have text in my notes. I did copy SampleHIDKeyboard.cpp and https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleKeyboardTypes.h to main folder. Try to run make menuconfig and switch log level to debug or verbose. Also try to switch options like described here
https://github.com/nkolban/esp32-snippets/issues/109#issuecomment-336687009

chegewara, 2018-01-22:
You dont even have to copy 
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleKeyboardTypes.h
because its included in library as HIDKeyboardTypes.h

JuanPotato, 2018-01-22:
Ok just to get this out of the way. How do I set up the project? I just
took the hello world example and in the main folder symlinked to cpp_utils.
Also of course removed the hello world code and replaced it with the sample
hid and then the main file. When I get the chance I'll setup everything how
described and set the log level to verbose. I appreciate your help

chegewara, 2018-01-22:
From description you are doing all its needed to make it works.

JuanPotato, 2018-01-22:
Changing the compiler settings worked, but why?

chegewara, 2018-01-22:
Because its know esp-idf issue that is causing characteristics are not seen by client apps.

JuanPotato, 2018-01-22:
weird, maybe we should put a notice somewhere in the files that are affected so that people know? And has an esp-idf issue already been made?

closed 2018-01-22 by JuanPotato

JuanPotato, 2018-01-23:
Any idea why this doesnt work on iOS?

chegewara, 2018-01-23:
Sorry but no, i have no ption to test it on iOS or MacOS. 
If you have some logs maybe i can see what is happening. Maybe this can be related to this
https://github.com/nkolban/esp32-snippets/issues/325#issuecomment-355872432

JuanPotato, 2018-01-23:
Do you just want me to set the logs to verbose and then send the result when i connect an iPad?

chegewara, 2018-01-23:
Yes.

JuanPotato, 2018-01-23:
https://pastebin.com/raw/qyT2NL9R

cleaned up the escape codes, but in case you wanted them: https://pastebin.com/raw/nQ4Xmjhq

chegewara, 2018-01-23:
From this log everything seems to be fine. iPhone is requesting MTU with value 185, esp32 accepts it and then is sending reportMap 65 bytes long in one packet. Then iPhone is writing to CCC descriptor to get notifications. Then bunch of packages are sent (each is key down or key up package):
```
D (65400) BLECharacteristic: >> setValue: length=8, data=0000280000000000, characteristic UUID=00002a4d-0000-1000-8000-00805f9b34fb
D (65410) BLECharacteristic: << setValue
D (65410) BLECharacteristic: >> notify: length: 8
D (65410) GeneralUtils:      00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f  ----------------
D (65420) GeneralUtils: 0000 00 00 28 00 00 00 00 00                          ..(.....
D (65430) FreeRTOS: Semaphore taking: name: ConfEvt (0x3ffdef78), owner: <N/A> for notify
D (65440) FreeRTOS: Semaphore taken:  name: ConfEvt (0x3ffdef78), owner: notify
```

There is no information than notification is `<< notifications disabled; ignoring` that means iPhone should have receiving notifications.
Maybe try to add this command:
`BLEDevice::setMTU(23);`
somewhere in here:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleHIDKeyboard.cpp#L85

Its long shot, but who knows.

JuanPotato, 2018-01-23:
Sadly nothing https://pastebin.com/raw/3zVg8BhK

chegewara, 2018-01-23:
Again, log is looking good, everything seems to work fine. There is nothing more i can do.

coletz, 2019-04-19:
I think that maybe I could know what is causing this issue. MacOS and iOS are not recognizing the esp32 as a keyboard completely. Let me try to explain my issue:
I can connect my esp32 to my mac, and it print characters. But it is not recognized as a keyboard in settings (while other official bt keyboard are recognized). Now, checking connected devices (something like device manager on Windows) I can see that the original keyboard has minor and major values which references to the CoD (as you can see [here](https://www.bluetooth.com/specifications/assigned-numbers/baseband)).
Some stuff will work only when macOS will recognize the esp32 as an actual keyboard, and **_in my opinion_** this will probably make iOS communicate correctly