nkolban/esp32-snippets #383: header esp_bt.h

gerisse, 2018-01-27:
Hi Mr Kolban
First, thanks a lot for your open ressources about esp32 and others. It 's great to share them.
a Question about  the header BLEDevice.h (used for me for the first time in a platformio project, with  BLE_client example) : when i build it , it does not find the header esp_bt.h ;
It said : /home/rp/.platformio/lib/ESP32 BLE Arduino_ID1841/src/BLEDevice.h:16:20: fatal error: esp_bt.h: No such file or directory
i make a mistake ? 
Thanks a lot

chegewara, 2018-01-27:
Hi,
there is change in esp-idf. bt.h has been changed to esp_bt.h, but it may be not updated on arduino yet. You can simply "fix" this error by creating file `esp_bt.h` and add just one line to it:
`#include "bt.h"`

Put this file in BLE library or your project folder.

forhuan, 2018-01-30:
Hi Mr Kolban
After i add this file to BLE library folder
it said 
```
\Documents\Arduino\hardware\espressif\esp32\libraries\BLE\src/BLEExceptions.h:13:2: error: #error "C++ exception handling must be enabled within make menuconfig. See Compiler Options > Enable C++ Exceptions."

#error "C++ exception handling must be enabled within make menuconfig. See Compiler Options > Enable C++ Exceptions."

```
Thanks a lot

forhuan, 2018-01-30:
i found a file which contains "CONFIG_CXX_EXCEPTIONS"
 \Documents\Arduino\hardware\espressif\esp32\tools\sdk\sdkconfig (1 hit)
	Line 95: CONFIG_CXX_EXCEPTIONS=

i try to modify it to CONFIG_CXX_EXCEPTIONS=1 or CONFIG_CXX_EXCEPTIONS=y
 but it shows an error the same as last comment

chegewara, 2018-01-30:
https://github.com/nkolban/esp32-snippets/issues/228#issuecomment-350660354

https://github.com/nkolban/esp32-snippets/issues/318

nkolban, 2018-02-05:
We are developing quite a chain here :-)

Lets see if we can decode.   First there is the ESP-IDF ... this is the SDK for building C/C++ apps for the native ESP32.  This project was designed / developed primarily for that.  Then we have the BLE C++ classes which are part of this cpp_utils project.  Again ... all normal.   Then we have the Arduino ESP32 project.  This is a separate project which attempts to make Arduino apps available for an ESP32 environment.  It was found that the BLE C++ classes worked great on Arduino ESP32.  Next the BLE classes (during their normal development work) were enhanced to take advantage of C++ exception handling.  That caused a problem as Arduino ESP32 wasn't being built / distributed to provide exception handling.   However, the author of the Arduino ESP32 functions switched that capability on in a recent distribution of Arduino ESP32 and we were back to working fine.  This takes us to "platformio".  That is an area I don't fully appreciate.  Do we know "how it works?".   My guess is that "platformio" builds on top of Arduino ESP32 which builds on top of ESP-IDF.  As such, we have a "chain" of dependencies.

Do we happen to know "which" Arduino ESP32 is used by platformio?

yutakau, 2018-09-09:
Hello,
Is this issue still unfixed? I am facing the same situation.

chegewara, 2018-09-09:
@yutakau it has been fixed few months ago. Have you updated esp32-arduino lately?

yutakau, 2018-09-09:
@chegewara  Thank you for your reply, but I got same error after esp32-arduino updating and tried again.
Do you have any suggestion or advice?

chegewara, 2018-09-09:
Delete whole espressif folder and install esp32 with boards manager.
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md
Or reinstall arduin-ide if its not a problem.

yutakau, 2018-09-10:
@chegewara Thank you, It worked well!  Re-installing arduino-esp32 resolved the problem.

closed 2018-11-29 by chegewara