nkolban/esp32-snippets #384: ESP32_BLE_Arduino/examples/BLE_client hags on reconnect

GrooverFromHolland, 2018-01-28:
Hi,
First of all my thanks for your effort on ESp32 code.
I have made a project combining IRremote.h, esp_now.h and your ESP32_BLE_ArduinoBLE_client example.
It all works excelent, except for reconnecting if connection is lost.
I use a timer to check if Ble connection still exists and  if not reconnect with the folowing code:

```
if (xSemaphoreTake(timerSemaphore, 0) == pdTRUE){	

		if (!pClient->isConnected())
		{
			Serial.println("Timeout; rescan for BLE ");
			doConnect = true;
			reconnect();
		}		
	}

void reconnect(){
	BLEScan* pBLEScan = BLEDevice::getScan();
	pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
	pBLEScan->setActiveScan(true);
	pBLEScan->start(40);
}

```
After this my ESP32 hangs with no activity at all.
I get the folowing message inmy serialmonitor:

> Timeout; rescan for BLE 
> E (48870920) BT: btc_search_callback  BLE observe complete. Num Resp 4
> 
>   Created client
> Forming a connection to 50:8c:b1:6a:e6:3e
> E (48871131) BT: btc_gattc_call_handler()
> E (48871141) BT: bta_gattc_process_api_open Failed, unknown client_if: 0

Is it my code that causes this?
If not:
What do the E-messages mean?
Is there a way to catch these errors?

I hope you can help me with this isue .

GrooverFromHolland, 2018-01-29:
After some deep digging in my code I found some ambiguous code.
 I use visual studio community 2017 with visualmicro plug in for my projects.
In if `(!pClient->isConnected())` if I highlight `isConnected() `and **Go To Declaration** it brings me to BLEClient.h line 42 :
`bool  isConnected();`
If I **Peek Definition** it brings me to WiFiSTA.ccp line 304:
```
bool WiFiSTAClass::isConnected()
{
    return (status() == WL_CONNECTED);
}
```

Could this cause my problem, or is it visual studio that's playing tricks on me?

nkolban, 2018-02-05:
In C++ we can have multiple methods with the same name but if they belong to different class types, they will not collide.  For example, if we have a class called `BLEClient` which has a method definition called `isConnected()` and a class called `WiFiSTAClass` which also has a method definition called `isConnected()`, we will find that they are distinct methods.  The context in which they are invoked will determine which method is run.

d-asnaghi, 2018-09-08:
Hi, I think this question has been addressed in [#498](https://github.com/nkolban/esp32-snippets/issues/498#issuecomment-384614251), for reference I will state Tuet's answer here as well:

> 1. When a onDisconnect() callback comes in, you have to manually invoke the disconnect() function of BLEClient.
> 2. After that, the instance of the BLEClient can be deleted. The destructor within this function seems to clean up all the hierarchy of many BLERemoteService and BLERemoteCharacteristics by itself. Don't try to delete these instances manually, otherwise a heap corruption will appear. It seems that most of the resources get released this way.
> 3. You can now connect and recreate the hierarchy again - without issues like "..already registered".