nkolban/esp32-snippets #387: How can you force a disconnect of a BLE client from the server?

SalvaPedraza, 2018-02-01:
Hi,

I am developing an application where the ESP32 act as server.
Some times if the client disconnect without any information to server (for example, application crash), the server continue in "connected to client" status and it is not advertising and do not accept any other connection.
I would like to do a whatchdog task that it disconnect the client if no activity (read or write) during certain time.
How can I force disconnection from client, so BLE server start advertising and accept new connections?

chegewara, 2018-02-02:
There is no function implemented in this library that can let you disconnect client. You have 2 options:
- use esp_ble_gatts_close(), here you have documentation http://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/esp_gatts.html#_CPPv219esp_ble_gatts_close13esp_gatt_if_t8uint16_t
- ask mr @nkolban to implement it

nkolban, 2018-02-05:
This is an interesting question.   I'm imagining a BLE Server (a peripheral) which is advertising happily.  It is then connected to by a BLE Client (a central) and hence it stops advertising.  Now the client crashes (powers down or goes out of communication range) and the BLE Server *should* start advertising again.  The interesting question is how should the server detect this?  In TCP communications, we have the notion of a keep-alive packet.  This is a packet periodically sent between peers to validate that the connection is still logically present.   In BLE ... is there an equivalent concept?  Is the responsibility down to us (BLE application writers) such that when we have to maintain our own timers in BLE servers and when we haven't received a request in a period, then we disconnect?

ankgt, 2018-02-27:
I have the exact same problem. I haven't tried it yet, but since I am in control of the client as well, I am thinking of having my app periodically write to a certain characteristic, which the ESP32 will check and reset regularly. If there is nothing written when the ESP32 checks, it will disconnect the client.

@SalvaPedraza Please let us know if you found any other solution.

SalvaPedraza, 2018-03-05:
I have been busy lately but finally I tried it.
Based on suggestion by chegewara, I have added a function to BLEServer library (Code below).
As my client is periodically (few seconds) communication with server once connected, I have a task in server that check the time-stamp of last communication received from client (updated in BLECharacteristicCallbacks.onWrite callback function). If older than certain threshold (10 seconds in my case) I call this function.

It works perfectly for me. It disconnect, start advertising and accept connection if requested by client. No need to do any additional action.

void BLEServer::disconnectClient() {
	ESP_LOGD(LOG_TAG, ">> disconnectClient. GATTS IF: %d. CONN ID: %d", m_gatts_if,  m_connId);
	if (m_gatts_if >= 0 and m_connId >= 0) {
		esp_err_t result = ::esp_ble_gatts_close(m_gatts_if, m_connId);
		if (result != ESP_OK) {
			ESP_LOGE(LOG_TAG, "Disconnection FAILED. Code: %d.", result);
		}
	} else {
		ESP_LOGD(LOG_TAG, "Can not disconnect Client.");
	}

	ESP_LOGD(LOG_TAG, "<< disconnectClient");
} // disconnectClient

posics, 2018-04-03:
This function seems to be critical to be available in the BLEServerCallbacks::onDisconnect() otherwise no client can ever re-connect. I had a situation where a client would get disconnected due to a timeout and the server would know it but after onDisconnect no client would ever be able to re-connect. Having now copied SalvaPedraza's function into BLEServer.cpp everything works once I call pServer.disconnectClient() from whithin onDisconnect() event handler. I tried to just do esp_ble_gatts_close but gatts_if and connId are private so no go.

posics, 2018-04-03:
I spoke to soon...

PaulEverblix, 2018-08-09:
Hello everyone !

I came across this conversation with the same idea of being able to remove the connection to a client.
@SalvaPedraza 's solution looks like it's the right one. Will this code be added in the library any time soon ?
Thanks !

chegewara, 2019-01-30:
Hi @nsamala 
i can add disconnect to BLEServer, but it will require you to track/store connected peer devices or at least their connID, because this is how esp-idf close function looks like:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/bluetooth/esp_gatts.html#_CPPv219esp_ble_gatts_close13esp_gatt_if_t8uint16_t

and this library implementation is build to connect different peer devices to one BLEServer which holds its own gattsIf but every client got different connId.

william-ferguson-au, 2019-02-20:
On a tangential note:
1) When the server stops advertising, it shouldn't automagically start advertising again.
2) When a client disconnects from the Server it shouldn't automagically start advertising again.

These should both be handled from callbacks within the app (IF that is required). Otherwise the app can't control when it is advertising a BT Server.

I have created an issue for this : https://github.com/nkolban/esp32-snippets/issues/816

mgulsoy, 2019-06-26:
I have a similar situation (could be counted as a use case):

"MyDevice" (server - peripheral) works on batteries and after say, 30 minutes of inactivity, it goes into deep sleep mode without wake up timer set. The purpose of that is to require the client (central - ex. Android phone) reset "MyDevice" using on/off switch.

My aim is to iterate on all client connections and close the connection one by one.

Mrgove10, 2020-04-24:
Hello @SalvaPedraza  Do you have an example how to implement your function ? I'm not that good in C++ / Adruino and i'm confused as to how to implement it .

jackjansen, 2021-07-29:
A word of warning: I stumbled upon this bug report, and it seemed to exactly match the symptoms I was seeing, with both BLE client and BLE server written by me and running on esp32, but it turned out to be different. Not a bug in the client (not disconnecting) but a bug in the server (not restarting advertising).

If I hard-reset the client the connection is closed correctly, however: advertising isn't restarted on the server.

So a simple
```
void onDisconnect(BLEServer* pServer) {
  pServer->startAdvertising();
}
```
did the trick.

Axliyor, 2022-06-18:
> I have been busy lately but finally I tried it. Based on suggestion by chegewara, I have added a function to BLEServer library (Code below). As my client is periodically (few seconds) communication [...]
/--------------------------------------------------------------------------------------------------------

In this case, sometimes there is an error in the library, and the esp32 reloads and stops advertising!!!

SalvaPedraza

jackjansen, 2022-06-20:
@Axliyor have you applied my `onDisconnect()` callback, form the comment above?

Axliyor, 2022-06-29:
> @Axliyor have you applied my `onDisconnect()` callback, form the comment above?

No i didn't use it but i tried it worked. The reason I didn't use it is because it disconnects and restarts esp32.
  BLEDevice::deinit();
  Serial.println("APP");
  BLEDevice::init(name);
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());
  BLEService *pService = pServer->createService(SERVICE_UUID);
  
  Men BLEDevice::deinit(); thus I stopped ble and managed to disconnect the clients without rebooting the device. I restarted the service and server

jackjansen, 2022-06-29:
Tangentially related: I'm switching from the default BLE library to Nimble (Arduino-nimble in my case, because I'm using the Arduino framework), and it seems to be much more stable in all respects. There was a little work in porting, but not much, and now memory use is much less and things appear more stable too.

sdetweil, 2023-05-10:
in my case, I need to change the service being advertised..
i connect from my app
send data from characteristic
    in esp32
   loop delay 1 second (wait for milis() to be after now+1000, 'not delay()')
   stop advertising
   remove service, remove service uuid.. data functionality (this UUID toggles for other purposes, and this works ok).
    ----- maybe a bug right here, if the service is removed, then all connections should be closed
   add service, add service uuid  , firmware upload functionality
   start advertising

app disconnects sometime in there
BUT .. myservercallbacks doesn't know .. I HAVE to disconnect, as the service i WAS connected to is gone..
but no callback

now, ble scanner on android sees new service and I can connect and interact with the new service characteristics
but  bluetoothctl on my linux box cannot see that device anymore.. (still connected?)
oops..  if I reboot the device, then the device appears in bluetoothctl scan.(with non-upload UUID as expected) 

I can't yet use Nimble, as removeservice and remove service UUID do not actually remove (results in TWO services being advertised..... but in my case the two are conflicting.. issued open at nimble

satishgaurav, 2024-03-26:
> I have been busy lately but finally I tried it. Based on suggestion by chegewara, I have added a function to BLEServer library (Code below). As my client is periodically (few seconds) communication [...]

It resets the ESP32 with the following error message. Is it possible to disconnect the client with restarting the device? 
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400da38b  PS      : 0x00060a30  A0      : 0x800d3892  A1      : 0x3ffd1980  
A2      : 0x00000000  A3      : 0x0000001d  A4      : 0x0000000a  A5      : 0x0000e8c0  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x800dcaf4  A9      : 0x3ffd1960  
A10     : 0x00000002  A11     : 0x3f4005eb  A12     : 0x00008fc3  A13     : 0x0000ef40  
A14     : 0x3ffe7eb8  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000030  LBEG    : 0x40090941  LEND    : 0x40090951  LCOUNT  : 0xfffffff8  

Backtrace: 0x400da388:0x3ffd1980 0x400d388f:0x3ffd19a0 0x400d4c4f:0x3ffd19c0 0x400de245:0x3ffd19f0

ELF file SHA256: bd78b932f617b35c

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
Waiting a client connection to notify...