nkolban/esp32-snippets #388: How to use https:// with the Webserver class

squonk11, 2018-02-01:
I would like to switch my Webserver, which uses the WebServer class, to https:// protocol. How can this be done? Is there a sample or an explanation? Do I also need to make a WPS connection on my WiFi in order to make it work? Any hint would be welcome.

nkolban, 2018-02-05:
@squonk11 Its been quite a while since I looked at the WebServer.h and WebServer.cpp source.  These classes wrapper the underlying Mongoose library.  I looked inside the WebServer.cpp source and could find no reference nor mention to SSL / TLS (which is the technology behind process "https" protocol).  What that makes me think is that we never did implement SSL / TLS in the wrapper classes.  Unfortunately, I don't have any immediate plans to enhance these classes.  While Mongoose networking is a super-cool library, my focus of later has been on the HttpServer class.  This does seem to offer SSL / TLS.   Please don't get me wrong, I am **not** trying to say that HttpServer is better than Mongoose ... I can't imagine it could even be close ... I'm merely saying that enhancements to WebServer class won't be happening any time soon based on the backlog of other work items that are in the queue.  Obviously if more folks come a calling asking for enhancements to WebServer, then we'll look at it again.  But right now we are event driven and interest in WebServer is very low as compared to many other areas.

squonk11, 2018-02-05:
Thank you for your answer. Unfortunately I made a mistake in my question: I wanted to ask for the procedure for setting up a https:// server using HttpServer class (not WebServer class) - sorry for that mistake.
Do you have a short description how to do that?