nkolban/esp32-snippets #389: How do I dealloc WiFiEventHandler after BootWiFi is completed?

zachjacobs, 2018-02-02:
I have a wrapper class working properly to obtain WiFi credentials and store in NVS.  Once that is finished, I use C to extract credentials from NVS and try to connect to a C based WiFi project.  When I do this, the WiFiEventHandler::getEventHandler() still seems to be called during WiFi events.   I have tried to resolve the issue by deleting the m_pWifiEventHandler pointer in the WiFi class but it still gets called.

I would like to use my own event handler in C that I have already written and previously used.
`esp_event_loop_init(&app_event_handler, NULL);`

Is there a preferred method for removing the reference to the old wifi class instance?

chegewara, 2018-02-02:
@zachjacobs Im not sure, but try to use "empty" event handler WiFi class and your own event handler from C.

nkolban, 2018-02-05:
@zachjacobs Thank you sir ... good catch.  We have now removed the old event handler after a successful boot so it is now "out of the event processing state".

However, I don't think you'll have luck with `esp_event_loop_init()`.  What this does is start a task to process events.  This task can only be created / initialized once and has already been created / initialized when we started boot processing.  However ... given that you want to process your own events AFTER the WiFi environment has already started, you can plugin in your own event handler by calling `esp_event_loop_set_cb()`.   The purpose of that function (which is provided by ESP-IDF) is to change the callback that is invoked and I think that is what you are after.

zachjacobs, 2018-02-05:
I tested f323d99245a68904106f8f9b3abaa29dfbb634ec and it works great in my application!  Thanks for the heads up on `esp_event_loop_set_cb()`.  I didn't run across that before and it makes my life much easier. 

Thanks for the quick response!

closed 2018-02-05 by zachjacobs