nkolban/esp32-snippets #391: Notifications/notify does not work

forhuan, 2018-02-05:
I use the example BLE_client to connect EPS32 with a ble device, but the program does't run to notifyCallback.
Here is the log
```
Forming a connection to a4:c1:38:98:78:a7
 - Created client
[D][BLEClient.cpp:75] connect(): >> connect(a4:c1:38:98:78:a7)
E (4848) BT: btc_gattc_call_handler()
[D][BLEDevice.cpp:105] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_REG_EVT
[D][BLEUtils.cpp:1344] dumpGattClientEvent(): GATT Event: ESP_GATTC_REG_EVT
[D][BLEUtils.cpp:1516] dumpGattClientEvent(): [status: ESP_GATT_OK, app_id: 0x0]
[D][BLEDevice.cpp:105] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_CONNECT_EVT
[D][BLEUtils.cpp:1344] dumpGattClientEvent(): GATT Event: ESP_GATTC_CONNECT_EVT
[D][BLEUtils.cpp:1374] dumpGattClientEvent(): [conn_id: 0, remote_bda: a4:c1:38:98:78:a7]
[D][BLEDevice.cpp:105] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_OPEN_EVT
[D][BLEUtils.cpp:1344] dumpGattClientEvent(): GATT Event: ESP_GATTC_OPEN_EVT
[D][BLEUtils.cpp:1469] dumpGattClientEvent(): [status: ESP_GATT_OK, conn_id: 0, remote_bda: a4:c1:38:98:78:a7, mtu: 23]
[D][BLEClient.cpp:104] connect(): << connect(), rc=1
 - Connected to server
```

```
 - Found our service
[D][BLERemoteService.cpp:165] retrieveCharacteristics(): >> getCharacteristics() for service: 0000ffe0-0000-1000-8000-00805f9b34fb
[D][BLERemoteService.cpp:250] retrieveCharacteristics(): Found a characteristic: Handle: 19, UUID: 0000ffe1-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:40] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 19 0x19, uuid: 0000ffe1-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:283] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: 0000ffe1-0000-1000-8000-00805f9b34fb
[E][BLERemoteCharacteristic.cpp:314] retrieveDescriptors(): 
...
[D][BLERemoteCharacteristic.cpp:472] registerForNotify(): >> registerForNotify(): Characteristic: uuid: 0000ffe1-0000-1000-8000-00805f9b34fb, handle: 19 0x13, props: broadcast: 0, read: 0, write_nr: 0, write: 0, notify: 1, indicate: 0, auth: 0
[D][BLEDevice.cpp:105] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_REG_FOR_NOTIFY_EVT
[D][BLEUtils.cpp:1344] dumpGattClientEvent(): GATT Event: ESP_GATTC_REG_FOR_NOTIFY_EVT
[D][BLEUtils.cpp:1532] dumpGattClientEvent(): [status: ESP_GATT_OK, handle: 19 0x13]
[D][BLERemoteCharacteristic.cpp:503] registerForNotify(): << registerForNotify()
We are now connected to the BLE Server.

```
and it does't print anymore....

I try to connect to the device with my Phone using FastBLE and my device notify ok

I think we can test it using example BLE_notify and BLE_client
emmm.... But I only have one ESP32 :(

chegewara, 2018-02-05:
Can you paste please notify callback?

forhuan, 2018-02-05:
I didn't modify the callback function.
```c++
static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
}
```

nkolban, 2018-02-05:
From what I can see here, it appears that your ESP32 is acting a BLE client to a remote BLE server.  It connects to the service with UUID `0000ffe0-0000-1000-8000-00805f9b34fb`, finds a characteristics with UUID `0000ffe1-0000-1000-8000-00805f9b34fb` and then registers for a notification on that characteristic.  What that means is that when the BLE server maintained value of the characteristic changes, the BLE server will publish a notification of the new value.

From your story, I am sensing that we aren't seeing any messages being issued from the notification callback.  Do we have confidence that the BLE server is in fact changing its characteristic values?

forhuan, 2018-02-05:
In my story, if i don't do some authentication, My BLE server keep notify the same value.
I want to figure out that My BLE server is really notifying instead of the app read the value every interval (Sorry for my poor English ,)

I download the example BLE_notify to ESP32 and connet it using my phone
[image]

```c++
void loop() {

  if (deviceConnected) {
    Serial.printf("*** NOTIFY: %d ***\n", value);
    pCharacteristic->setValue(&value, 1);
    pCharacteristic->notify();
    //pCharacteristic->indicate();
    value++;
  }
  delay(2000);
}
```

then I Connect my BLE Server, it notify about every 200ms , although the value didn't change.
[image]

pit001, 2018-02-06:
```
if (pRemoteCharacteristic_RX->canIndicate() == true) {
    Serial.println("{\"debug\":\"Indicate supported\"}");
  } else {
    Serial.println("{\"debug\":\"Indicate not supported\"}");
  }

  if (pRemoteCharacteristic_RX->canNotify() == true) {
    Serial.println("{\"debug\":\"Notify supported\"}");
  } else {
    Serial.println("{\"debug\":\"Notify not supported\"}");
  }
...
// Read the value of the characteristic. Very important.
  std::string value = pRemoteCharacteristic_RX->readValue();
...
const uint8_t bothOff[]       = {0x0, 0x0};
const uint8_t notificationOn[] = {0x1, 0x0};
const uint8_t indicationOn[]   = {0x2, 0x0};
const uint8_t bothOn[]         = {0x3, 0x0};
  pRemoteCharacteristic_RX->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)indicationOn, 2, true);
```

forhuan, 2018-02-06:
thanks for ur code,
```c++
    Serial.println(" - Found our characteristic");
    
    Serial.println(pRemoteCharacteristic->canNotify());
    Serial.println(pRemoteCharacteristic->canRead());
    Serial.println(pRemoteCharacteristic->canIndicate());
    
    Serial.println(pwRemoteCharacteristic->canNotify());
    Serial.println(pwRemoteCharacteristic->canWrite());         

    pwRemoteCharacteristic->writeValue(con,9);                           //auth
```

Hear is the Serial output
```
 - Found our characteristic
1
0
0
0
1
```

if I auth success , my device will show a bluetooth icon , I'm sure the authentication is successful. 
after the authentication , the value notify like this

[image]

pit001, 2018-02-06:
add this in your init of the notifications on the client side:

pRemoteCharacteristic->readValue();
pwRemoteCharacteristic->readValue();
const uint8_t notificationOn[] = {0x1, 0x0};
pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)notificationOn, 2, true);

forhuan, 2018-02-06:
the pRemoteCharacteristic->canRead() returns false ...
but it doesn't matter if i read the Characteristic
```C++
 if (connected) {
      // Read the value of the characteristic.
    std::string value = pRemoteCharacteristic->readValue();
    byte buf[64]= {0};
    memcpy(buf,value.c_str(),value.length());
    Serial.print("The characteristic value was: ");
    for(int i=0; i < value.length(); i++)
    {
      Serial.print(buf[i],HEX);
      Serial.print(" ");
     }
     Serial.println();
    
  }
```

```
The characteristic value was: 12 F 20 A7 78 98 38 C1 A4 36 3 36 0 1 5 0 0 0 0 0 
[D][BLERemoteCharacteristic.cpp:433] readValue(): >> readValue(): uuid: 0000ffe1-0000-1000-8000-00805f9b34fb, handle: 19 0x13
[D][BLEDevice.cpp:105] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_READ_CHAR_EVT
[D][BLEUtils.cpp:1344] dumpGattClientEvent(): GATT Event: ESP_GATTC_READ_CHAR_EVT
[D][BLEUtils.cpp:1493] dumpGattClientEvent(): [status: ESP_GATT_OK, conn_id: 0, handle: 19 0x13, value_len: 20]
[D][BLERemoteCharacteristic.cpp:455] readValue(): << readValue(): length: 20
The characteristic value was: 12 F 20 A7 78 98 38 C1 A4 36 3 36 0 1 5 0 0 0 0 0 
[D][BLERemoteCharacteristic.cpp:433] readValue(): >> readValue(): uuid: 0000ffe1-0000-1000-8000-00805f9b34fb, handle: 19 0x13
[D][BLEDevice.cpp:105] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_READ_CHAR_EVT
[D][BLEUtils.cpp:1344] dumpGattClientEvent(): GATT Event: ESP_GATTC_READ_CHAR_EVT
[D][BLEUtils.cpp:1493] dumpGattClientEvent(): [status: ESP_GATT_OK, conn_id: 0, handle: 19 0x13, value_len: 20]
[D][BLERemoteCharacteristic.cpp:455] readValue(): << readValue(): length: 20
The characteristic value was: 12 F 20 A7 78 98 38 C1 A4 36 3 36 0 1 5 0 0 0 0 0 
```
The characteristic value is correct , 
but  the value should be change when i auth  ok .
later i will show you a gif

robertklep, 2018-07-06:
@forhuan not sure if this is still a issue (it's been a while since this issue was opened), but I ran into the same problem.

I was able to fix it (at least for my use case) and issued a PR (nkolban/ESP32_BLE_Arduino#17).

chegewara, 2018-07-06:
@robertklep thanks for patch, but the problem is that PR in ESP32_BLE_Arduino wont be merged. Its because this repo is master and arduino repo is just rewrited from this repo, so after new version realease your patch will vanish.

robertklep, 2018-07-06:
@chegewara oh dang :( I'll issue a PR against this repo then!

hansmbakker, 2018-12-23:
@pit001 and @robertklep thank you for your code!
For `arduino-esp32 1.0.0` I see that both your contributions are required to get notifications working: #595 and the code below are not merged in there.
```c
const uint8_t notificationOn[] = {0x1, 0x0};
pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)notificationOn, 2, true);
```

The code from @pit001 is added as https://github.com/nkolban/esp32-snippets/blob/c48cb19186744f5792b37060b4ae9b1c36b422df/cpp_utils/BLERemoteCharacteristic.cpp#L461-L464.

closed 2018-12-29 by chegewara