nkolban/esp32-snippets #394: Update WifiEventHandler and support Ethernet interface

mseerig, 2018-02-07:
Hey!
I have updated the WifiEventHandler Class. Now all idf-supported events are included, als well as the ethernet events. To support also the ethternet interface, i wrote a suitable class. This is totaly not a final and good looking peace of code, but it works so faw. 

By adding the etheret events to the "WiFiEventHandler" I relaced, that the name WifiEvent is not longer the best choise.. Maybe, it's better to change that to NetworkEventHandler. ;)
In the attachments are my current Ethernet class, als well as the new, updated NetworkEventHandler class. 

What do you think, makes this sense?

[NetworkEventHandler.zip](https://github.com/nkolban/esp32-snippets/files/1703220/NetworkEventHandler.zip)

burkulesomesh43, 2018-08-13:
Hi,
I am using esp32 gateway.
I want use ethernet and wifi both simaltaneously.
is it possible??
if yes, then how it can be use?

pinki, 2018-12-14:
I think it's great to have both, Ethernet and WiFi connectivity.