nkolban/esp32-snippets #396: Updating the BLE_server code in ESP32_BLE_Arduino repo

SindhujaNarra, 2018-02-07:
Hello @nkolban,

I had faced the same issue (#144) as @weizenbock when I was testing BLE_server and BLE_client in Arduino IDE with 2 ESP32 DevKitC devices (one as the server and another as the client). It worked only after I made changes to the BLE_server code as suggested by @chegewara i.e., to reduce the length of device name to 3 (5 characters device name didn't work for me) and adding "pAdvertising->addServiceUUID(pService->getUUID());" to the code.

I have seen that few others have the similar issue (#144 #145 #163). So, I was wondering if you could please update the code for BLE_server in ESP32_BLE_Arduino repo so that in future no one else faces this same issue.

Thanks

pit001, 2018-03-01:
you can do it on yourself:

Part of my script:

copy the \cpp_utils\BLE*.* and \cpp_utils\HIDTypes.* files to your Arduino hardware\espressif\esp32\libraries\BLE\src directory

delete the \BLE\src\BLEExceptions.* again (or do not copy it :-) )

comment all lines of `#include "BLEExceptions.h"` -> `//#include "BLEExceptions.h"`
also comment the lines beginning with `throw new BLEUuidNotFoundException` and `throw BLEDisconnectedException();`

enjoy

@nkolban : maybe you can set the exception-handling in the not #defined part for the Arduino IDE !?

-Pit

devinsewell, 2018-10-13:
This worked for me:  pAdvertising->addServiceUUID(pService->getUUID()); before pAdvertising->start(); (my device name is 20 characters no problem)

chegewara, 2018-10-13:
@devinsewell And i can tell that your UUID is 16 bit type, but when you want to addServiceUUID with UUID 128 bit long it wont work. I will fix it soon, but there is many more important bugs and issues to fix.

devinsewell, 2018-10-13:
Ah the 20 character device name worked then stopped, changed to a 3 character name it fixed

closed 2018-11-29 by chegewara