nkolban/esp32-snippets #397: BLE client not receiving remote characteristics indications

mkol5222, 2018-02-08:
Hello, I have adapted BLE client example for my device and unfortunately registered indication callback is never called on BLE indication received.

I have verified in source that characteristic should really be indicate one by canIndicate call (see code and the output).

I am using ESP32_BLE_Arduino version 0.4.9 in my project.

Source of my ESP32 BLE client is
https://gist.github.com/mkol5222/3321e7fcc550a21d1d8a94a3b225bf47
and full debug output is
https://gist.github.com/mkol5222/f004eb5ec54da4ae50fb35341b27cee6

Would you please be able to give me advice how to proceed 
making handling of indications from my device characteristic work?

Very same device works well in the browser with Web BLE API client implementation.

Thanks!

chegewara, 2018-02-15:
If this is all that is logged out then problem is here:
https://gist.github.com/mkol5222/3321e7fcc550a21d1d8a94a3b225bf47#file-ble_client_modified-ino-L37

Code cant pass thru this line

mkol5222, 2018-02-15:
@chegewara thanks for review. It is my mistake sharing incomplete debug.

We are able to connect and list service and characteristics and even call registration for notification by registerForNotify() call,
However ESP_GATTC_REG_FOR_NOTIFY_EVT is last GATT event received and idication is not received.

Any idea how to progress identifying cause?

 

```
[D][BLEClient.cpp:370] getServices(): << getServices
[D][BLEClient.cpp:331] getService(): << getService: found the service with uuid: c1b25000-caaf-6d0e-4c33-7dae30052840
 - Found our service
[D][BLERemoteService.cpp:167] retrieveCharacteristics(): >> getCharacteristics() for service: c1b25000-caaf-6d0e-4c33-7dae30052840
[D][BLERemoteService.cpp:198] retrieveCharacteristics(): Found a characteristic: Handle: 11, UUID: c1b25010-caaf-6d0e-4c33-7dae30052840
[D][BLERemoteCharacteristic.cpp:41] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 11 0x11, uuid: c1b25010-caaf-6d0e-4c33-7dae30052840
[D][BLERemoteCharacteristic.cpp:284] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: c1b25010-caaf-6d0e-4c33-7dae30052840
[E][BLERemoteCharacteristic.cpp:315] retrieveDescriptors(): 
[D][BLERemoteCharacteristic.cpp:316] retrieveDescriptors(): Found a descriptor: Handle: 12, UUID: 00002902-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:330] retrieveDescriptors(): << retrieveDescriptors(): Found 1 descriptors.
[D][BLERemoteCharacteristic.cpp:49] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:198] retrieveCharacteristics(): Found a characteristic: Handle: 14, UUID: c1b25011-caaf-6d0e-4c33-7dae30052840
[D][BLERemoteCharacteristic.cpp:41] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 14 0x14, uuid: c1b25011-caaf-6d0e-4c33-7dae30052840
[D][BLERemoteCharacteristic.cpp:284] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: c1b25011-caaf-6d0e-4c33-7dae30052840
[E][BLERemoteCharacteristic.cpp:308] retrieveDescriptors(): esp_ble_gattc_get_all_descr: ESP_GATT_NOT_FOUND
[D][BLERemoteCharacteristic.cpp:330] retrieveDescriptors(): << retrieveDescriptors(): Found 0 descriptors.
[D][BLERemoteCharacteristic.cpp:49] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:198] retrieveCharacteristics(): Found a characteristic: Handle: 16, UUID: c1b25012-caaf-6d0e-4c33-7dae30052840
[D][BLERemoteCharacteristic.cpp:41] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 16 0x16, uuid: c1b25012-caaf-6d0e-4c33-7dae30052840
[D][BLERemoteCharacteristic.cpp:284] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: c1b25012-caaf-6d0e-4c33-7dae30052840
[E][BLERemoteCharacteristic.cpp:308] retrieveDescriptors(): esp_ble_gattc_get_all_descr: ESP_GATT_NOT_FOUND
[D][BLERemoteCharacteristic.cpp:330] retrieveDescriptors(): << retrieveDescriptors(): Found 0 descriptors.
[D][BLERemoteCharacteristic.cpp:49] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:198] retrieveCharacteristics(): Found a characteristic: Handle: 18, UUID: c1b25013-caaf-6d0e-4c33-7dae30052840
[D][BLERemoteCharacteristic.cpp:41] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 18 0x18, uuid: c1b25013-caaf-6d0e-4c33-7dae30052840
[D][BLERemoteCharacteristic.cpp:284] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: c1b25013-caaf-6d0e-4c33-7dae30052840
[E][BLERemoteCharacteristic.cpp:315] retrieveDescriptors(): 
[D][BLERemoteCharacteristic.cpp:316] retrieveDescriptors(): Found a descriptor: Handle: 19, UUID: 00002902-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:330] retrieveDescriptors(): << retrieveDescriptors(): Found 1 descriptors.
[D][BLERemoteCharacteristic.cpp:49] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:214] retrieveCharacteristics(): << getCharacteristics()
 - Found our characteristic
Can indicate?: YES
Can notify?: NO
We are now connected to the BLE Server.
registering for notification
[D][BLERemoteCharacteristic.cpp:479] registerForNotify(): >> registerForNotify(): Characteristic: uuid: c1b25010-caaf-6d0e-4c33-7dae30052840, handle: 11 0xb, props: broadcast: 0, read: 1, write_nr: 0, write: 0, notify: 0, indicate: 1, auth: 0
[D][BLEDevice.cpp:140] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_REG_FOR_NOTIFY_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_REG_FOR_NOTIFY_EVT
[D][BLEUtils.cpp:1566] dumpGattClientEvent(): [status: ESP_GATT_OK, handle: 11 0x0b]
[D][BLERemoteCharacteristic.cpp:510] registerForNotify(): << registerForNotify()
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT
[D][BLEUtils.cpp:1343] dumpGapEvent(): [status: 0, bd_addr: c8:d5:43:1e:8c:a1, min_int: 32, max_int: 48, latency: 0, conn_int: 48, timeout: 50]
[D][BLEClient.cpp:398] handleGAPEvent(): BLEClient ... handling GAP event!

```

chegewara, 2018-02-16:
Here is the thing. I did not need to check it before because i have no need to register for indicate. Seems there is no proper api to register for indicate in esp-idf, which is weird. I asked to add this to esp-idf but it may take some time.
https://github.com/espressif/esp-idf/issues/1628

You may write simple function that will register for indications:
- find characteristic you want to register for indications,
- find BLERemoteDescriptor(0x2902) on this characteristic
- write value 0x0200 to this descriptor.

mkol5222, 2018-02-23:
@chegewara many thanks for hint. You are corrent - BLE client sets callback,
but remote device was not told to turn on indications.

This can be fixed in the code by
```
const uint8_t indicationOn[] = {0x2, 0x0};
pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)indicationOn, 2, true);
```
exactly as you have suggested.

Andreas Spiess' video was great help explaining the principles of BLE notifications/indications with ESP32
https://www.youtube.com/watch?v=osneajf7Xkg
Related code https://github.com/SensorsIot/Bluetooth-BLE-on-Arduino-IDE

closed 2018-02-23 by mkol5222

cristian-spiescu, 2020-10-18:
Thanks @chegewara and @mkol5222. I'd like to add something that I found out by trial/error:

```c++
        pRemoteCharacteristicReceive->registerForNotify(notifyCallback);
        
        const uint8_t indicationOn[] = {0x2, 0x0};
        pRemoteCharacteristicReceive->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)indicationOn, 2, true);
```

The call to `registerForNotify()` needs to happen BEFORE activating the indication. Initially I was invoking `registerForNotify()` after, which doesn't seem to work.