nkolban/esp32-snippets #398: ESP32 C++ with MQTT

Koulwa, 2018-02-09:
So far, have loved the nkolban/esp32-snippets. many thanks to nkolban!

Really, just looking for the best way to include MQTT with C++ only using the IDF (NOT Arduino).

Trying to get MQTT to work, while using the other C++ libraries (namely wifi boot and BLE scan) to work as well. I really don't need anything super fancy for the time. I was trying to use the AWS Cpp library.

I'm new to binding c/c++ code together, so i'm trying to understand how to pull it off. 
Following the logic BLEDevice::init("");
I implemented 
AWS::init("192.168.0.132", 1883);

but get compiler errors, I included the "AWS" options in make config and compiled fine, but then had call back panic error/reboot.

I don't plan on using AWS, but was trying to use this for MQTT.

Also, was able to get another library for MQTT working on the ESP32, https://github.com/tuanpmt/esp32-mqtt by including the c files and wrapping them much like the app_main() wrapping. 

This seems like i'm hacking it together though. I'd like a clean way to use an MQTT library that gets updated/bug fixed in the future and seems like this would be a common enough thing that others would want to get used.

I then looked at the paho examples, which were in C, and looked at the C++ versions but were only for Linux and Arduino which uses the stack classes, and couldn't get that to compile either.

So, is there a C++ example that would work with the ESP32 in conjunction with the CPP_UTILS?

Thanks!

mseerig, 2018-02-09:
Hey there.
I had the exactly same problems. I tryed also to use the AWS class at my MQTT Client, but without success. Than, I looked for a other solution, but there are no one witch is a clean C++ and esp32 suitable solution.

For me, I chooceed the way to port the Arduino PubSubClient  in a way, that I use the CPP_UTILS in this class. Here are my PubSubClient class for the cpp_utils and an example. Pleace be patient, I will create al pullrequest, or an extra projekt to public this.

Hope, that helps. ;)
[PubSubClient.zip](https://github.com/nkolban/esp32-snippets/files/1710282/PubSubClient.zip)

johnhmacleod, 2018-05-25:
I have a gateway that connects to my SensorTag via BLE, then forwards the packets using MQTT to IBM Watson IoT Platform.  It all works very well with the following 3 caveats -

1) You must connect to your BLE device before attempting the WiFi connection
2) Do not use WiFiSecure (TLS) for MQTT - it never ends up connecting to the server

(MQTT using TLS works just fine when not running BLE, though it took me a while to track that down as the cause!)

joedeveloper, 2018-07-31:
I have also been having more trouble than I expected getting mqtt integrated in my ESP-IDF based project.
Some thoughts:
1. I looked at cpp_utils/PubSubClient.cpp, but couldn't convince myself that I knew how to actually hook it up - Having just now inspected the full PubSibClient.zip in this thread I see there is an implementation example - this would have been super useful to have found in esp32-snippets.

2. As a newcomer it is pretty hard to evaluate pros / cons of PAHO, AWS-MQTT (IDF Blessed?) and the 'home grown' PubSubClient.cpp implementation, is there a place that discusses them? A quick note on caveats or 'historic but discouraged' approaches would be nice.

3. As I am adding more functionality ( BLE, WiFI, MQTT, actual application features ) I am increasingly running into issues stemming from resource constraints and contention. 
3.1 Is there a guide on how to go about stripping down what gets compiled in? 
3.2 Is there a guide or design pattern implementation for on-demand start / stop of services /  components?
3.3 Does BLE _really_ need to be that resource hungry?

4. I am currently trying to use the ESP-IDF AWS mqtt lib with what seems a high volume of messages - imagine replacing ESP_LOGI with mqtt calls, has anyone made wrappers that support anything in the vein of async / queuing / bundling / discard rules? Is there interest in something like that or am i just doing it wrong?

closed 2023-10-14 by HiveTechDev