nkolban/esp32-snippets #399: BLE crashed when requesting a characteristic

borin-z, 2018-02-09:
## Hardware:
Board: ESP32 Dev Module
Core Installation/update date: Yesterday
IDE name: Arduino IDE
Flash Frequency: 80MHz
Upload Speed: 921600

## Description:
There is xiaomi mijia temperature + ESP32 with BLE.  I would like to receive various characteristics from the device. I decided to start with the simplest characteristics, the battery charge. 
I used the code from the example of ESP32 BLE Arduino:

<pre>#include

 "BLEDevice.h"

void setup() {
  Serial.begin(115200);
  Serial.println("Starting Arduino BLE Client application...");

  BLEAddress addr("4C:65:A8:D0:B1:68");
  BLEUUID srv("0000180f-0000-1000-8000-00805f9b34fb");
  BLEUUID btr("226cbb55-6476-4566-7562-66734470666d");

  BLEDevice::init("");
  BLEClient* pClient = BLEDevice::createClient();
  if (pClient->connect(addr)){
    Serial.println("Connection OK");
    BLERemoteService* pRemoteService = pClient->getService(srv);
    BLERemoteCharacteristic* pRemoteCharacteristic = pRemoteService->getCharacteristic(btr);
  } 
}
  
void loop() {}</pre>

<pre>Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
Starting Arduino BLE Client application...
E (882) BT: btc_gattc_call_handler()
Connection OK
abort() was called at PC 0x4014991b on core 1

Backtrace: 0x4008a77c:0x3ffd3610 0x4008a87b:0x3ffd3630 0x4014991b:0x3ffd3650 0x40149962:0x3ffd3670 0x4013742b:0x3ffd3690 0x400d4e59:0x3ffd36b0 0x400d2bf2:0x3ffd3710 0x401518e6:0x3ffd3780</pre>

<pre># gatttool -b 4C:65:A8:D0:B1:68 --char-read -u 00002a19-0000-1000-8000-00805f9b34fb
handle: 0x0018   value: 64</pre>

<pre>
Decoding 8 results
0x4008a77c: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 578
0x4008a87b: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 578
0x4014991b: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 112
0x40149962: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 112
0x4013742b: __cxa_throw at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc line 87
0x400d4e59: BLERemoteService::getCharacteristic(BLEUUID) at c:\users\borin\documents\arduino\hardware\espressif\esp32\tools\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/ios_base.h line 1014
0x400d2bf2: setup() at C:\Users\borin\Desktop\gatt/gatt.ino line 59
0x401518e6: loopTask(void*) at C:\Users\borin\Documents\Arduino\hardware\espressif\esp32\cores\esp32/main.cpp line 15
</pre>

## Debug mode
The result is the characteristic is, and then panic falls.
<pre>ets Jun  8 2016 00:22:57
[... 63 lines cut ...]
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_SEARCH_RES_EVT
[D][BLEUtils.cpp:1602] dumpGattClientEvent(): [conn_id: 0, start_handle: 45 0x2d, end_handle: 65535 0xffff, srvc_id: uuid: 0000fe95-0000-1000-8000-00805f9b34fb, inst_id: 1
[D][BLERemoteService.cpp:29] BLERemoteService(): >> BLERemoteService()
[D][BLERemoteService.cpp:37] BLERemoteService(): << BLERemoteService()
[D][BLEDevice.cpp:140] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_SEARCH_CMPL_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_SEARCH_CMPL_EVT
[D][BLEUtils.cpp:1581] dumpGattClientEvent(): [status: ESP_GATT_OK, conn_id: 0]
[D][BLEClient.cpp:370] getServices(): << getServices
[D][BLEClient.cpp:331] getService(): << getService: found the service with uuid: 0000180f-0000-1000-8000-00805f9b34fb
[D][BLERemoteService.cpp:167] retrieveCharacteristics(): >> getCharacteristics() for service: 0000180f-0000-1000-8000-00805f9b34fb
E ([D][BLERemoteService.cpp:198] retrieveCharacteristics(): Found a characteristic: Handle: 24, UUID: 00002a19-0000-1000-8000-00805f9b34fb
77[D][BLERemoteCharacteristic.cpp:41] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 24 0x24, uuid: 00002a19-0000-1000-8000-00805f9b34fb
3[D][BLERemoteCharacteristic.cpp:284] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: 00002a19-0000-1000-8000-00805f9b34fb
[E][BLERemoteCharacteristic.cpp:315] retrieveDescriptors(): 
5[D][BLERemoteCharacteristic.cpp:316] retrieveDescriptors(): Found a descriptor: Handle: 25, UUID: 00002902-0000-1000-8000-00805f9b34fb
) [D][BLERemoteCharacteristic.cpp:330] retrieveDescriptors(): << retrieveDescriptors(): Found 1 descriptors.
[D][BLERemoteCharacteristic.cpp:49] BLERemoteCharacteristic(): << BLERemoteCharacteristic
B[D][BLERemoteService.cpp:214] retrieveCharacteristics(): << getCharacteristics()
T: No pending command
abort() was called at PC 0x4014da1b on core 1

Backtrace: 0x4008a778:0x3ffd35c0 0x4008a877:0x3ffd35e0 0x4014da1b:0x3ffd3600 0x4014da62:0x3ffd3620 0x4013b35f:0x3ffd3640 0x400d6895:0x3ffd3660 0x400d3262:0x3ffd36c0 0x40155ac6:0x3ffd3730

Rebooting...
</pre>

pit001, 2018-03-01:
#419

pit001, 2018-03-11:
additional info: xiaomi devices sometimes need to be initialized, if not they disconnect after some ms/s. Check it in parallel with the nordic app.

EvgenyBatalin, 2018-07-18:
Any news on that?

fransakeson, 2019-05-17:
Same issue when trying to getCharacteristic on a iGrill3 thermometer. iGrill, like some Xiaomi devices need initialization/authentication. In the case of iGrill, you need to write a challenge to a specific characteristic (which is write capable), get a reply back through another characteristic (read capable), and finally after applying AES encryption to the reply, send that to a third characteristic. So in short:
1. Connect to device, service UUID _64ac0000-4a4b-4b58-9f37-94d3c52ffdf7_
1. Write "0000000000000000" to characteristic UUID _64ac0002-4a4b-4b58-9f37-94d3c52ffdf7_ (yes, that's a flaw in the iGrill device, using all zeros makes us not need the AES :-))
2. Read characteristic UUID _64ac0003-4a4b-4b58-9f37-94d3c52ffdf7_
3. Write the same string/bytearray to characteristic UUID _64ac0004-4a4b-4b58-9f37-94d3c52ffdf7_
4. Done!

Now, since all getCharacteristic calls makes the ESP32 crash, we can't progress.

chegewara, 2019-05-17:
Try this just before your code crash:
`size_t val = uxTaskGetStackHighWaterMark(NULL)`.

It will return minimum ever free stack size.

fransakeson, 2019-05-18:
Many thanks for your replay, @chegewara. I found that the patched BLERemoteService.cpp from https://github.com/wakwak-koba/ESP32_BLE_Arduino/blob/master/src/ fixes my problem. So this seems to be a duplicate to https://github.com/nkolban/esp32-snippets/issues/765

evanthomas, 2020-01-17:
I'm not sure what the status of this defect is or whether there is a release anywhere where it is fixed. I've just spent many hours fixing this and another defect in this library. The `retrieveCharacteristics` function not does not only fail to correctly allocate enough memory (as others have pointed out) but seems to be using the underlying esp-idf API incorrectly. This works for me:
```
void BLERemoteService::retrieveCharacteristics() {
	ESP_LOGD(LOG_TAG, ">> getCharacteristics() for service: %s", getUUID().toString().c_str());

	removeCharacteristics(); // Forget any previous characteristics.

	uint16_t offset = 0;
	esp_gattc_char_elem_t results[10];
    uint16_t count = 10;  // this value is used as in parameter that allows to search max 10 chars with the same uuid
    esp_gatt_status_t status = ::esp_ble_gattc_get_all_char(
        getClient()->getGattcIf(),
        getClient()->getConnId(),
        m_startHandle,
        m_endHandle,
        results,
        &count,
        offset
    );

    if (status != ESP_GATT_OK) {   // If we got an error, end.
        ESP_LOGE(LOG_TAG, "esp_ble_gattc_get_all_char: %s", BLEUtils::gattStatusToString(status).c_str());
        return;
    }

    for (int i = 0; i < count; i ++) {
        esp_gattc_char_elem_t &result = results[i];
		ESP_LOGD(LOG_TAG, "Found a characteristic: Handle: %d, UUID: %s", result.char_handle, BLEUUID(result.uuid).toString().c_str());

		// We now have a new characteristic ... let us add that to our set of known characteristics
		BLERemoteCharacteristic *pNewRemoteCharacteristic = new BLERemoteCharacteristic(
			result.char_handle,
			BLEUUID(result.uuid),
			result.properties,
			this
		);

		m_characteristicMap.insert(std::pair<std::string, BLERemoteCharacteristic*>(pNewRemoteCharacteristic->getUUID().toString(), pNewRemoteCharacteristic));
		m_characteristicMapByHandle.insert(std::pair<uint16_t, BLERemoteCharacteristic*>(result.char_handle, pNewRemoteCharacteristic));
	}

	m_haveCharacteristics = true; // Remember that we have received the characteristics.
	ESP_LOGD(LOG_TAG, "<< getCharacteristics()");
} // getCharacteristics
```
i.e. only one call to `esp_ble_gattc_get_all_char` is necassary.

In addition `m_rawData` needs to be set to zero in the `BLERemoteCharacteristic` constructor otherwise the event handler tries to free it which leads to tears.