nkolban/esp32-snippets #405: BLE Crashes when setting device name to `com.companyname.peripheral`.

eman6576, 2018-02-14:
Hello,

When initializing a `BLEDevice` like this:
```c++
BLEDevice::init("com.companyname.peripheral");
server = BLEDevice::createServer();
BLEAdvertising *advertising = server->getAdvertising();
advertising->start();
```
BLE crashes after beginning to advertise. But when creating a `BLEDevice` like this:
```c++
BLEDevice::init("MyCompanyl");
server = BLEDevice::createServer();
BLEAdvertising *advertising = server->getAdvertising();
advertising->start();
```
It doesn't crash. Why does this happen?

chegewara, 2018-02-15:
Hi,
i just did quick test and its working in my example. Even with this very long string name it works:
`BLEDevice::init("com.companyname.peripheral");`

Are working on esp-idf or arduino-ide? What commit are you using?

eman6576, 2018-02-15:
I'm using the arduino-ide with the arduino-esp32 library.