nkolban/esp32-snippets #406: BLE notifications issues

chegewara, 2018-02-15:
Hi,
there is a lot (way too much) opened issues about problem with notifications andor notification callbacks. I would like to answer all and help somehow but i lost track of it and i decided to open this issue and try to help all in here. 

Im not saying there is no issue, but i just compiled 2 simple examples on esp-idf, flashed it to 2 esp32 and client is connecting to server with no problem, registering to notifications and receiving notifications. 
I can understand there may be some issue trying to communicate with other products and like i said, i will try to help with all issues.

Mistikan, 2018-02-17:
Hello! I do not receive a response to a message that is processed in the onWrite function. If I understood correctly, this is due to the fact that the 2 characteristics are in one pCharacheristic. I tried to smash them, but nothing happened.
I attach a screenshot of the code and send the message through uart.
[image]
How can I better solve this problem?

chegewara, 2018-02-17:
Hi, if you are using BLE_uart code without changing characteristic setting then you are making mistake. 
Characteristic that is receiving write requests cant send notifications, it is characteristic that only receive data, and the other characteristic is to send data from esp32.

But it is strange how this code works. You should be able to change characteristic value with this code which clearly is not happened, and code should not allow you to send notifications, which i never check before because i never thought to about sending notifications from characteristic that is not setup to do it. I will try to find the problem.

chegewara, 2018-02-17:
@Mistikan 
I see the problem, now i have to find the cause of it.
When you are trying to send notification code just stall and causing diconnection, am i right?

Mistikan, 2018-02-17:
@chegewara
Yes, you right.

chegewara, 2018-02-17:
@Mistikan 
Im guessing it has something to do with semaphores, but semaphores is mr. Neil's job (@nkolban). I know nothing about it, sorry.

GDX64, 2018-02-18:
Hi, sorry for my bad english, I'm not a native english speaker. I'm from Brazil,  :) 

Maybe it's not the correct place to ask this, because it is not quite a real issue. If it is the case, just tell me and I will delete the question. I don't know the github either, I created this account just to ask this question. hahah 

I'm working with some robots for a Very Small Size Soccer (VSSS) team, and I'm trying to use the ESP32 to do the communication between the computer and the robots. I will be sending only speed comands to the motors, two int16 numbers. There are three robots and I was thinking about using one esp32 as a server and send the data to the other three esp32. I must do this about every 10ms, sending two int16 numbers (4 bytes) to three clients every 10ms.

I would like to know if is it possible using the BLE, or using the Wifi. I tryed to check the maximum rate that I can send data using the BLE exemples and I could only reach about 30ms between the packages sending data to only one client. I don't know if is there some limit defined by the current library.

It may be that it's not a good idea to use the esp32 for that, or the BLE, So whatever you may say to me will be useful, and I will be very grateful.

Mistikan, 2018-02-25:
@chegewara Hello.
It seems I found the perfect solution. We need to create a Serial analogue for Bluetooth, and I did it. Could you estimate my code and what should we do to integrate it into nkolban's repository?
Link: https://yadi.sk/d/XGb14z3x3SkzRH
Example:
[image]
[image]
P.S. just put these two files in libraries/BLE/src
P.P.S. there are still some bugs, but the core works

chegewara, 2018-02-25:
Hi @Mistikan, 
You can always create PR and mr @nkolban will merge it if he accept code.

Mistikan, 2018-02-27:
@chegewara Hello.
Look up:
https://github.com/hobzcalvin/ESP32_BLE_Arduino/blob/42a93a9bd0da6bcdd0f160da3aa76671b3bd075c/src/BLESerial.cpp
https://github.com/hobzcalvin/ESP32_BLE_Arduino/blob/42a93a9bd0da6bcdd0f160da3aa76671b3bd075c/src/BLESerial.h
It's already written ...

closed 2018-11-26 by chegewara