nkolban/esp32-snippets #408: Memory leak when using BLEscan

1technophile, 2018-02-16:
Hello,

Thanks for all this work on ESP32 a lot of data for working with this really powerfull chip!

I'm working on presence detection or sensor integration with BLE and our ESP32s (BLE to MQTT solution). After a few hours the ESP32 disconnects from the wifi or are stuck (https://github.com/1technophile/OpenMQTTGateway/issues/173)

After some analysis I found what seems to be a memory leak when using BLE scan.

I took the BLE scan example and run it every 20 seconds by checking RAM remaining:

```
/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

int scanTime = 10; //In seconds
//Time used to wait for an interval before checking system measures
unsigned long timer_sys_measures = 0;

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
    }
};

void setup() {
  Serial.begin(115200);
  BLEDevice::init("");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);

  unsigned long now = millis();
  if (now > (timer_sys_measures + 20000)) {//retriving value of temperature and humidity of the box from DHT every xUL
    timer_sys_measures = millis();
    Serial.println("Remaining memory");
    uint32_t freeMem;
    freeMem = ESP.getFreeHeap();
    Serial.println(freeMem);
    
    Serial.println("Scanning...");
    BLEScan* pBLEScan = BLEDevice::getScan(); //create new scan
    pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
    pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
    BLEScanResults foundDevices = pBLEScan->start(scanTime);
    Serial.print("Devices found: ");
    Serial.println(foundDevices.getCount());
    Serial.println("Scan done!");
  }
}
```

Here are my results:
```
Remaining memory
131032
Scanning...
Advertised Device: Name: , Address: xxxxxxxxxx, manufacturer data: xxxxxxxxxx, serviceUUID: xxxxxxxxxx 
Advertised Device: Name: Flower care, Address: xxxxxxxxxx, serviceUUID: xxxxxxxxxx
Advertised Device: Name: Flower care, Address: xxxxxxxxxx, serviceUUID: xxxxxxxxxx
E (43006) BT: btc_search_callback  BLE observe complete. Num Resp 3

Devices found: 3
Scan done!
Remaining memory
129424
Scanning...
Advertised Device: Name: , Address: xxxxxxxxxx, manufacturer data: xxxxxxxxxx, serviceUUID: xxxxxxxxxx 
Advertised Device: Name: Flower care, Address: xxxxxxxxxx, serviceUUID: xxxxxxxxxx
Advertised Device: Name: Flower care, Address: xxxxxxxxxx, serviceUUID: xxxxxxxxxx
E (63009) BT: btc_search_callback  BLE observe complete. Num Resp 3

Devices found: 3
Scan done!
Remaining memory
129404
Scanning...
Advertised Device: Name: , Address: xxxxxxxxxx, manufacturer data: xxxxxxxxxx, serviceUUID: xxxxxxxxxx 
Advertised Device: Name: Flower care, Address: xxxxxxxxxx, serviceUUID: xxxxxxxxxx
Advertised Device: Name: Flower care, Address: xxxxxxxxxx, serviceUUID: xxxxxxxxxx
E (83012) BT: btc_search_callback  BLE observe complete. Num Resp 3

Devices found: 3
Scan done!
Remaining memory
129380
Scanning...
Advertised Device: Name: , Address: xxxxxxxxxx, manufacturer data: xxxxxxxxxx, serviceUUID: xxxxxxxxxx 
Advertised Device: Name: Flower care, Address: xxxxxxxxxx, serviceUUID: xxxxxxxxxx
Advertised Device: Name: Flower care, Address: xxxxxxxxxx, serviceUUID: xxxxxxxxxx
E (103015) BT: btc_search_callback  BLE observe complete. Num Resp 3

Devices found: 3
Scan done!
Remaining memory
129360
```
We can see that remaining memory decreases over the time.

Any idea about an incorrect implementation in my side?
Or maybe you need more infos about this?

nkolban, 2018-02-16:
In your source code, you have the following line:

```
pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
```

When ever you see a `new` call, this allocates storage (in this case for a new instance of a class).  When ever you explicitly `new` data, someone, somewhere has to release it.  Have a look at the following code and see if this corrects the issue.

```
/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

int scanTime = 10; //In seconds
//Time used to wait for an interval before checking system measures
unsigned long timer_sys_measures = 0;

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
    }
};

void setup() {
  Serial.begin(115200);
  BLEDevice::init("");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);

  unsigned long now = millis();
  if (now > (timer_sys_measures + 20000)) {//retriving value of temperature and humidity of the box from DHT every xUL
    timer_sys_measures = millis();
    Serial.println("Remaining memory");
    uint32_t freeMem;
    freeMem = ESP.getFreeHeap();
    Serial.println(freeMem);
    
    Serial.println("Scanning...");
    BLEScan* pBLEScan = BLEDevice::getScan(); //create new scan
    MyAdvertisedDeviceCallbacks myCallbacks;
    pBLEScan->setAdvertisedDeviceCallbacks(&myCallbacks);
    pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
    BLEScanResults foundDevices = pBLEScan->start(scanTime);
    Serial.print("Devices found: ");
    Serial.println(foundDevices.getCount());
    Serial.println("Scan done!");
  }
}
```

1technophile, 2018-02-17:
This is really better thanks a lot!!

```
Remaining memory
129456
Scanning...
Advertised Device: Name: , Address: ***************, manufacturer data: ***************, serviceUUID: ***************
Advertised Device: Name: Flower care, Address: ***************, serviceUUID: *************** 
Advertised Device: Name: Flower care, Address: ***************, serviceUUID: *************** 
E (303052) BT: btc_search_callback  BLE observe complete. Num Resp 3

Devices found: 3
Scan done!
Remaining memory
129456
Scanning...
Advertised Device: Name: , Address: ***************, manufacturer data: ***************, serviceUUID: ***************
Advertised Device: Name: Flower care, Address: ***************, serviceUUID: *************** 
Advertised Device: Name: Flower care, Address: ***************, serviceUUID: *************** 
E (323055) BT: btc_search_callback  BLE observe complete. Num Resp 3

Devices found: 3
Scan done!
Remaining memory
129456
Scanning...
Advertised Device: Name: , Address: ***************, manufacturer data: ***************, serviceUUID: ***************
Advertised Device: Name: Flower care, Address: ***************, serviceUUID: *************** 
Advertised Device: Name: Flower care, Address: ***************, serviceUUID: *************** 
E (343058) BT: btc_search_callback  BLE observe complete. Num Resp 3

Devices found: 3
Scan done!
Remaining memory
129456
Scanning...
Advertised Device: Name: , Address: ***************, manufacturer data: ***************, serviceUUID: ***************
Advertised Device: Name: Flower care, Address: ***************, serviceUUID: *************** 
Advertised Device: Name: Flower care, Address: ***************, serviceUUID: *************** 
E (363061) BT: btc_search_callback  BLE observe complete. Num Resp 3

```

closed 2018-02-17 by 1technophile

Kus-t, 2018-08-04:
Hello, 

this is my first post/comment here, i don't know if it would have been better to open a new issue.

I'm trying to run a code that scan for BLE devices and open a door if a known one is detected.
I have the same issue of memory decreasing so i tried to make the modifications you mentioned above but the problem still remain.

I suppose it could be due to `pServerAddress = new BLEAddress(advertisedDevice.getAddress());`
but I didn't find how to change it and still get the same results.

Could you please help me and tell me how to fix that ?

Thank you !

Here is the code :

```
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

static BLEAddress *pServerAddress;

#define KEY 22

bool deviceFound = false;

int CarOpen;
int BuzPin = 23;

String knownAddresses[] = { "90:94:97:35:c5:66"};

 unsigned long entry;

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
  Serial.print("Notify callback for characteristic ");
  Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
  Serial.print(" of data length ");
  Serial.println(length);
}

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    /**
        Called for each advertising BLE server.
    */
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.print("BLE Advertised Device found: ");
      delay(30);
      Serial.println(advertisedDevice.toString().c_str());
      pServerAddress = new BLEAddress(advertisedDevice.getAddress());
      delay(200);

      bool known = false;
      for (int i = 0; i < (sizeof(knownAddresses) / sizeof(knownAddresses[0])); i++) {
        if (strcmp(pServerAddress->toString().c_str(), knownAddresses[i].c_str()) == 0) known = true;
      }
      if (known) {
        Serial.print("Device found: ");
        Serial.println(advertisedDevice.getRSSI());
        if (advertisedDevice.getRSSI() > -85) deviceFound = true;
        else deviceFound = false;
        Serial.println(pServerAddress->toString().c_str());
        advertisedDevice.getScan()->stop();
        delay(200);
      }
    }
}; // MyAdvertisedDeviceCallbacks

void setup() {
  Serial.begin(115200);
  Serial.println("Starting Arduino BLE Client application...");
  pinMode(KEY, OUTPUT);
  
  ledcSetup(0, 5000, 13);
  ledcAttachPin(BuzPin, 0);

  BLEDevice::init("");

  CarOpen = 0;
  digitalWrite(KEY, LOW);
  Serial.println("CLOSED");
  
   ledcWriteTone(0, 400); // 4KHz
  delay(2000);                          // 583 ms
  ledcWriteTone(0, 0);
  
}

void loop() {

  Serial.println();
  
  uint32_t Memory;
    Memory = ESP.getFreeHeap();
    Serial.println(Memory);
    delay(30);
    
  Serial.println("BLE Scan restarted.....");
  delay(300);
  deviceFound = false;
  
   BLEScan* pBLEScan = BLEDevice::getScan();
   MyAdvertisedDeviceCallbacks myCallbacks;
  pBLEScan->setAdvertisedDeviceCallbacks(&myCallbacks);
  pBLEScan->setActiveScan(true);
  
  BLEScanResults scanResults = pBLEScan->start(4);
  
  if (deviceFound && CarOpen == 0) {
  
    ledcWriteTone(0, 2000); // 4KHz
    delay(200);
    ledcWriteTone(0, 0);
    
    Serial.println("open");
    digitalWrite(KEY, HIGH);
    CarOpen = 1;
    delay(2000);
  }

  if (deviceFound && CarOpen == 1) {
    Serial.println("Allready open");
    delay(2000);
  }
  
  if (deviceFound == false && CarOpen == 1) {
    
    ledcWriteTone(0, 2000); // 4KHz
    delay(200);
    ledcWriteTone(0, 0);
    delay(100);
    ledcWriteTone(0, 2000); // 4KHz
    delay(200);
    ledcWriteTone(0, 0);
    
    Serial.println("closed");
    digitalWrite(KEY, LOW);
    CarOpen = 0;
    delay(500);
  }

  if (deviceFound == false && CarOpen == 0) {
    Serial.println("Allready closed");;
    delay(500);
  }
} // End of loop

```

patlunb, 2018-10-10:
@nkolban 
Seems to me I have the same issue.

Using  Arduino 1.8.5. BLE libraries latest.

Attaching simplified program. MACs and service UUIDs have been hidden (sorry for txt extension): 
[sketch_oct10a.txt](https://github.com/nkolban/esp32-snippets/files/2464554/sketch_oct10a.txt)

Here's the getFreeHeap output (after each connection and read cycle):

 - FREE HEAP SIZE: 101676
 - FREE HEAP SIZE: 95804
 - FREE HEAP SIZE: 89924
 - FREE HEAP SIZE: 84060

And it goes all the way down to <20000, and then crashes and reboots.
Scans without connection (no devices found) doesn't consume any memory, but if the connection is made, then up to 6KB is gone in one go.

Any advice is appreciated, thanks!

patlunb, 2018-10-18:
Thanks, the issue was resolved. 

Please disregard previous post

RoyOltmans, 2019-10-05:
Patlunb I would appreciate the solution...