nkolban/esp32-snippets #410: BLEScan::start - blocks when duration is 0 (infinite)

QuadCorei8085, 2018-02-23:
`BLEScanResults BLEScan::start(uint32_t duration)`

can be called with duration=0, the C function `esp_ble_gap_start_scanning` will then scan indefinetly.
The callback mechanism however can not be used in this case as the semaphore `m_semaphoreScanEnd.wait("start");` will block the function from returning.

Simple use case of duration=0 is when you just listen and want to use the BLE as beacon locator (e.g. RSSI based distance).

I suggest to introduce `start_async()` where the function will not return a list of found devices but requires a `BLEAdvertisedDeviceCallbacks* pAdvertisedDeviceCallbacks`

beats0126, 2018-05-24:
it is this is what u mention? [https://github.com/nkolban/esp32-snippets/issues/496](https://github.com/nkolban/esp32-snippets/issues/496)

QuadCorei8085, 2018-05-24:
It was a long time ago. I used instead the C implementation with the
callbacks available.
I did not mean to create a separate task but that is also an option (my
point was only that with duration=0 a callback would be enough to register
using the CPP api).