nkolban/esp32-snippets #412: Error running SampleServer_authentication_numeric_confirmation

thomasvdwege, 2018-02-23:
I'm trying to run the examples in the security folder (/cpp_utils/tests/BLETests/security), but some of them don't work because of this error:

`CXX build/main/app_main_security.o
AR build/main/libmain.a
LD build/bleTemperature.elf
/home/magikarpxl/esp/bleTemperature/build/main/libmain.a(SampleServer_authentication_numeric_confirmation.o):(.literal._ZN10MySecurityD5Ev[MySecurity::~MySecurity()]+0x0): undefined reference to `vtable for BLESecurityCallbacks'
collect2: error: ld returned 1 exit status
/home/magikarpxl/esp/esp-idf/make/project.mk:388: recipe for target '/home/magikarpxl/esp/bleTemperature/build/bleTemperature.elf' failed
make: *** [/home/magikarpxl/esp/bleTemperature/build/bleTemperature.elf] Error 1`

Looks like undefined reference to `vtable for BLESecurityCallbacks' is the error in the log. Does anyone know a fix for this problem?

thomasvdwege, 2018-02-25:
It looks like the destructors are not implemented the right way, but I couldn't find out how to fix the error (I tried implementing the virtual destructors, but it didn't help). Looks like @chegewara implemented these examples according to the pull requests, could you look at it?

chegewara, 2018-02-25:
I really dont know why your code is complaining about destructor. BLESecurityCallbacks is just interface and there is nothing that needs more than virtual destructor. Maybe try `make clean`.

thomasvdwege, 2018-02-25:
`make clean` didn't work, it looks like the virtual destruction needs an implementation, like `~MySecurity(){}`. Can you reproduce the issue?

chegewara, 2018-02-25:
@thomasvdwege 
I have one more idea, can you see if SM module is turned on in menuconfig? Its security module in bluetooth settings.

thomasvdwege, 2018-02-25:
Include BLE Security Module (SMP) is turned on, did you mean that setting?

chegewara, 2018-02-25:
Yes, thats the one

thomasvdwege, 2018-02-25:
Does it need to be turned off? Having it on doesn't solve the problem.

chegewara, 2018-02-25:
No, it needs to be turned on. I will try to compile this example and let you know.

chegewara, 2018-02-25:
Thats strange, but i have recreated this issue and its been working before. I will look at it closer, thanks for reporting and for patience.

thomasvdwege, 2018-02-25:
Let me know if i need to test something, or if you need help. ```(SampleServer_authentication_numeric_confirmation.o):(.literal._ZN10MySecurityD5Ev[MySecurity::~MySecurity()]+0x0): undefined reference to ``vtable for BLESecurityCallbacks'``` probably means there's something wrong with the deconstructor, but I'm not 100% sure.

chegewara, 2018-02-25:
This issue has something to do with virtual class in C++. Im not expert, my knowledge about C++ is very small, but ive found the solution. You can make this destructor non-virtual and it should have works:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLESecurity.h#L40

thomasvdwege, 2018-02-25:
Making it non virtual looks like some kind of hack, I think it still needs to be virtual if we want to do something with the deconstructor (it's needed when you want to do something when you delete an object, for example deleting other objects associated with the object). If it works right now it's usable, I'll look into it another time. Thanks for the fix!

chegewara, 2018-02-25:
Thanks for using this code and reporting issue.

gourryinverse, 2018-06-05:
issue still persists, not sure what the problem was but for not i hacked around it by simply doing the same method of un-virtualizing the destructor

rkone, 2018-06-12:
I also ran into this issue and finally came upon this page after much time spent trying to find the solution on my own..

hetlelid, 2018-09-16:
I also get this, with the latest libs.

Workaround:
- Use the cpp_utils from kolban instead of the BLE lib in arduino-esp32

Step by step:
cd components
git clone https://github.com/nkolban/esp32-snippets.git
cp -r esp32-snippets/cpp_utils .
rm -rf esp32-snippets
rm -rf arduino-esp32/libraries/BLE/