nkolban/esp32-snippets #414: Save/remember BLE pairing in NVS

thedoginthewok, 2018-02-24:
I'm using code based on the following example:   
[BLE_server_passkey.ino](../../esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/security/BLE_server/BLE_server_passkey/BLE_server_passkey.ino)

I can successfully pair and read/write to a Characteristic with "setAccessPermissions(ESP_GATT_PERM_READ_ENCRYPTED | ESP_GATT_PERM_WRITE_ENCRYPTED)", but after removing power from the ESP32 I need to repeat the pairing process.

Is there a way to save the pairing information to non volatile storage?

chegewara, 2018-02-25:
Hi,
there is no need to save anything in NVS, it should be done by esp-idf. You just need to change this line https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/security/BLE_server/BLE_server_passkey/BLE_server_passkey.ino#L65

and use this in it:
ESP_LE_AUTH_REQ_SC_BOND
or this:
ESP_LE_AUTH_BOND 

Then your device will be bonded and stored and no more authentication will be required unless esp32 or device will delete bonding info.

BTW Im glad it works for you

thedoginthewok, 2018-02-26:
Thanks, it's working!

Kinda unrelated, but I had to implement method `bool onConfirmPIN(uint32_t pin)` in class MySecurity to get the example to compile.

closed 2018-02-26 by thedoginthewok

chegewara, 2018-02-26:
Example seems to be a little outdated, because `bool onConfirmPIN(uint32_t pin)` has been added later on, sorry for that.

robson31, 2018-07-09:
How do I get the example to compile? I've been looking into the library, but I can't find a function definition for `bool on ConfirmPIN(unsigned int)` that there's undefined reference to, only `bool onConfirmPIN (uint32_t pin)`.

chegewara, 2018-07-10:
You have to use callback with this value as parameter:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLESecurity.h#L67

Here is only example, but you have to implement it to return true or false, depend on input value:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/security/SampleClient_authentication_passkey.cpp#L39-L43
You can have for example 2 buttons where one is to return true and one to return false.

chegewara, 2018-07-10:
If you have somewhere in your code (BLESecurity or example) `onConfirmPIN(unsigned int)` then you need to change it to:
`onConfirmPIN(uint32_t pin)`
This must be veeery old code and sorry for that.

robson31, 2018-07-10:
No problem, I've checked the libraries and there's no instance of `onConfirmPIN(unsigned int)` anywhere, not even in my code, which is exactly why this compiler error doesn't make any sense to me.

chegewara, 2018-07-10:
The thing is i am not using `unsigned int` because i am trying to follow esp-idf convention.
Just try to add `onConfirmPIN(unsigned int)` in your callback and see what happen, and if you can post all logs with error you are getting.

robson31, 2018-07-10:
`//bool onConfirmPIN(uint32_t pass_key);
bool onConfirmPIN(usigned int);`
and the code compiles with no errors...

chegewara, 2018-07-10:
Good for you, now you have to remember this to fix when it will complain next time.