nkolban/esp32-snippets #420: BLERemoteCharacteristic::writeValue fails with '0x00'

pit001, 2018-03-01:
SdkVersion: v3.0-dev-1325-ge5b2c1cf
Arduino IDE 1.8.5 with BT sources from esp32-snippets/cpp_utils/BLE*.*
Example: ESP32_BLE_Arduino/examples/BLE_client/

Bug in the methode: `BLERemoteCharacteristic::writeValue(uint8_t* data, size_t length, bool response)` in BLERemoteCharacteristic.cpp

`data` is a bytearray with binary data. 

`writeValue` calls `writeValue(std::string newValue, bool response)` . Here `.length()` gets the length of the string up to the first \0 (0x00) -> if the binary data includes `0x00`'s the length is wong and only the first part of the bytearray is send...

Possible solution: Add a write**Binary** fnc

```
void BLERemoteCharacteristic::writeBinary(uint8_t* data,size_t length, bool response) {
	ESP_LOGD(LOG_TAG, ">> writeBinary(), length: %d", length());

	// Check to see that we are connected.
	if (!getRemoteService()->getClient()->isConnected()) {
		ESP_LOGE(LOG_TAG, "Disconnected");
		//////////throw BLEDisconnectedException();
	}

	m_semaphoreWriteCharEvt.take("writeValue");

	// Invoke the ESP-IDF API to perform the write.
	esp_err_t errRc = ::esp_ble_gattc_write_char(
		m_pRemoteService->getClient()->getGattcIf(),
		m_pRemoteService->getClient()->getConnId(),
		getHandle(),
		length,
		data,
		response?ESP_GATT_WRITE_TYPE_RSP:ESP_GATT_WRITE_TYPE_NO_RSP,
		ESP_GATT_AUTH_REQ_NONE
	);

	if (errRc != ESP_OK) {
		ESP_LOGE(LOG_TAG, "esp_ble_gattc_write_char: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
		return;
	}

	m_semaphoreWriteCharEvt.wait("writeBinary");

	ESP_LOGD(LOG_TAG, "<< writeBinary");
} // writeBinary
```

-Pit

RobsyRocket, 2018-03-02:
Integrating this change would be great, because this could explain and solve the problem I have in my project. I need to write the value 0x00000000 but it doesn't work as expected.

nkolban, 2018-03-10:
It is my understanding that std::string in C++ standard library doesn't use the NULL character string termination convention but instead a std::string is an array of bytes of a given explicit length.  Do you have a calling example where it is failing?

For example:

```
BLERemoteCharacteristic *pMyCharacteristic;
...
pMyCharacteristic->writeValue((char*)"AB\0DE", 5);
```
Should set the value to be the 5 bytes 'A', 'B', 0x00, 'D' and 'E'.

pit001, 2018-03-11:
I can follow your understanding, it's also my estimation, but the reality teachs me something else.

char ON[] =  {0x42, 0x03, 0x4E, 0x04, 0xB2, 0x10, 0x81}; //working
char OFF[] = {0x40, 0x03, 0xB2, 0x10, 0x81, 0x00, 0x00}; //fail - get only 0x40, 0x03, 0xB2, 0x10, 0x81

Flojolo, 2021-01-19:
Hi Guys, 
is there an update on this topic? I think, I still run into this error. 
When I try to send [0x45, 0x01] to my device, it reacts as I expect. In contrast to this, when I try to send [0x45, 0x00] it does not react at all. 

I can confirm, that the device is able to process [0x45, 0x00] when I send it with RFConnect.

This is my code snippet which im working on:
```
uint8_t newValue[2] = {0x45, 0x00};
  pCommandRemoteCharacteristic->writeValue(newValue, 2, false);
```

Flojolo, 2021-01-20:
Hello again,

I solved this issue on my end, but it totally confused me. The issue was:
- passing a char array to `writeValue` and a length only, results in the execution of `void BLERemoteCharacteristic::writeValue(std::string newValue, bool response)`. The key in my case was, that the value `2` is interpreted as `true` for the argument `response`. 
- Using this insight, I understood (in contrast to my expectation) that I need the parameter `response` to be set to `true`. Applying this and calling `void BLERemoteCharacteristic::writeValue(uint8_t *data, size_t length, bool response)` explicitly with 3 parameters solved my issue.

chegewara, 2021-01-20:
Is there any reason you cant explicitly make std::string before passing it? Like for example here you can send `0x0`:
```
char tmp[] = {0x0}; 
std::string value(tmp);
```
Im not 100% sure, but i believe this should works, because value is type of string and you can get its length (is not null terminated).

positron96, 2021-04-08:
> s there any reason you cant explicitly make std::string before passing it? Like for example here you can send `0x0`:
> 
> ```
> char tmp[] = {0x0}; 
> std::string value(tmp);
> ```

It does not work. The `length` of such string is 0, not 1. See small snippet here: https://ideone.com/5qIxPy

chegewara, 2021-04-08:
Maybe this:
```
std::string value = "\0x0";
std::string value("\0x0");
std::string value{"\0x0"};
```

One of those should works.

positron96, 2021-04-08:
Sorry, my understanding of C++ is limited, but as I understand it, these 3 lines produce identical strings. Their `std::string::length()` will just stop at first encountered zero. 
Here is the checking code: https://ideone.com/yQ1M2r (all 3 have zero length)

chegewara, 2021-04-08:
No worries, it was just my wrong assumption it should works. This time i tested code and this works for sure:
```
#include <iostream>
void setup() {
  // put your setup code here, to run once:
Serial.begin(115200);
delay(1000);
  std::string value1 = "\0x0";
  std::string value2("\0x0");
  std::string value3{"\0x0"};
  value1.resize(1);
  value2.resize(1);
  value3.resize(1);
 
  Serial.println(value1.length());
  Serial.println(value2.length());
  Serial.println(value3.length());

  Serial.printf("%d\n", value1[0]);
  Serial.printf("%d\n", value2[0]);
  Serial.printf("%d\n", value3[0]);

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

positron96, 2021-04-08:
Yeah, thanks, this one works! Looks a bit hacky to me though.

chegewara, 2021-04-08:
I am guessing its the case when you are using conversion from char string to std::string, where the first one is null terminated and later is not.