nkolban/esp32-snippets #423: BLERemoteDescriptor::readValue() doesn't work, and here is why

stefaandesmet2003, 2018-03-03:
The BLERemoteDescriptor::readValue() function doesn't work, i.e. it never returns, because it waits forever on the semaphore m_semaphoreReadDescrEvt.
I use this function for reading the descriptor 2902 (checking if notifications/indications are on on a BLE server characteristic).

I solved the issue by adding a gattClientEventHandler in the BLERemoteDescriptor class that handles the ESP_GATTC_READ_DESCR_EVT event. The BLERemoteCharacteristic class also needs a small change in its gattClientEventHandler : passing on the received event to the underlying BLERemoteDescriptor's from the m_descriptorMap
The code is in attached zipfile. I hope you find it useful to add to your library.
[BLERemoteDescriptor.zip](https://github.com/nkolban/esp32-snippets/files/1776993/BLERemoteDescriptor.zip)

vtongcoder, 2018-03-09:
I still have the error when read Descriptor after copying your files.
It cause the connection auto disconnecting.

(16683) BLERemoteCharacteristic: << readValue(): length: 20
D (16689) BLERemoteDescriptor: >> readValue: handle: 46, uuid: 00002901-0000-1000-8000-00805f9b34fb
D (16698) FreeRTOS: Semaphore taking: name: ReadDescrEvt (0x3ffdbec8), owner: <N/A> for readValue
D (16707) FreeRTOS: Semaphore taken:  name: ReadDescrEvt (0x3ffdbec8), owner: readValue
E (46716) BT: gatt_rsp_timeout disconnecting...
E (46718) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0016
D (46720) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_READ_DESCR_EVT
D (46727) BLEUtils: GATT Event: ESP_GATTC_READ_DESCR_EVT
D (46733) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_DISCONNECT_EVT
D (46741) BLEUtils: GATT Event: ESP_GATTC_DISCONNECT_EVT

stefaandesmet2003, 2018-03-09:
You are trying to read descriptor 2901; is that supposed to be readable? https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.characteristic_user_description.xml 
Maybe you're trying an invalid operation, and that's why your device doesn't reply, and you get the timeout.
My BLE device doesn't have descriptor 2901 on any characteristic, so I can't repeat your case. My device has a descriptor 2902 on a few characteristics, and descriptor 2902 is read/write by specification, and it works ok.

A bit off topic, but how did you get the timing values in the debug output? 

This is part of the output from a test sketch (with core debug level set to 'verbose') , with the ESP_GATTC_READ_DESCR_EVT event arriving correctly.

Characteristic: uuid: 00002a5b-0000-1000-8000-00805f9b34fb, handle: 16 0x10, props: broadcast: 0, read: 0, write_nr: 0, write: 0, notify: 1, indicate: 0, auth: 0
[D][BLERemoteDescriptor.cpp:115] readValue(): >> readValue: handle: 17, uuid: 00002902-0000-1000-8000-00805f9b34fb
[D][BLEDevice.cpp:140] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_READ_DESCR_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_READ_DESCR_EVT
[D][BLERemoteDescriptor.cpp:141] readValue(): << readValue(): length: 2

stefaandesmet2003, 2018-03-09:
E (46718) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0016
reason 0x0016 means GATT_CONN_TERMINATE_LOCAL_HOST
timeout on a connection would be reason 0x0008 = GATT_CONN_TIMEOUT. 
So not sure what's going on in your setup.