nkolban/esp32-snippets #425: Calling get on an NVS variable before it's set causes a crash

damonsmith, 2018-03-03:
I made function to get status and it tries to retrieve a couple of strings from NVS but if they haven't been set it causes a reboot.

If they've been set then it's all good, but if they haven't been set yet and I run this bit of code:

NVS myNamespace(WIFI_STORAGE_NAMESPACE, NVS_READONLY);
myNamespace.get(KEY_SETTINGS_SSID, &ssid, false);
myNamespace.get(KEY_SETTINGS_PASSWORD, &password, false);

then I get:

abort() was called at PC 0x400fc6f3 on core 1

And the esp reboots. There doesn't seem to be a "check" function to see if the namespace has been used or the variables have been set. Is there a workaround for this? 

Sorry, once I get my head around the esp-idf API a bit more I'll probably be able to fix it myself and send a patch but for now I'm a bit stuck. Any help appreciated!

nkolban, 2018-03-10:
Looks like you implemented an elegant resolution.  Thank you kind sir.  Is this issue now ready for closure?

damonsmith, 2018-03-11:
Thanks! Im really happy I can contribute.

closed 2018-03-11 by damonsmith