nkolban/esp32-snippets #430: ANCS library

S-March, 2018-03-09:
Hi Neil,

I decided to start putting together an ANCS library to use the ESP32 as a smart watch. Right now I'm a little stuck and was wondering if I could pick your brain. 

Normally, for ANCS to work, the device must first advertise while soliciting service from the ANCS UUID.

I added the following function to BLEAdvertising so that you can include a service solicitation in the advert packet (this allows the device to show up in the settings panel of an iOS device).

void setServiceSolicitation(BLEUUID uuid);

```
/**
* @brief Set the service solicitation (UUID)
* @param [in] uuid The UUID to set with the service solicitation data.  Size of UUID will be used.
*/
void BLEAdvertisementData::setServiceSolicitation(BLEUUID uuid)
{
	char cdata[2];
	switch(uuid.bitSize()) {
		case 16: {
			// [Len] [0x14] [UUID16] data
			cdata[0] = 3;
			cdata[1] = ESP_BLE_AD_TYPE_SOL_SRV_UUID;  // 0x14
			addData(std::string(cdata, 2) + std::string((char *)&uuid.getNative()->uuid.uuid16,2));
			break;
		}

		case 128: {
			// [Len] [0x15] [UUID128] data
			cdata[0] = 17;
			cdata[1] = ESP_BLE_AD_TYPE_128SOL_SRV_UUID;  // 0x15
			addData(std::string(cdata, 2) + std::string((char *)uuid.getNative()->uuid.uuid128,16));
			break;
		}

		default:
			return;
	}
} // setServiceSolicitationData
```
Once you find the device, you can connect in the settings panel. I added the ability to bond in my code, which it does just fine. After bonded, the device shows as connected - perfect.

Here is where I'm stuck. After the device is bonded and connected, the ESP needs to switch out of a server role and into a client role to then subscribe to the ANCS service on the iOS device and read/write to the three characteristics exposed. 

I may just be in way over my head, but I can't seem to find a way to force a connection to a white-listed device. Or if I'm already connected, to maintain that connection while I dismantle the server and spin up the client. Any ideas?

Here is the code that I have so far (gets as far as bonding/connecting to device after the advertising stage)

```
/**
 * Create a new BLE server.
 */
#include "BLEDevice.h"
#include "BLEServer.h"
#include "BLEUtils.h"
#include "BLE2902.h"
#include <esp_log.h>
#include <string>
#include <Task.h>

#include "sdkconfig.h"

static char LOG_TAG[] = "SampleServer";

class MySecurity : public BLESecurityCallbacks {

    uint32_t onPassKeyRequest(){
        ESP_LOGI(LOG_TAG, "PassKeyRequest");
        return 123456;
    }

    void onPassKeyNotify(uint32_t pass_key){
        ESP_LOGI(LOG_TAG, "On passkey Notify number:%d", pass_key);
    }

    bool onSecurityRequest(){
        ESP_LOGI(LOG_TAG, "On Security Request");
        return true;
    }
    
    bool onConfirmPIN(unsigned int){
        ESP_LOGI(LOG_TAG, "On Confrimed Pin Request");
        return true;
    }

    void onAuthenticationComplete(esp_ble_auth_cmpl_t cmpl){
        ESP_LOGI(LOG_TAG, "Starting BLE work!");
        if(cmpl.success){
            uint16_t length;
            esp_ble_gap_get_whitelist_size(&length);
            ESP_LOGD(LOG_TAG, "size: %d", length);
        }
    }
};

class MainBLEServer: public Task {
    void run(void *data) {
        ESP_LOGD(LOG_TAG, "Starting BLE work!");
        esp_log_buffer_char(LOG_TAG, LOG_TAG, sizeof(LOG_TAG));
        esp_log_buffer_hex(LOG_TAG, LOG_TAG, sizeof(LOG_TAG));

        // Initialize device
        BLEDevice::init("Watch");
        BLEServer* pServer = BLEDevice::createServer();
        BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);
        BLEDevice::setSecurityCallbacks(new MySecurity());

        // Advertising parameters:
        // Soliciting ANCS
        BLEAdvertising *pAdvertising = pServer->getAdvertising();
        BLEAdvertisementData oAdvertisementData = BLEAdvertisementData();
        oAdvertisementData.setFlags(0x01);
        oAdvertisementData.setServiceSolicitation(BLEUUID("7905F431-B5CE-4E99-A40F-4B1E122D00D0"));
        pAdvertising->setAdvertisementData(oAdvertisementData);        

        // Set security
        BLESecurity *pSecurity = new BLESecurity();
        pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_BOND);
        pSecurity->setCapability(ESP_IO_CAP_OUT);
        pSecurity->setInitEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);

        //Start advertising
        pAdvertising->start();

        ESP_LOGD(LOG_TAG, "Advertising started!");
        delay(portMAX_DELAY);
    }
};

void SampleSecureServer(void)
{
    //esp_log_level_set("*", ESP_LOG_DEBUG);
    MainBLEServer* pMainBleServer = new MainBLEServer();
    pMainBleServer->setStackSize(20000);
    pMainBleServer->start();
} // app_main

void setup()
{
    SampleSecureServer();
    
}
void loop()
{
    delay(100);
}
```

Thanks a ton!

chegewara, 2018-03-10:
Hi,
i have question. Because bluetooth LE security is part of bluetooth v4.2, can you confirm that your iOS device is at least v4.2? If not, can you try to rewrite your code to not use security, or use only encrypted connection?

S-March, 2018-03-10:
I’m working with an iPhone 8, so it should b able to handle up to 5.0. I think the 6 was the first iOS device with 4.2. I believe the security is a requirement of the ANCS protocol.

I was thinking about it a bit more during my commute home. Is there a getWhitelistedDeviceAddress call? Or a way to just force the BLEClient to a whitelisted device?

chegewara, 2018-03-10:
Can you provide some logs?

S-March, 2018-03-10:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
[D][ANCS_ADV_CLIENT_TEST.ino:57] run(): Starting BLE work!
[D][BLEDevice.cpp:69] createServer(): >> createServer
[D][BLEServer.cpp:304] registerApp(): >> registerApp - 0
[D][BLEDevice.cpp:95] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_REG_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_REG_EVT
[D][BLEUtils.cpp:1791] dumpGattServerEvent(): [status: ESP_GATT_OK, app_id: 0]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_REG_EVT
[D][BLEServer.cpp:308] registerApp(): << registerApp
[D][BLEDevice.cpp:76] createServer(): << createServer
[D][[LESBLEAr.ert:29ng.cpnd14GATsetrvvervenem): <ataand >GATetArvertvsnt
tData
[D][BLEA[verBLEinglsppp1410 se duveGapsemnt(D:taec: <ed atGAdvevensemenP_Gta
L[_AD[BLATAvertiSingCOMp:LET4]EVT
[()[BL> silst: cp:1omA] Datp: 1Eveustom [snReus: 0eDa[D] 0L
[DrveLEApp:13isihancpp:19Ev st(): ): Seno r .v. ha ading ise event[BLEAdvertising.cpp:231[ staLt()ils<cst:10
9[D]dAmCGaADVenLIENT_eceT.idoa 4APruv(n: AESertAsiBL _taCAed!
_DATA_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1285] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
[D][BLEUtils.cpp:1123] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEDevice.cpp:95] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CONNECT_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CONNECT_EVT
[D][BLEUtils.cpp:1707] dumpGattServerEvent(): [conn_id: 0, remote_bda: 75:cf:fe:dc:40:fa]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONNECT_EVT
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[0;31mE (58047) BT: lmp_version_below LMP version 6 < 8[0m
[0;31mE (58316) BT: Call back not found for application conn_id=3[0m
[D][BLEDevice.cpp:95] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_MTU_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_MTU_EVT
[D][BLEUtils.cpp:1766] dumpGattServerEvent(): [conn_id: 0, mtu: 517]
[I][BLEDevice.cpp:110] gattServerEventHandler(): ESP_GATTS_MTU_EVT, MTU 517
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT
[D][BLEUtils.cpp:1343] dumpGapEvent(): [status: 16, bd_addr: 75:cf:fe:dc:40:fa, min_int: 6, max_int: 6, latency: 0, conn_int: 24, timeout: 72]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[0;31mE (59996) BT: FOR LE SC LTK IS USED INSTEAD OF STK[0m
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_KEY_EVT
[D][BLEUtils.cpp:1358] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[I][BLEDevice.cpp:231] gapEventHandler(): key type = ESP_LE_KEY_LENC
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_KEY_EVT
[D][BLEUtils.cpp:1358] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[I][BLEDevice.cpp:231] gapEventHandler(): key type = ESP_LE_KEY_PENC
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_KEY_EVT
[D][BLEUtils.cpp:1358] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[I][BLEDevice.cpp:231] gapEventHandler(): key type = ESP_LE_KEY_LID
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_KEY_EVT
[D][BLEUtils.cpp:1358] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[I][BLEDevice.cpp:231] gapEventHandler(): key type = ESP_LE_KEY_PID
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_AUTH_CMPL_EVT
[D][BLEUtils.cpp:1160] dumpGapEvent(): [bd_addr: 75:cf:fe:dc:40:fa, key_present: 0, key: ***, key_type: 0, success: 1, fail_reason: 0, addr_type: ***, dev_type: ESP_BT_DEVICE_TYPE_BLE]
[I][ANCS_ADV_CLIENT_TEST.ino:46] onAuthenticationComplete(): Starting BLE work!
[D][ANCS_ADV_CLIENT_TEST.ino:50] onAuthenticationComplete(): size: 12
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT
[D][BLEUtils.cpp:1343] dumpGapEvent(): [status: 0, bd_addr: 75:cf:fe:dc:40:fa, min_int: 24, max_int: 24, latency: 0, conn_int: 36, timeout: 72]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
```

chegewara, 2018-03-10:
2 questions:
- this message is saying that you are connecting with device capable only bluetooth v 4.0 `BT: lmp_version_below LMP version 6 < 8`, do you have this message each time when you are connecting this device to esp32 or just sometimes?

- i see your device is exchanging authentication keys, which means your device is not bonded with esp32, esp32 does not store this device as bonded or device does not store esp32 as bonded; if your device is bonded with esp32 you should not see those lines when you connect again:

```
FOR LE SC LTK IS USED INSTEAD OF STK�[0m
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_KEY_EVT
[D][BLEUtils.cpp:1358] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[I][BLEDevice.cpp:231] gapEventHandler(): key type = ESP_LE_KEY_LENC
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_KEY_EVT
[D][BLEUtils.cpp:1358] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[I][BLEDevice.cpp:231] gapEventHandler(): key type = ESP_LE_KEY_PENC
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_KEY_EVT
[D][BLEUtils.cpp:1358] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[I][BLEDevice.cpp:231] gapEventHandler(): key type = ESP_LE_KEY_LID
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_KEY_EVT
[D][BLEUtils.cpp:1358] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[I][BLEDevice.cpp:231] gapEventHandler(): key type = ESP_LE_KEY_PID
```

S-March, 2018-03-10:
Sorry, I should have mentioned: that last log I purposefully forced phone to forget the ESP before rebooting. I wanted to show the whole pairing process. Below is another log. In it, I did the following:

1.  Forced phone to forget device
2. Booted ESP
3. Opened settings panel on phone
4. Found device and hit connect
5. When pairing pop up showed up, I hit pair
6. Turned off BT on phone to force a disconnect
7. Turned BT back on and reconnected to ESP

The second time I connected, I didn't exchange keys or have to acknowledge the pairing. Only the first time. As for why the phone is only showing as 4.0 I'm not sure...

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
[D][ANCS_ADV_CLIENT_TEST.ino:57] run(): Starting BLE work!
[D][BLEDevice.cpp:69] createServer(): >> createServer
[D][BLEServer.cpp:304] registerApp(): >> registerApp - 0
[D][BLEDevice.cpp:95] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_REG_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_REG_EVT
[D][BLEUtils.cpp:1791] dumpGattServerEvent(): [status: ESP_GATT_OK, app_id: 0]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_REG_EVT
[D][BLEServer.cpp:308] registerApp(): << registerApp
[D][BLEDevice.cpp:76] createServer(): << createServer
[D][[D]SerEAdvcptis94g cppdle0] TStrverrtint()nt<< a()dleGAsTSedverEisetentData
[D][BLEAdvert[BLng.ils:1pp: se1] verpGsemvenD(ta R: eivsetaAGAertvente tDa_GAP_BLED]DV_EAdA_rAW_ing_COpMPLET] sVTrt(D][>> Utart: cpsto111]Daump 1,Evest():cantatpon 0D
t[: 0B
ESe[BLr.cver134ingandl:G94]venart()BLE nover vic haadvintiGAd event!B
EAdvertising.cpp:231] sta][B): <ils.artp:[D][]NCSmpGa_ELInNT): ST.enoe84a rAn ):endveEtisiAg PtarE_SCAN_RSP_DATA_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1285] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
[D][BLEUtils.cpp:1123] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEDevice.cpp:95] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CONNECT_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CONNECT_EVT
[D][BLEUtils.cpp:1707] dumpGattServerEvent(): [conn_id: 0, remote_bda: AA:AA:AA:AA:AA:AA]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONNECT_EVT
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[0;31mE (9791) BT: lmp_version_below LMP version 6 < 8[0m
[0;31mE (10179) BT: Call back not found for application conn_id=3[0m
[D][BLEDevice.cpp:95] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_MTU_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_MTU_EVT
[D][BLEUtils.cpp:1766] dumpGattServerEvent(): [conn_id: 0, mtu: 517]
[I][BLEDevice.cpp:110] gattServerEventHandler(): ESP_GATTS_MTU_EVT, MTU 517
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT
[D][BLEUtils.cpp:1343] dumpGapEvent(): [status: 16, bd_addr: AA:AA:AA:AA:AA:AA, min_int: 6, max_int: 6, latency: 0, conn_int: 24, timeout: 72]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[0;31mE (12699) BT: FOR LE SC LTK IS USED INSTEAD OF STK[0m
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_KEY_EVT
[D][BLEUtils.cpp:1358] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[I][BLEDevice.cpp:231] gapEventHandler(): key type = ESP_LE_KEY_LENC
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_KEY_EVT
[D][BLEUtils.cpp:1358] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[I][BLEDevice.cpp:231] gapEventHandler(): key type = ESP_LE_KEY_PENC
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_KEY_EVT
[D][BLEUtils.cpp:1358] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[... 32 lines cut ...]
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CONNECT_EVT
[D][BLEUtils.cpp:1707] dumpGattServerEvent(): [conn_id: 0, remote_bda: AA:AA:AA:AA:AA:AA]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONNECT_EVT
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[0;31mE (25958) BT: lmp_version_below LMP version 6 < 8[0m
[0;31mE (26137) BT: l2cble_start_conn_update, the last connection update command still pending.[0m
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_AUTH_CMPL_EVT
[D][BLEUtils.cpp:1160] dumpGapEvent(): [bd_addr: AA:AA:AA:AA:AA:AA, key_present: 0, key: ***, key_type: 0, success: 1, fail_reason: 0, addr_type: ***, dev_type: ESP_BT_DEVICE_TYPE_BLE]
[I][ANCS_ADV_CLIENT_TEST.ino:46] onAuthenticationComplete(): Starting BLE work!
[... cut ...]
[I][BLEDevice.cpp:110] gattServerEventHandler(): ESP_GATTS_MTU_EVT, MTU 517
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT
[D][BLEUtils.cpp:1343] dumpGapEvent(): [status: 0, bd_addr: AA:AA:AA:AA:AA:AA, min_int: 24, max_int: 24, latency: 0, conn_int: 36, timeout: 72]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
```

chegewara, 2018-03-10:
No problem. Looks like device is connected with esp32, which means bonding works good.
Now you can try in server callback in onConnect try to create BLEClient and connect to iOS device.

S-March, 2018-03-10:
Oh, interesting. I will try that in the morning! Thanks, will report back

S-March, 2018-03-10:
Hey @chegewara or @nkolban , any chance you can point me in the right direction to find the remote device address that connected to a server? I think I know where I need to go next, but I can't seem to find how to spin up a client connected to an address I'm already connected to.

Basically, what I'm looking for is something like:
"advertisedDevice.getAddress().toString().c_str()"

So I can then take that and pass that to the client for connection.

Thanks!

chegewara, 2018-03-10:
If it can be printed out here, then it can be retrieved somewhere:
[D][BLEUtils.cpp:1707] dumpGattServerEvent(): [conn_id: 0, remote_bda: AA:AA:AA:AA:AA:AA]

S-March, 2018-03-10:
That was my exact thought. But as I was digging through the dumpGattServerEvent(), I couldn't seem to find where the param was sent to it from.

S-March, 2018-03-10:
I see this line, where the connect.remote_bda is exactly what I want. I have no idea where param is being passed to it though.
esp_ble_set_encryption(param->connect.remote_bda, BLEDevice::m_securityLevel);

chegewara, 2018-03-10:
Its passed from low level bluedroid stack as events parameter. Now, as you've found where it can be retrieved from, you can add variable to this class and you will be able to get it when you need it. Its not much work, 2 to 5 lines of code.

S-March, 2018-03-10:
I'm sorry if this is a bit of a silly question, I'm just rather new to this. Would I just be adding it to the BLEDevice class?

Something like:
```
esp_bd_addr_t m_remoteBda;
```
And then in the BLEDevice::gattServerEventHandler in the ESP_GATTS_CONNECT_EVT case, add something like 
```
BLEDevice::m_remoteBda=param->connect.remote_bda
```
Then in the future I can just call BLEDevice::m_remoteBda to put the address where I need it?

chegewara, 2018-03-11:
Check if this `esp_bd_addr_t m_remoteBda;` is private or public. Needs to be public.

S-March, 2018-03-11:
In BLEDevice.h
```
/**
 * @brief %BLE functions.
 */
class BLEDevice {
public:

	static BLEClient*  createClient();    // Create a new BLE client.
	static BLEServer*  createServer();    // Cretae a new BLE server.
	static BLEAddress  getAddress();      // Retrieve our own local BD address.
	static BLEScan*    getScan();         // Get the scan object
	static std::string getValue(BLEAddress bdAddress, BLEUUID serviceUUID, BLEUUID characteristicUUID);	  // Get the value of a characteristic of a service on a server.
	static void        init(std::string deviceName);   // Initialize the local BLE environment.
	static void        setPower(esp_power_level_t powerLevel);  // Set our power level.
	static void        setValue(BLEAddress bdAddress, BLEUUID serviceUUID, BLEUUID characteristicUUID, std::string value);   // Set the value of a characteristic on a service on a server.
	static std::string toString();        // Return a string representation of our device.
	static void        whiteListAdd(BLEAddress address);    // Add an entry to the BLE white list.
	static void        whiteListRemove(BLEAddress address); // Remove an entry from the BLE white list.
	static void		   setEncryptionLevel(esp_ble_sec_act_t level);
	static void		   setSecurityCallbacks(BLESecurityCallbacks* pCallbacks);
	static esp_err_t   setMTU(uint16_t mtu);
	static uint16_t	   getMTU();
        static esp_bd_addr_t m_remoteBda;
```

Then in BLEDevice.cpp
```
/**
 * Singletons for the BLEDevice.
 */
BLEServer* BLEDevice::m_pServer = nullptr;
BLEScan*   BLEDevice::m_pScan   = nullptr;
BLEClient* BLEDevice::m_pClient = nullptr;
bool       initialized          = false;   // Have we been initialized?
esp_ble_sec_act_t 	BLEDevice::m_securityLevel = (esp_ble_sec_act_t)0;
BLESecurityCallbacks* BLEDevice::m_securityCallbacks = nullptr;
uint16_t   BLEDevice::m_localMTU = 23;
esp_bd_addr_t   BLEDevice::m_remoteBda;
```
And 

```
/**
 * @brief Handle GATT server events.
 *
 * @param [in] event The event that has been newly received.
 * @param [in] gatts_if The connection to the GATT interface.
 * @param [in] param Parameters for the event.
 */
/* STATIC */ void BLEDevice::gattServerEventHandler(
   esp_gatts_cb_event_t      event,
   esp_gatt_if_t             gatts_if,
   esp_ble_gatts_cb_param_t* param
) {
	ESP_LOGD(LOG_TAG, "gattServerEventHandler [esp_gatt_if: %d] ... %s",
		gatts_if,
		BLEUtils::gattServerEventTypeToString(event).c_str());

	BLEUtils::dumpGattServerEvent(event, gatts_if, param);

	switch(event) {
		case ESP_GATTS_CONNECT_EVT: {
			BLEDevice::m_localMTU = 23;
			if(BLEDevice::m_securityLevel){
                BLEDevice::m_remoteBda=param->connect.remote_bda;
				esp_ble_set_encryption(param->connect.remote_bda, BLEDevice::m_securityLevel);
			}
			break;
		} // ESP_GATTS_CONNECT_EVT
```
And I'm still getting the error: 
>error: 'remoteBda' is not a member of 'BLEDevice'
>Serial.println(BLEDevice::remoteBda.toString().c_str());

chegewara, 2018-03-11:
Try to initialize it here:
```
/**
 * Singletons for the BLEDevice.
 */
BLEServer* BLEDevice::m_pServer = nullptr;
BLEScan*   BLEDevice::m_pScan   = nullptr;
BLEClient* BLEDevice::m_pClient = nullptr;
bool       initialized          = false;   // Have we been initialized?
esp_ble_sec_act_t 	BLEDevice::m_securityLevel = (esp_ble_sec_act_t)0;
BLESecurityCallbacks* BLEDevice::m_securityCallbacks = nullptr;
uint16_t   BLEDevice::m_localMTU = 23;
esp_bd_addr_t   BLEDevice::m_remoteBda = {00,00,00,00,00,00}; <-- HERE, but im not sure
```

S-March, 2018-03-11:
New and exciting error!

>error: invalid array assignment
>BLEDevice::m_remoteBda=param->connect.remote_bda;

chegewara, 2018-03-11:
This is definition, you need to initialize it:
http://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/esp_bt_defs.html#_CPPv213esp_bd_addr_t

Im not good at C and C++, sorry.

S-March, 2018-03-11:
Got it!
```
/**
 * Singletons for the BLEDevice.
 */
BLEServer* BLEDevice::m_pServer = nullptr;
BLEScan*   BLEDevice::m_pScan   = nullptr;
BLEClient* BLEDevice::m_pClient = nullptr;
bool       initialized          = false;   // Have we been initialized?
esp_ble_sec_act_t 	BLEDevice::m_securityLevel = (esp_ble_sec_act_t)0;
BLESecurityCallbacks* BLEDevice::m_securityCallbacks = nullptr;
uint16_t   BLEDevice::m_localMTU = 23;
esp_bd_addr_t   BLEDevice::m_remoteBda;
```
And here is the important part! Have to use memcpy to copy the array.
```
/**
 * @brief Handle GATT server events.
 *
 * @param [in] event The event that has been newly received.
 * @param [in] gatts_if The connection to the GATT interface.
 * @param [in] param Parameters for the event.
 */
/* STATIC */ void BLEDevice::gattServerEventHandler(
   esp_gatts_cb_event_t      event,
   esp_gatt_if_t             gatts_if,
   esp_ble_gatts_cb_param_t* param
) {
	ESP_LOGD(LOG_TAG, "gattServerEventHandler [esp_gatt_if: %d] ... %s",
		gatts_if,
		BLEUtils::gattServerEventTypeToString(event).c_str());

	BLEUtils::dumpGattServerEvent(event, gatts_if, param);

	switch(event) {
		case ESP_GATTS_CONNECT_EVT: {
			BLEDevice::m_localMTU = 23;
			if(BLEDevice::m_securityLevel){
                                memcpy(m_remoteBda, (param->connect.remote_bda), sizeof(param->connect.remote_bda));
				esp_ble_set_encryption(param->connect.remote_bda, BLEDevice::m_securityLevel);
			}
			break;
		} // ESP_GATTS_CONNECT_EVT
```

S-March, 2018-03-11:
Back to work on the rest of the library, will continue to update as I make progress.

S-March, 2018-03-11:
[image]

Wooo! Client is connecting and found the characteristics on phone!

Next step is to ID those notifications and make use of them!

S-March, 2018-03-12:
If anyone is looking for an ANCS library in the future, here is a fully functional ANCS example.
 
I apologize the code is not the cleanest, but it works! I get all notifications from my phone streaming onto the UART lines. I will clean things up later and create a pull request.

```
/**
 * Create a new BLE server.
 */
#include "BLEDevice.h"
#include "BLEServer.h"
#include "BLEClient.h"
#include "BLEUtils.h"
#include "BLE2902.h"
#include <esp_log.h>
#include <string>
#include <Task.h>

#include "sdkconfig.h"

static char LOG_TAG[] = "SampleServer";

static BLEUUID ancsServiceUUID("7905F431-B5CE-4E99-A40F-4B1E122D00D0");
static BLEUUID notificationSourceCharacteristicUUID("9FBF120D-6301-42D9-8C58-25E699A21DBD");
static BLEUUID controlPointCharacteristicUUID("69D1D8F3-45E1-49A8-9821-9BBDFDAAD9D9");
static BLEUUID dataSourceCharacteristicUUID("22EAC6E9-24D6-4BB5-BE44-B36ACE7C7BFB");

class MySecurity : public BLESecurityCallbacks {

    uint32_t onPassKeyRequest(){
        ESP_LOGI(LOG_TAG, "PassKeyRequest");
        return 123456;
    }

    void onPassKeyNotify(uint32_t pass_key){
        ESP_LOGI(LOG_TAG, "On passkey Notify number:%d", pass_key);
    }

    bool onSecurityRequest(){
        ESP_LOGI(LOG_TAG, "On Security Request");
        return true;
    }
    
    bool onConfirmPIN(unsigned int){
        ESP_LOGI(LOG_TAG, "On Confrimed Pin Request");
        return true;
    }

    void onAuthenticationComplete(esp_ble_auth_cmpl_t cmpl){
        ESP_LOGI(LOG_TAG, "Starting BLE work!");
        if(cmpl.success){
            uint16_t length;
            esp_ble_gap_get_whitelist_size(&length);
            ESP_LOGD(LOG_TAG, "size: %d", length);
        }
    }
};

static void dataSourceNotifyCallback(
  BLERemoteCharacteristic* pDataSourceCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pDataSourceCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
}

static void NotificationSourceNotifyCallback(
  BLERemoteCharacteristic* pNotificationSourceCharacteristic,
[... 161 lines cut ...]

        ESP_LOGD(LOG_TAG, "Advertising started!");
        delay(portMAX_DELAY);
    }
};

void SampleSecureServer(void)
{
    //esp_log_level_set("*", ESP_LOG_DEBUG);
    MainBLEServer* pMainBleServer = new MainBLEServer();
    pMainBleServer->setStackSize(20000);
    pMainBleServer->start();
} // app_main

void setup()
{
    Serial.begin(115200);
    SampleSecureServer();
    
}
void loop()
{
    delay(100);
}
```

closed 2018-03-12 by S-March

chegewara, 2018-03-12:
Nice work. You can create new project on github with this project, its your project, and then you can share with community on forum.

S-March, 2018-03-12:
I plan on it. I have a few more goodies up my sleeve that will come along with it. For now, I figured I'd leave this here in case anyone else needs it while I get the rest of the project ready :)

chegewara, 2018-03-27:
@S-March What IOS do you have on iPhone 8?

S-March, 2018-03-27:
I’m running on 11.2.6

krewtib, 2018-05-16:
@S-March 
I am using Arduino 1.8.5 and tried to compile your ANCS example with the result that i get such errors:

ESP32_ANCS:129: error: expected class-name before '{' token
 class MyClient: public Task {
                             ^
/Users/bitwerk/Documents/Arduino ESP32/ESP32_ANCS/ESP32_ANCS.ino: In member function 'virtual void MyServerCallbacks::onConnect(BLEServer*)':
ESP32_ANCS:185: error: 'm_remoteBda' is not a member of 'BLEDevice'
         Serial.println(BLEAddress(BLEDevice::m_remoteBda).toString().c_str());
                                   ^
ESP32_ANCS:188: error: 'class MyClient' has no member named 'setStackSize'
         pMyClient->setStackSize(18000);
                    ^
ESP32_ANCS:189: error: 'class MyClient' has no member named 'start'
         pMyClient->start(new BLEAddress(BLEDevice::m_remoteBda));
                    ^
ESP32_ANCS:189: error: 'm_remoteBda' is not a member of 'BLEDevice'
         pMyClient->start(new BLEAddress(BLEDevice::m_remoteBda));
                                         ^
/Users/bitwerk/Documents/Arduino ESP32/ESP32_ANCS/ESP32_ANCS.ino: At global scope:
ESP32_ANCS:199: error: expected class-name before '{' token
 class MainBLEServer: public Task {
                                  ^
/Users/bitwerk/Documents/Arduino ESP32/ESP32_ANCS/ESP32_ANCS.ino: In member function 'void MainBLEServer::run(void*)':
ESP32_ANCS:217: error: 'class BLEAdvertisementData' has no member named 'setServiceSolicitation'
         oAdvertisementData.setServiceSolicitation(BLEUUID("7905F431-B5CE-4E99-A40F-4B1E122D00D0"));
                            ^
/Users/bitwerk/Documents/Arduino ESP32/ESP32_ANCS/ESP32_ANCS.ino: In function 'void SampleSecureServer()':
ESP32_ANCS:239: error: 'class MainBLEServer' has no member named 'setStackSize'
     pMainBleServer->setStackSize(20000);
                     ^
ESP32_ANCS:240: error: 'class MainBLEServer' has no member named 'start'
     pMainBleServer->start();
                     ^

What am i doing wrong?

S-March, 2018-05-16:
Make sure you read through the previous comments. I had to add m_remoteBda to BLEDevice.h If you scroll up a bit you will see where I added it.

chegewara, 2018-05-16:
There is no Task class in arduino library. You can try to copy it, but i dont remember if it works in arduino environment.

S-March, 2018-05-16:
This is what I used:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/Task.h

krewtib, 2018-05-16:
Wow. Thanks for so much help.

After making all the changes there is still one error left:

/var/folders/xj/nnxznz2d4c1bvst5v_mjq3yr0000gn/T/arduino_modified_sketch_654806/ESP32_ANCS.ino: In member function 'virtual void MainBLEServer::run(void*)':
ESP32_ANCS:217: error: 'class BLEAdvertisementData' has no member named 'setServiceSolicitation'
         oAdvertisementData.setServiceSolicitation(BLEUUID("7905F431-B5CE-4E99-A40F-4B1E122D00D0"));
                            ^
'class BLEAdvertisementData' has no member named 'setServiceSolicitation'

S-March, 2018-05-16:
Take a look at the very first post in this thread, I created a setServiceSolicitation() in BLEAdvertising.h

krewtib, 2018-05-16:
first i added in BLEAdvertising.h:

in class BLEAdvertisementData {
	
in public:
behind	
    void setShortName(std::string name);
    void setServiceSolicitation(BLEUUID uuid);
    
then i added in BLESdvertising.cpp

behind
std::string BLEAdvertisementData::getPayload() {
	return m_payload;
} // getPayload

the routine from your post
/**
* @brief Set the service solicitation (UUID)
* @param [in] uuid The UUID to set with the service solicitation data.  Size of UUID will be used.
*/
void BLEAdvertisementData::setServiceSolicitation(BLEUUID uuid)
{ ......
}

but still i get a error:
ESP32_ANCS:217: error: 'class BLEAdvertisementData' has no member named 'setServiceSolicitation'

S-March, 2018-05-16:
Hmm... In my BLEAdvertising.cpp I put the following section of code immediately after the following function.
```void BLEAdvertisementData::setPartialServices(BLEUUID uuid)```
```
/**
 * @brief Set the service solicitation (UUID)
 * @param [in] uuid The UUID to set with the service solicitation data.  Size of UUID will be used.
 */
void BLEAdvertisementData::setServiceSolicitation(BLEUUID uuid)
{
	char cdata[2];
	switch(uuid.bitSize()) {
		case 16: {
			// [Len] [0x14] [UUID16] data
			cdata[0] = 3;
			cdata[1] = ESP_BLE_AD_TYPE_SOL_SRV_UUID;  // 0x14
			addData(std::string(cdata, 2) + std::string((char *)&uuid.getNative()->uuid.uuid16,2));
			break;
		}

		case 128: {
			// [Len] [0x15] [UUID128] data
			cdata[0] = 17;
			cdata[1] = ESP_BLE_AD_TYPE_128SOL_SRV_UUID;  // 0x15
			addData(std::string(cdata, 2) + std::string((char *)uuid.getNative()->uuid.uuid128,16));
			break;
		}

		default:
			return;
	}
} // setServiceSolicitationData
```

S-March, 2018-05-17:
On your phone, forget the device and try repairing it. If it is pairing, it should just work with code as is. What version of iOS are you running? Also when you send an email or text or something to phone, make sure you aren't accidentally reading it or removing it from the notification center. If your phone thinks you saw the notification, it won't forward them to the ESP

krewtib, 2018-05-17:
I am using iPhone X with IOS 11.3.1.
Did as you suggested. 
Forgot the Device (The Device disconnected properly)
Paired again. I did not see any pairing data from onPassKeyRequest. But Device connected.
Sent email to phone.
Did not read the mail on the phone.
Still no respond.

S-March, 2018-05-18:
Hmmm... When you paired the device what was the output on the UART stream?

You can also try turning on verbose debugging and seeing what the output is as the connection is taking place.

CoretechR, 2018-05-19:
Great work with the ANCS implementation! I am also trying to run this in the Arduino environment. But even after applying the changes mentioned here, I get tons of errors. 
For example: 'class MainBLEServer' has no member named 'start'
and 'class MyClient' has no member named 'setStackSize'
I know that you are not finished with your project yet, but would one of you mind putting the working code and modified library files somewhere on github?

chegewara, 2018-05-19:
https://github.com/nkolban/esp32-snippets/issues/430#issuecomment-389659442

CoretechR, 2018-05-22:
OK, I got to the same point as @krewtib now. The behavior is the same, no notifications are displayed and it disconnects after a few seconds. 
```
********************
**Device connected**
00:00:00:00:00:00
********************
E (29473) BT: lmp_version_below LMP version 6 < 8
E (29477) BT: earlier enc was not done for same device

E (29561) BT: resolving list can not be edited, EnQ now
E (29621) BT: Call back not found for application conn_id=3
E (29621) BT: Call back not found for application conn_id=5
E (30978) BT: FOR LE SC LTK IS USED INSTEAD OF STK
E (59478) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=259 reason=0x0100
E (59479) BT: p_conn is NULL in bta_gattc_conn_cback

E (59480) BT: bta_gattc_conn_cback() - cif=5 connected=0 conn_id=261 reason=0x0100
E (59487) BT: p_conn is NULL in bta_gattc_conn_cback

************************
**Device  disconnected**
************************
```
What did the output looks like with your working solution, @S-March?
Maybe it is just a problem with Arduino.

krewtib, 2018-05-22:
@Max K

Maybe also a difference in the ESP32 module? Mine has following print on the lid:

ESP-WROOM-32
211-161007
FCC ID:2AC7Z—ESPWRoom32

I have got 2 of these but none of them work with this sketch.
Other sketches are running fine.

CoretechR, 2018-05-22:
My board has the identical print on it. Same ID. But I doubt that this problem is hardware related. There is so much that could be configured incorrectly just because we are using the files from the Arduino library.

chegewara, 2018-05-22:
It would be nice if @S-March could share his code as he promissed.

krewtib, 2018-05-22:
Okay. 
Could have been a source of errors. Let us wait for the code.

CoretechR, 2018-05-22:
I might be doing something wrong but I put the files from your library folder into the ESP32_BLE_Arduino library directory. Now there is this error:

> C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp: In member function 'bool BLEClient::connect(BLEAddress)':
> 
> C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:114:2: error: invalid conversion from 'int' to 'esp_ble_addr_type_t' [-fpermissive]
> 
>   );
> 
>   ^
> 
> C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:114:2: error: too few arguments to function 'esp_err_t esp_ble_gattc_open(esp_gatt_if_t, uint8_t*, esp_ble_addr_type_t, bool)'
> 
> In file included from C:\Users\Max\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEClient.cpp:13:0:
> 
> C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid/esp_gattc_api.h:294:11: note: declared here
> 
>  esp_err_t esp_ble_gattc_open(esp_gatt_if_t gattc_if, esp_bd_addr_t remote_bda, esp_ble_addr_type_t remote_addr_type, bool is_direct);
> 
>            ^
> 
> exit status 1

krewtib, 2018-05-22:
> I don't have my full project (more than just ANCS) done yet, but I pulled out the ANCS portion of it. This code works for me on an esp32 module. I uploaded it to a module just now to verify and it is working.
>
> https://github.com/S-March/esp32_ANCS <https://github.com/S-March/esp32_ANCS>

What Version of Arduino are you using? I am using 1.8.5 on a iMac with macOS High Sierra Vers. 10.13.4

I just downloaded your source from GitHub. Created a complete new directory as Arduino ANCS_TEST. In this directory i copied your ANCS Folder and the Libraries Folder.
Compiling brings this result:

/Users/krewtib/Documents/Arduino ESP32 ANCS_TEST/ANCS/ANCS.ino:10:18: fatal error: Task.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board ESP32 Dev Module.

This is a error i really cannot understand as the files Task.h and Task.cpp are in your Libraries Folder.

Funny ain’t it?

CoretechR, 2018-05-22:
It works! 😄
I am getting notifications from my iPhone. Thank you so much, @S-March !
These are changes needed for the latest ESP32 core:

In File BLEDescriptor.cpp replace:
`m_bleUUID.equals(BLEUUID(param->add_char_descr.char_uuid)) &&`
with:
`m_bleUUID.equals(BLEUUID(param->add_char_descr.descr_uuid)) &&`

In file BLEClient.cpp replace;
```
errRc = ::esp_ble_gattc_open(
	getGattcIf(),
	*getPeerAddress().getNative(), // address																																  
	1                              // direct connection
);
```
with:
```
errRc = ::esp_ble_gattc_open(
	getGattcIf(),
	*getPeerAddress().getNative(), // address
	BLE_ADDR_TYPE_PUBLIC,
	1                              // direct connection
);
```

S-March, 2018-05-22:
Fantastic! Glad it works. Let me know if future questions come up.

krewtib, 2018-05-22:
@Max
Are you sure that these are the only files to be changed?

As soon as i change In File BLEDescriptor.cpp
„add_char_descr.char_uuid" to „add_char_descr.descr_uuid" i get compiler errors.

> Am 22.05.2018 um 21:09 schrieb Max K <notifications@github.com>:
> 
> m_bleUUID.equals(BLEUUID(param->add_char_descr.descr_uuid)) &&

CoretechR, 2018-05-22:
@krewtib Only these two files. I installed the ESP32 Arduino core just yesterday so it should be the latest version. Attached is a zip folder with the files that are working for me: 
[ANCS Arduino.zip](https://github.com/nkolban/esp32-snippets/files/2028346/ANCS.Arduino.zip)

krewtib, 2018-05-22:
@Max
Now i am fully confused. Please just tell me where to copy which directories.
Must i have the ESP32_BLE_Arduino in my Sketch Folder Libraries or in the Contents/Java/hardware/espressif/esp32/libraries or in both ? Only in the Core gives errors regarding Task.h

CoretechR, 2018-05-22:
@krewtib Just noticed you are on a Mac but it should be the same as in Windows:
Folder ESP32_BLE_Arduino needs to be in "documents/Arduino/libraries"
Folder ANCS in "documents/Arduino"

krewtib, 2018-05-22:
Sorry that doesn’t work.
I have a Folder named Arduino ESP32 ANCS_Test where i have to put my Sketch folder ANCS with ANC.ino in it. Within This folder (Arduino ESP32 ANCS_TEST) i have  a folder call libraries where i can put my Arduino libs like SSD1306_lib or TaskScheduler.
My main Arduino.App  is in my Application folder and here i have the folders Contents/Java/hardware/espressif/esp32/libraries.
As soon as i put the ESP32_BLE_Arduino folder im my Sketch folder libraries i get compile errors.
I copied the ESP32_BLE_Arduino folder into  Contents/Java/hardware/espressif/esp32/libraries and compiled my existing Sketches. Worked.
But compiling ANCS.ino gives errors.
(Documents/Arduino ESP32 ANCS_TEST/ANCS/ANCS.ino:10:18: fatal error: Task.h: No such file or directory
compilation terminated.)

CoretechR, 2018-05-22:
I have never placed libraries inside the folder of the current sketch.
There should be a libraries folder in documents/Arduino/
If you install libraries via the Arduino IDE library manager, where do they usually go?

krewtib, 2018-05-22:
In this case they go into the /Arduino ESP32 ANCS_TEST/Libraries.

I use these Libraries directories in the sketch folders as i can separate them according to my Arduino applications and don’t have a huge Lib directory.

CoretechR, 2018-05-22:
Including libraries in the sketch folder does not work for me at all.
Have you tried using the full path to the source files like: `#include "Libraries/ESP32_BLE_Arduino/Task.h"`

krewtib, 2018-05-22:
Will give it try tomorrow.

CoretechR, 2018-05-23:
I've made some progress today with retrieving the notification content:
In addition to the category the Host also sends the UID of the notification. This can be used to request more details such as app and title. For example:
```
const uint8_t v[]={0x0, ID[0],ID[1],ID[2],ID[3], 0x5};
pControlPointCharacteristic->writeValue((uint8_t*)v,6,true);
```

This is the output after requesting app, title, content and date:

> New notification!
> Category: Other
> Requesting details...
> Notify callback for characteristic 22eac6e9-24d6-4bb5-be44-b36ace7c7bfb of data length 26
> 0 17 0 0 0 0 12 0 com.apple.podcasts
> Notify callback for characteristic 22eac6e9-24d6-4bb5-be44-b36ace7c7bfb of data length 8
> 0 17 0 0 0 1 0 0 
> Notify callback for characteristic 22eac6e9-24d6-4bb5-be44-b36ace7c7bfb of data length 114
> 0 17 0 0 0 3 6A 0 Es gibt eine neue Folge „Stuff You Should Know“, SYSK Selects: Is brain size related to intelligence?.
> Notify callback for characteristic 22eac6e9-24d6-4bb5-be44-b36ace7c7bfb of data length 23
> 0 17 0 0 0 5 F 0 20180519T190930

S-March already implemented the callback for receiving the requested data.
It is working perfectly fine but my code is kind of a mess. I didn't know where to put the request other than in the myclient class. The code is running in a loop, waiting for NotificationSourceNotifyCallback to set a flag:

```
/** END ANCS SERVICE **/
        
delay(1000);

while(1){
    if(pendingNotification == true){
        // CommandID: CommandIDGetNotificationAttributes
        // 32bit uid
        // AttributeID
        Serial.println("Requesting details...");
        const uint8_t vIdentifier[]={0x0,   latestMessageID[0],latestMessageID[1],latestMessageID[2],latestMessageID[3],   0x0};
        pControlPointCharacteristic->writeValue((uint8_t*)vIdentifier,6,true);
        const uint8_t vTitle[]={0x0,   latestMessageID[0],latestMessageID[1],latestMessageID[2],latestMessageID[3],   0x1, 0x0, 0x10};
        pControlPointCharacteristic->writeValue((uint8_t*)vTitle,8,true);
        const uint8_t vMessage[]={0x0,   latestMessageID[0],latestMessageID[1],latestMessageID[2],latestMessageID[3],   0x3, 0x0, 0x10};
        pControlPointCharacteristic->writeValue((uint8_t*)vMessage,8,true);
        const uint8_t vDate[]={0x0,   latestMessageID[0],latestMessageID[1],latestMessageID[2],latestMessageID[3],   0x5};
        pControlPointCharacteristic->writeValue((uint8_t*)vDate,6,true);
        pendingNotification = false;
    }
    delay(100); //does not work without small delay
}
```
Any ideas how this could be improved?

krewtib, 2018-05-23:
@Max K
just tried  #include "Libraries/ESP32_BLE_Arduino/Task.h“. Didn’t work so i changed to #include "/Applications/Arduino.app/Contents/Java/hardware/espressif/esp32/libraries/BLE/src/Task.h“. This did it.
But there is a new error:

/Documents/Arduino ESP32 ANCS_TEST/ANCS/ANCS.ino: In member function 'virtual void MyServerCallbacks::onConnect(BLEServer*)':
ANCS:176: error: 'm_remoteBda' is not a member of 'BLEDevice'
         Serial.println(BLEAddress(BLEDevice::m_remoteBda).toString().c_str());
                                   ^
ANCS:180: error: 'm_remoteBda' is not a member of 'BLEDevice'
         pMyClient->start(new BLEAddress(BLEDevice::m_remoteBda));
                                         ^
exit status 1
'm_remoteBda' is not a member of ‚BLEDevice'

[44 more replies]