nkolban/esp32-snippets #431: BLE Server:  Service can't be connected to when # of characteristics > 5

joehui, 2018-03-10:
With the following code, when there are only 5 characteristics, everything works fine.  But, once I increase the # of characteristics to 6 or more, the BLE client on my smartphone can't see or connect to the Service any more.   I've tested this on more than 1 BLE client so I am sure it is not a problem with the client.

Is there any limit on the number of characteristics I can have?  Or, can you anything I have done wrong?

Your help will be greatly appreciated.

```
#include <esp_log.h>
#include "freertos/FreeRTOS.h"
#include <driver/adc.h>
#include "esp_system.h"
#include "nvs_flash.h"
#include "driver/gpio.h"
#include "BLEServer.h"
#include "BLEDevice.h"

#define BLE_DEVICE_NAME "BleTest"

#define SERVICE_UUID	"e87c9761-597f-4e0c-b20d-d49052a827a0"
#define CHARACTERISTIC1 "e87c9761-597f-4e0c-b20d-d49052a827a1"
#define CHARACTERISTIC2 "e87c9761-597f-4e0c-b20d-d49052a827a2"
#define CHARACTERISTIC3 "e87c9761-597f-4e0c-b20d-d49052a827a3"
#define CHARACTERISTIC4 "e87c9761-597f-4e0c-b20d-d49052a827a4"
#define CHARACTERISTIC5 "e87c9761-597f-4e0c-b20d-d49052a827a5"
#define CHARACTERISTIC6 "e87c9761-597f-4e0c-b20d-d49052a827a6"

BLECharacteristic *pCharacteristic1;
BLECharacteristic *pCharacteristic2;
BLECharacteristic *pCharacteristic3;
BLECharacteristic *pCharacteristic4;
BLECharacteristic *pCharacteristic5;
BLECharacteristic *pCharacteristic6;

// handler for BLE write
class myCharacteristicCallbacks: public BLECharacteristicCallbacks {
	void onWrite(BLECharacteristic *pCharacteristic) {
	};

	void onRead(BLECharacteristic *pCharacteristic) {
	}
};

// BLE Manager
void startBleManager() {
	char myBLEDeviceName[40];
	uint8_t chipid[6];

	esp_efuse_mac_get_default(chipid);
	sprintf(myBLEDeviceName, BLE_DEVICE_NAME":%02x:%02x:%02x:%02x:%02x:%02x\n",chipid[0], chipid[1], chipid[2], chipid[3], chipid[4], chipid[5]);

	BLEDevice::init(myBLEDeviceName);

	BLEServer *pServer = BLEDevice::createServer();
	BLEService *pService = pServer->createService(BLEUUID(SERVICE_UUID));

	pCharacteristic1 =  pService->createCharacteristic(
				BLEUUID(CHARACTERISTIC1),
				BLECharacteristic::PROPERTY_READ
			);
        pCharacteristic2 =  pService->createCharacteristic(
                                BLEUUID(CHARACTERISTIC2),
                                BLECharacteristic::PROPERTY_READ
                        );
        pCharacteristic3 =  pService->createCharacteristic(
                                BLEUUID(CHARACTERISTIC3),
                                BLECharacteristic::PROPERTY_READ
                        );
        pCharacteristic4 =  pService->createCharacteristic(
                                BLEUUID(CHARACTERISTIC4),
                                BLECharacteristic::PROPERTY_READ
                        );
        pCharacteristic5 =  pService->createCharacteristic(
                                BLEUUID(CHARACTERISTIC5),
                                BLECharacteristic::PROPERTY_READ
                        );
        pCharacteristic6 =  pService->createCharacteristic(
                                BLEUUID(CHARACTERISTIC6),
                                BLECharacteristic::PROPERTY_READ
                        );

	pCharacteristic1->setCallbacks(new myCharacteristicCallbacks());
	pCharacteristic2->setCallbacks(new myCharacteristicCallbacks());
	pCharacteristic3->setCallbacks(new myCharacteristicCallbacks());
	pCharacteristic4->setCallbacks(new myCharacteristicCallbacks());
	pCharacteristic5->setCallbacks(new myCharacteristicCallbacks());
	pCharacteristic6->setCallbacks(new myCharacteristicCallbacks());

	pService->start();
	BLEAdvertising *pAdvertising = pServer->getAdvertising();
	pAdvertising->start();
}

extern "C" void app_main(void)
{
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES) {
        // NVS partition was truncated and needs to be erased
        // Retry nvs_flash_init
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK( err );

    // starts the Wifi Connection Manager, which will start other tasks once connected
    startBleManager();
}

```

chegewara, 2018-03-10:
Hi,
did you try to add second parameter (handles number) in createService?
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEService.cpp#L47

joehui, 2018-03-10:
Just tried this:

`        BLEService *pService = pServer->createService(BLEUUID(SERVICE_UUID), 200);
`
No difference at all.

chegewara, 2018-03-10:
Its been working some time ago with over 20 characteristics + descriptors. If you are using android device to connect then you have to remember that android is caching connections and you need to connect few times to see difference.

joehui, 2018-03-11:
I am testing this using an ios app called "BLE Scanner" (also testing with my own React Native app).  I tried reloading the client app many times.  It still doesn't work unless I reduce the # of Characteristics to 5.  I even increased the handle count to 2000.  No change to the behavior at all.

joehui, 2018-03-11:
More info.   If I do not increase the max # of handles and I have 8 or more characteristics, I would get the following error when looking at serial monitor:

`E (558) BLEUtils: [status: ESP_GATT_ERROR, attr_handle: 0 0x00, service_handle: 40 0x28, char_uuid: e87c9761-597f-4e0c-b20d-d49052a827a8]
`

Increasing the handle count solves this error, but the client would still not be able to connect to the service.  Thus, my originally reported issue at 6 or more characteristics is a different.

Let me summarize again:

**Scenario#1**      # of characteristics <= 5 
    Everything works!

**Scenario#2**      # of characteristics >= 6  (regardless of what max # of handles is)
    Cannot connect to the BLE server service

**Scenario#3**      # of characteristics >= 8  AND using default # of handles
    ESP_GATT_ERROR, attr_handle: 0 0x00 error

**Scenario#4**      # of characteristics >= 8  AND using increased # of handles
    No more ESP_GATT_ERROR, attr_handle error, but still cannot connect to the BLE server service

justind000, 2018-04-23:
I have also found this error. I was able to use @joehui comment above to fix my issue. I don't have the exact numbers, but I did notice that I at first tried using the exact number of handles I was using and it didn't work, it still only worked for some many. I increased it from something like 20 to 50 and it started working.

mrengineer, 2018-08-01:
Does somebody solved this problem?

chegewara, 2018-08-03:
There is no problem, you have to create service with handles number parameter. I didnt know how many handles you need to setup, but lately someone corrected me and said that you need 2 handles per each characteristic + 1 handle per each descriptor + 1 handle for service.