nkolban/esp32-snippets #435: BLE communication

mingshima, 2018-03-14:
Hello all, I am doing a project of using BLE to transmitter audio between two ESP32, I tried the examples in ESP32_BLE_Arduino. I use one board as a server and another as a client. The server connects with a microphone and the client connects with a speaker .

I have tried to read value of microphone and write the value to a speaker in a same ESP32 board, and it works very well, but when I tried to  separate mic and speaker and connect them with bluetooth, problem occurs. The speaker can react to the mic, but can't make smooth voice, so I tried to find the problems.

I find the BLE server can read the value of mic and send it to BLE client about 10000 times in  a second, that is to say, the sample frequency is about 10KHz, which can basically meet the needs, but in BLE client, 1000 times of reading the data received from server will take about 30s, the frequency is about 30Hz, this caused a huge data lose, so the voice is very strange.

ps: I use the official example of ESP32, when I  delete the read-value code of BLE client, it will run very fast, so I am sure this is the problem.

Can anyone help me to solve this problem? Any suggestion is admired.

this is the code, just changed a little.
the server
```
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <driver/adc.h>

BLECharacteristic *pCharacteristic;

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

void setup() {
 
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("ESP");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );
                                       
  //pCharacteristic->setValue("Hello World says Neil");
  pService->start();
  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->addServiceUUID(pService->getUUID());
  pAdvertising->start();
  //Serial.println("Characteristic defined! Now you can read it in your phone!");
}

void loop() {
  //ADC, ADC1_CHANNEL_5 is GPIO33, 12bits, 0db->full 1.1V
  adc1_config_width(ADC_WIDTH_12Bit);
  adc1_config_channel_atten(ADC1_CHANNEL_5,ADC_ATTEN_0db);
  int val = adc1_get_voltage(ADC1_CHANNEL_5);
  //Serial.println(val);
  
  char txString0[10];
  dtostrf(val, 1, 2, txString0);
  pCharacteristic->setValue(txString0); 

  //to calculate the speed of sample
  /*
   static int time1 = millis();
    static int i = 0;
    i++;
    if (i == 10000)
      Serial.println(millis()-time1);
  */
  
}
```

the client
```
#include <cstdlib>
#include <string>
//#include <driver/dac.h>
#include "BLEDevice.h"
//#include "BLEScan.h"

// The remote service we wish to connect to.
static BLEUUID serviceUUID("4fafc201-1fb5-459e-8fcc-c5c9c331914b");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID("beb5483e-36e1-4688-b7f5-ea07361b26a8");

static BLEAddress *pServerAddress;
static boolean doConnect = false;
static boolean connected = false;

static BLERemoteCharacteristic* pRemoteCharacteristic;

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
}

bool connectToServer(BLEAddress pAddress) {
    Serial.print("Forming a connection to ");
    Serial.println(pAddress.toString().c_str());
    
    BLEClient*  pClient  = BLEDevice::createClient();
    Serial.println(" - Created client");

    // Connect to the remove BLE Server.
    pClient->connect(pAddress);
    Serial.println(" - Connected to server");

    // Obtain a reference to the service we are after in the remote BLE server.
    BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService == nullptr) {
      Serial.print("Failed to find our service UUID: ");
      Serial.println(serviceUUID.toString().c_str());
      return false;
    }
    Serial.println(" - Found our service");

    // Obtain a reference to the characteristic in the service of the remote BLE server.
    pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
    if (pRemoteCharacteristic == nullptr) {
      Serial.print("Failed to find our characteristic UUID: ");
      Serial.println(charUUID.toString().c_str());
      return false;
    }
    Serial.println(" - Found our characteristic");

    // Read the value of the characteristic.
    //std::string value = pRemoteCharacteristic->readValue();
    //Serial.print("The characteristic value was: ");
    //Serial.println(value.c_str());

    pRemoteCharacteristic->registerForNotify(notifyCallback);
}
/**
 * Scan for BLE servers and find the first one that advertises the service we are looking for.
 */
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
 /**
   * Called for each advertising BLE server.
   */
  void onResult(BLEAdvertisedDevice advertisedDevice) {
    Serial.print("BLE Advertised Device found: ");
    Serial.println(advertisedDevice.toString().c_str());

    // We have found a device, let us now see if it contains the service we are looking for.
    if (advertisedDevice.haveServiceUUID() && advertisedDevice.getServiceUUID().equals(serviceUUID)) {

      // 
      Serial.print("Found our device!  address: "); 
      advertisedDevice.getScan()->stop();

      pServerAddress = new BLEAddress(advertisedDevice.getAddress());
      doConnect = true;

    } // Found our server
  } // onResult
}; // MyAdvertisedDeviceCallbacks

void setup() {
  Serial.begin(115200);
  Serial.println("Starting Arduino BLE Client application...");
  BLEDevice::init("");

  // Retrieve a Scanner and set the callback we want to use to be informed when we
  // have detected a new device.  Specify that we want active scanning and start the
  // scan to run for 30 seconds.
  BLEScan* pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true);
  pBLEScan->start(30);
  
} // End of setup.

// This is the Arduino main loop function.
void loop() {

  // If the flag "doConnect" is true then we have scanned for and found the desired
  // BLE Server with which we wish to connect.  Now we connect to it.  Once we are 
  // connected we set the connected flag to be true.
  if (doConnect == true) {
    if (connectToServer(*pServerAddress)) {
      Serial.println("We are now connected to the BLE Server.");
      connected = true;
    } else {
      Serial.println("We have failed to connect to the server; there is nothing more we will do.");
    }
    doConnect = false;
  }

  // If we are connected to a peer BLE Server, update the characteristic each time we are reached
  // with the current time since boot.
 /* if (connected) {
    String newValue = "Time since boot: " + String(millis()/1000);
    Serial.println("Setting new characteristic value to \"" + newValue + "\"");
    
    // Set the characteristic's value to be the array of bytes that is actually a string.
    //pRemoteCharacteristic->writeValue(newValue.c_str(), newValue.length());
    
  }*/
  if (connected) {
    
    std::string str = pRemoteCharacteristic->readValue();
    //int number = std::atoi( str.c_str() );
    //int output = map(number, 0, 4095, 0, 255);
    //Serial.println(output);
    //dacWrite(25, output);

    /*static int time1 = millis();
    static int i = 0;
    i++;
    if (i == 10000)
      Serial.println(millis()-time1);*/
  }
    
  //delay(50); // Delay a second between loops.
}
```

chegewara, 2018-03-14:
Hi,
first suggestion is to set mtu value to about 500 (512 max). Second suggestion is to use classic bluetooth instead. Bluetooth LE is designed to transfer small amount of data, so sending audio data can be hard or even impossible to achieve.

mingshima, 2018-03-14:
Yeah, that is the problem. I find that BLE can't transfer large data such as audio, but there is not enough example in esp32 arduino of classic bluetooth, just an example of BluetoothSerial. Do you know other examples of classic bluetooth? Thanks.

chegewara, 2018-03-14:
https://github.com/kodera2t/ESP32_OLED_webradio

https://github.com/MrBuddyCasino/ESP32_MP3_Decoder

hansmbakker, 2018-12-04:
@mingshima is this still an issue? If it is solved, could you please close it to keep the issue list clean?

cc @chegewara

closed 2018-12-06 by mingshima