nkolban/esp32-snippets #436: Can't compile with Platform.io

ayhoung, 2018-03-15:
I'm trying to compile with the below code in Platform.io
```
#include <Arduino.h>
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

int scanTime = 30; //In seconds

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
    }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

  BLEDevice::init("");
  BLEScan* pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  BLEScanResults foundDevices = pBLEScan->start(scanTime);
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);
}
```

This works perfectly in the Arduino IDE, but for some reason I'm getting the below error when compiling with Platformio:
```
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 32 > Espressif ESP32 Dev Module
SYSTEM: ESP32 240MHz 288KB RAM (1.25MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 19 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP32 BLE Arduino> v0.4.9
Compiling .pioenvs/esp32dev/src/main.cpp.o
Generating partitions .pioenvs/esp32dev/partitions.bin
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLE2902.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLE2904.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLEAddress.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLEAdvertisedDevice.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLEAdvertising.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLEBeacon.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLECharacteristic.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLECharacteristicMap.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLEClient.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLEDescriptor.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLEDescriptorMap.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLEDevice.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLEExceptions.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLEHIDDevice.cpp.o
Compiling .pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLERemoteCharacteristic.cpp.o
/Users/allen/.platformio/lib/ESP32 BLE Arduino_ID1841/src/BLEClient.cpp: In member function 'BLERemoteService* BLEClient::getService(BLEUUID)':
/Users/allen/.platformio/lib/ESP32 BLE Arduino_ID1841/src/BLEClient.cpp:336:12: error: exception handling disabled, use -fexceptions to enable
throw new BLEUuidNotFoundException;
^
*** [.pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLEClient.cpp.o] Error 1
/Users/allen/.platformio/lib/ESP32 BLE Arduino_ID1841/src/BLERemoteCharacteristic.cpp: In member function 'std::__cxx11::string BLERemoteCharacteristic::readValue()':
/Users/allen/.platformio/lib/ESP32 BLE Arduino_ID1841/src/BLERemoteCharacteristic.cpp:439:34: error: exception handling disabled, use -fexceptions to enable
throw BLEDisconnectedException();
^
*** [.pioenvs/esp32dev/libc0e/ESP32 BLE Arduino_ID1841/BLERemoteCharacteristic.cpp.o] Error 1
```

I've used both configs
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/stage
board = esp32dev
framework = arduino
```
and
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
framework = arduino
```

but both don't seem to work. Has anyone else run into this issue?

dy008, 2018-03-20:
same issue!
but i can used one choice:
on platformio.ini 
change the line
platform =espressif32@0.11.1
and build is ok!

ayhoung, 2018-03-20:
awesome thanks!

ayhoung, 2018-04-01:
thanks dy008!

closed 2018-12-06 by chegewara