nkolban/esp32-snippets #437: BLEEdddystoneTLM.cpp assumes Arduino environment: won't compile in esp-idf env.

mws-rmain, 2018-03-15:
Last pull from github/pcbreflux grabbed BLEEddystoneTLM.cpp, which assumes an arduino environment (fwiw, it _is_ under an 'arduino' directory in his repo).
The dependency is due to inclusion of 'Arduino.h' header, and use of arduino 'String' function within the 'toString()' function of the class.
This could easily be changed to use standard C++ functions in <iomanip> and <sstring>, which would be more portable.

mainr, 2018-03-17:
fixed in new release from pcbreflux

rweller59, 2018-03-28:
I'm using espidf and just ran into this same (similar?) problem after updating to latest espidf, toolchain and cpp_utils - can't flash chip 

CXX build/cpp_utils/BLEEddystoneTLM.o
/home/rick/eclipse-test/OldOrb/components/cpp_utils/./BLEEddystoneTLM.cpp:7:21: fatal error: Arduino.h: No such file or directory
compilation terminated.

I've never heard of pcbreflux. How is it related to espidf? Am I supposed to be installing something?

chegewara, 2018-03-28:
@rweller59 If you did not heard about him just check YT, he has a lot usefull vids about esp32. 
If you dont need BLEEdystone in your project then my advise is to just delete both files (.h and .cpp) from cpp_folder. This class has been prepared to work in arduino environment.

mainr, 2018-03-28:
If you have sync'd your repo with esp32-snippets latest, this should work. PcbReflux fixed it, and Neil pulled his fix, so all should run without manual intervention.

rweller59, 2018-03-28:
works for me. thanks for the quick response.