nkolban/esp32-snippets #439: Using a predefined BLE GATT Service

thomasvdwege, 2018-03-17:
I can't get my predefined BLE Services to work. I try using one of the Service Identifiers from the [https://www.bluetooth.com/specifications/gatt/services](BLE specification site), but when I use them they don't show up in my bluetooth scanner. If I use a random generated UUID it shows up as a 'custom service', and I can use custom charasteristics (from [https://www.bluetooth.com/specifications/gatt/characteristics](this list).

This is the code I use:

```
		BLEDevice::init("ESP32");
		BLEServer *pServer = BLEDevice::createServer();
		BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);

		// Required in authentication process
		// to provide displaying and/or input passkey or yes/no butttons confirmation
		GenericSecurity *security = new GenericSecurity();
		BLEDevice::setSecurityCallbacks(security);

		std::string serviceuuid = "78489c82-5e4c-4c0b-b3ca-54390af7bf8c";
		// https://www.bluetooth.com/specifications/gatt/services
		BLEService *pService = pServer->createService(BLEUUID((uint16_t)0x180F));

		// https://www.bluetooth.com/specifications/gatt/characteristics
		BLECharacteristic *ssidCharacteristic = pService->createCharacteristic(
			BLEUUID((uint16_t) 0x2A80),
			BLECharacteristic::PROPERTY_WRITE);
		
		BLECharacteristic *passwordCharacteristic = pService->createCharacteristic(
			BLEUUID((uint16_t)0x2A01),
			BLECharacteristic::PROPERTY_WRITE);

		ssidCharacteristic->setAccessPermissions(ESP_GATT_PERM_WRITE_ENCRYPTED);
		ssidCharacteristic->setCallbacks(new BleCallbackWifi());

		passwordCharacteristic->setAccessPermissions(ESP_GATT_PERM_WRITE_ENCRYPTED);
		passwordCharacteristic->setCallbacks(new BleCallbackWifi());

		// https://www.bluetooth.com/specifications/gatt/descriptors
		BLE2902 *ssid2902Descriptor = new BLE2902();
		BLE2902 *password2902Descriptor = new BLE2902();
		// p2902Descriptor->setNotifications(true);
		ssidCharacteristic->addDescriptor(ssid2902Descriptor);
		passwordCharacteristic->addDescriptor(password2902Descriptor);

		pService->start();

		BLEAdvertising *pAdvertising = pServer->getAdvertising();
		pAdvertising->addServiceUUID(BLEUUID(pService->getUUID()));

		BLESecurity *pSecurity = new BLESecurity();
		pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_MITM_BOND);
		pSecurity->setCapability(ESP_IO_CAP_OUT);
		pSecurity->setInitEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);

		pAdvertising->start();
		ESP_LOGI(tag, "Advertising started!");
```

chegewara, 2018-03-17:
Are you using arduino-ide or esp-idf? If esp-idf then you have to change settings in `make menuconfig` like here:
https://github.com/nkolban/esp32-snippets/issues/264#issuecomment-357353186

thomasvdwege, 2018-03-17:
I am using esp-idf, thanks for the quick fix. I'll post it here again, for anyone else encountering the problem:

This fixes your BLE device not broadcasting predefined BLE Services:

I now remember where is issue you can have just now, try to run menuconfig and change Compiler options->Optimization Level->Release. We already posted this issue but still its not fixed. Sorry for inconvenience

closed 2018-03-17 by thomasvdwege