nkolban/esp32-snippets #442: Bug when using two characteristics with the same identifier

thomasvdwege, 2018-03-18:
When I create two different characteristics with the same 'identifier'
 `BLEUUID((uint16_t)0x2A3D`
it looks like the onWrite handlers are not assigned correctly. Both values report
`"SSID: BLE received on ESP32/write : %s"`,
but the passwordhandler should respond with
`"Password: BLE received on ESP32/write: %s"`. It worked one time, but I didn't even change anything, I just rebooted the ESP32.

Using two different characteristics fixes this, but is there any way I can use two (or more) of the same characteristic?

```
BLEDevice::init(name);
	BLESecurity *pSecurity = new BLESecurity();
	pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_MITM_BOND);
	pSecurity->setCapability(ESP_IO_CAP_OUT);
	pSecurity->setInitEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);
	BLEServer *pServer = BLEDevice::createServer();
	BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);
		// Required in authentication process to provide displaying and/or input passkey or yes/no butttons confirmation
	BLEDevice::setSecurityCallbacks(new GenericSecurity());
	// https://www.bluetooth.com/specifications/gatt/services
	BLEService *pService = pServer->createService(BLEUUID((uint16_t)0x180A));
	// https://www.bluetooth.com/specifications/gatt/characteristics
	BLECharacteristic *ssidCharacteristic = pService->createCharacteristic(
		BLEUUID((uint16_t)0x2A3D),
		BLECharacteristic::PROPERTY_WRITE);
	
	BLECharacteristic *passwordCharacteristic = pService->createCharacteristic(
		BLEUUID((uint16_t)0x2A3D),
		BLECharacteristic::PROPERTY_WRITE);

	ssidCharacteristic->setAccessPermissions(ESP_GATT_PERM_WRITE_ENCRYPTED);
	ssidCharacteristic->setCallbacks(new BleCallbackSSID());

	passwordCharacteristic->setAccessPermissions(ESP_GATT_PERM_WRITE_ENCRYPTED);
	passwordCharacteristic->setCallbacks(new BleCallbackPassword());
	// https://www.bluetooth.com/specifications/gatt/descriptors
	BLE2902 *ssid2902Descriptor = new BLE2902();
	BLE2902 *password2902Descriptor = new BLE2902();
	ssidCharacteristic->addDescriptor(ssid2902Descriptor);
	passwordCharacteristic->addDescriptor(password2902Descriptor);
```

```
// SSID
void BleCallbackSSID::onWrite(BLECharacteristic *pCharacteristic)
{
    std::string msg = pCharacteristic->getValue();
    ESP_LOGI(tag, "SSID: BLE received on ESP32/write: %s", msg.c_str());
}

void BleCallbackSSID::onRead(BLECharacteristic *pCharacteristic)
{
    std::string msg = pCharacteristic->getValue();
    ESP_LOGI(tag, "BLE read: %s, %i", msg.c_str(), msg.length());
}

// Password
void BleCallbackPassword::onWrite(BLECharacteristic *pCharacteristic)
{
    std::string msg = pCharacteristic->getValue();
    ESP_LOGI(tag, "Password: BLE received on ESP32/write: %s", msg.c_str());
}

void BleCallbackPassword::onRead(BLECharacteristic *pCharacteristic)
{
    std::string msg = pCharacteristic->getValue();
    ESP_LOGI(tag, "BLE read: %s, %i", msg.c_str(), msg.length());
}
```

chegewara, 2018-03-18:
It may be issue with your client app. If you have 2 characteristics with the same UUID it requires to access characteristics with handle, but your client app can access it with UUID and sends value randomly.

thomasvdwege, 2018-03-19:
For now I'll stick to 2 different UUID's, when I've written my own client I'll report back. My client being buggy could definitely be the case.

TimonPeng, 2019-12-24:
> It may be issue with your client app. If you have 2 characteristics with the same UUID it requires to access characteristics with handle, but your client app can access it with UUID and sends value randomly.

How to write data to characteristics with handle?