nkolban/esp32-snippets #445: Can't discover device

lbalzs, 2018-03-22:
I just flashed the server example, theres no errors while flashing. The serial monitor output is:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
Starting BLE work!
Characteristic defined! Now you can read it in your phone!

Unfortunately I can't see my device neither on my phone, and can't discover it with hcitool neither!:(
I don't know what could be wrong. Can you help me please?

chegewara, 2018-03-22:
hi,
could you change log level to debug and paste new logs?

lbalzs, 2018-03-22:
https://pastebin.com/8vn56kY6
Here it is.

chegewara, 2018-03-22:
I meant logs from serial, when you run esp32, and those are compilation logs.

lbalzs, 2018-03-22:
Starting BLE work!
D (395) FreeRTOS: Semaphore taking: name: RegisterAppEvt (0x3ffde5cc), owner: <N/A> for registerApp
D (396) FreeRTOS: Semaphore taken:  name: RegisterAppEvt (0x3ffde5cc), owner: registerApp
D (938) FreeRTOS: Semaphore taking: name: CreateEvt (0x3ffdfa34), owner: <N/A> for createService
D (945) FreeRTOS: Semaphore taken:  name: CreateEvt (0x3ffdfa34), owner: createService
D (953) FreeRTOS: Semaphore taking: name: CreateEvt (0x3ffdfe34), owner: <N/A> for executeCreate
D (961) FreeRTOS: Semaphore taken:  name: CreateEvt (0x3ffdfe34), owner: executeCreate
D (971) FreeRTOS: Semaphore taking: name: CreateEvt (0x3ffdea4c), owner: <N/A> for executeCreate
D (978) FreeRTOS: Semaphore taken:  name: CreateEvt (0x3ffdea4c), owner: executeCreate
D (986) FreeRTOS: Semaphore taking: name: StartEvt (0x3ffdfe94), owner: <N/A> for start
D (993) FreeRTOS: Semaphore taken:  name: StartEvt (0x3ffdfe94), owner: start
Characteristic defined! Now you can read it in your phone!

Sorry.

chegewara, 2018-03-22:
No problem.

I made mistake assuming you are using arduino-ide.

When you updated esp-idf and cpp_utils last time? If you dont have any errors during compilation then it was some time ago. Another suggestion is to change settings following those instructions:
https://github.com/nkolban/esp32-snippets/issues/264#issuecomment-357353186

lbalzs, 2018-03-22:
I'm quite newbie to Arduino, i might have done something badly.
Here is what I did (as i found in ESP32 guide):
mkdir -p ~/Documents/Arduino/hardware/espressif
cd ~/Documents/Arduino/hardware/espressif
git clone https://github.com/espressif/arduino-esp32.git esp32
cd esp32
git submodule update --init --recursive
cd tools
python get.py

Im using Arduino-IDE.
Did i mess up something?

chegewara, 2018-03-22:
This is my log from BLE_server example. As you can see there is more log messages than in your log:
If you are using code without changes then all should be fine. (i updated my arduino (on windows) to see if there is problem)

```
Starting BLE work!
[D][BLEDevice.cpp:69] createServer(): >> createServer
[D][BLEServer.cpp:304] registerApp(): >> registerApp - 0
[D][BLEDevice.cpp:95] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_REG_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_REG_EVT
[D][BLEUtils.cpp:1791] dumpGattServerEvent(): [status: ESP_GATT_OK, app_id: 0]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_REG_EVT
[D][BLEServer.cpp:308] registerApp(): << registerApp
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:76] createServer(): << createServer
[D][BLEServer.cpp:76] createService(): >> createService - 4fafc201-1fb5-459e-8fcc-c5c9c331914b
[D][BLEDevice.cpp:95] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CREATE_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CREATE_EVT
[D][BLEUtils.cpp:1716] dumpGattServerEvent(): [status: ESP_GATT_OK, service_handle: 40 0x28, service_id: [uuid: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, inst_id: 0]]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CREATE_EVT
[D][BLEService.cpp:160] setHandle(): >> setHandle - Handle=0x28, service UUID=4fafc201-1fb5-459e-8fcc-c5c9c331914b)
[D][BLEService.cpp:166] setHandle(): << setHandle
[D][BLEService.cpp:89] executeCreate(): << executeCreate
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEServer.cpp:93] createService(): << createService
[D][BLEService.cpp:188] addCharacteristic(): >> addCharacteristic()
[D][BLEService.cpp:191] addCharacteristic(): Adding characteristic: uuid=beb5483e-36e1-4688-b7f5-ea07361b26a8 to service: UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, handle: 0x28
[D][BLEService.cpp:203] addCharacteristic(): << addCharacteristic()
[D][BLECharacteristic.cpp:664] setValue(): >> setValue: length=21, data=48656c6c6f20576f726c642073617973204e65696c, characteristic UUID=beb5483e-36e1-4688-b7f5-ea07361b26a8
[D][BLECharacteristic.cpp:671] setValue(): << setValue
[D][BLEService.cpp:125] start(): >> start(): Starting service (esp_ble_gatts_start_service): UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, handle: 0x28
[D][BLECharacteristic.cpp:83] executeCreate(): >> executeCreate()
[D][BLECharacteristic.cpp:94] executeCreate(): Registering characteristic (esp_ble_gatts_add_char): uuid: beb5483e-36e1-4688-b7f5-ea07361b26a8, service: UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, handle: 0x28
[D][BLEDevice.cpp:95] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_ADD_CHAR_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_ADD_CHAR_EVT
[D][BLEUtils.cpp:1669] dumpGattServerEvent(): [status: ESP_GATT_OK, attr_handle: 42 0x2a, service_handle: 40 0x28, char_uuid: beb5483e-36e1-4688-b7f5-ea07361b26a8]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_ADD_CHAR_EVT
[D][BLECharacteristic.cpp:609] setHandle(): >> setHandle: handle=0x2a, characteristic uuid=beb5483e-36e1-4688-b7f5-ea07361b26a8
[D][BLECharacteristic.cpp:611] setHandle(): << setHandle
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_ADD_CHAR_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:135] executeCreate(): << executeCreate
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:95] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_START_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_START_EVT
[D][BLEUtils.cpp:1805] dumpGattServerEvent(): [status: ESP_GATT_OK, service_handle: 0x28]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_START_EVT
[D][BLEService.cpp:151] start(): << start()
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_START_EVT
[D][BLEAdvertising.cpp:164] start(): >> start: customAdvData: 0, customScanResponseData: 0
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEAdvertising.cpp:184] start(): - no services advertised
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEAdvertising.cpp:221] start(): << start
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT
Characteristic defined! Now you can read it in your phone!
[D][BLEUtils.cpp:1099] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RSP_DATA_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1285] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
[D][BLEUtils.cpp:1123] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!

```

lbalzs, 2018-03-22:
Starting BLE work!
[D][BLEDevice.cpp:69] createServer(): >> createServer
[D][BLEServer.cpp:304] registerApp(): >> registerApp - 0
D (400) FreeRTOS: Semaphore taking: name: RegisterAppEvt (0x3ffde60c), owner: <N/A> for registerApp
D (407) FreeRTOS: Semaphore taken:  name: RegisterAppEvt (0x3ffde60c), owner: registerApp
[D][BLEDevice.cpp:95] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_REG_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_REG_EVT
[D][BLEUtils.cpp:1791] dumpGattServerEvent(): [status: ESP_GATT_OK, app_id: 0]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_REG_EVT
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEServer.cpp:308] registerApp(): << registerApp
[D][BLEDevice.cpp:76] createServer(): << createServer
[D][BLEServer.cpp:76] createService(): >> createService - 4fafc201-1fb5-459e-8fcc-c5c9c331914b
D (1060) FreeRTOS: Semaphore taking: name: CreateEvt (0x3ffdfd3c), owner: <N/A> for createService
D (1069) FreeRTOS: Semaphore taken:  name: CreateEvt (0x3ffdfd3c), owner: createService
D (1077) FreeRTOS: Semaphore taking: name: CreateEvt (0x3ffdfc4c), owner: <N/A> for executeCreate
D (1085) FreeRTOS: Semaphore taken:  name: CreateEvt (0x3ffdfc4c), owner: executeCreate
[D][BLEDevice.cpp:95] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CREATE_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CREATE_EVT
[D][BLEUtils.cpp:1716] dumpGattServerEvent(): [status: ESP_GATT_OK, service_handle: 40 0x28, service_id: [uuid: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, inst_id: 0]]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CREATE_EVT
[D][BLEService.cpp:160] setHandle(): >> setHandle - Handle=0x28, service UUID=4fafc201-1fb5-459e-8fcc-c5c9c331914b)
[D][BLEService.cpp:166] setHandle(): << setHandle
[D][BLEService.cpp:89] executeCreate(): << executeCreate
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEServer.cpp:93] createService(): << createService
[D][BLEService.cpp:188] addCharacteristic(): >> addCharacteristic()
[D][BLEService.cpp:191] addCharacteristic(): Adding characteristic: uuid=beb5483e-36e1-4688-b7f5-ea07361b26a8 to service: UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, handle: 0x28
[D][BLEService.cpp:203] addCharacteristic(): << addCharacteristic()
[D][BLECharacteristic.cpp:664] setValue(): >> setValue: length=21, data=48656c6c6f20576f726c642073617973204e65696c, characteristic UUID=beb5483e-36e1-4688-b7f5-ea07361b26a8
[D][BLECharacteristic.cpp:671] setValue(): << setValue
[D][BLEService.cpp:125] start(): >> start(): Starting service (esp_ble_gatts_start_service): UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, handle: 0x28
[D][BLECharacteristic.cpp:83] executeCreate(): >> executeCreate()
[D][BLECharacteristic.cpp:94] executeCreate(): Registering characteristic (esp_ble_gatts_add_char): uuid: beb5483e-36e1-4688-b7f5-ea07361b26a8, service: UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, handle: 0x28
D (1248) FreeRTOS: Semaphore taking: name: CreateEvt (0x3ffdeb74), owner: <N/A> for executeCreate
D (1257) FreeRTOS: Semaphore taken:  name: CreateEvt (0x3ffdeb74), owner: executeCreate
[D][BLEDevice.cpp:95] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_ADD_CHAR_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_ADD_CHAR_EVT
[D][BLEUtils.cpp:1669] dumpGattServerEvent(): [status: ESP_GATT_OK, attr_handle: 42 0x2a, service_handle: 40 0x28, char_uuid: beb5483e-36e1-4688-b7f5-ea07361b26a8]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_ADD_CHAR_EVT
[D][BLECharacteristic.cpp:609] setHandle(): >> setHandle: handle=0x2a, characteristic uuid=beb5483e-36e1-4688-b7f5-ea07361b26a8
[D][BLECharacteristic.cpp:611] setHandle(): << setHandle
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_ADD_CHAR_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:135] executeCreate(): << executeCreate
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
D (1344) FreeRTOS: Semaphore taking: name: StartEvt (0x3ffdfcac), owner: <N/A> for start
D (1358) FreeRTOS: Semaphore taken:  name: StartEvt (0x3ffdfcac), owner: start
[D][BLEDevice.cpp:95] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_START_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_START_EVT
[D][BLEUtils.cpp:1805] dumpGattServerEvent(): [status: ESP_GATT_OK, service_handle: 0x28]
[D][BLEServer.cpp:176] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_START_EVT
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_START_EVT
[D][BLEService.cpp:151] start(): << start()
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEAdvertising.cpp:164] start(): >> start: customAdvData: 0, customScanResponseData: 0
[D][BLEServer.cpp:294] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEAdvertising.cpp:184] start(): - no services advertised
[D][BLEAdvertising.cpp:221] start(): << start
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT
Characteristic defined! Now you can read it in your phone!
[... cut ...]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RSP_DATA_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1285] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
[D][BLEUtils.cpp:1123] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:134] handleGAPEvent(): BLEServer ... handling GAP event!

mine seems quite same

chegewara, 2018-03-22:
Yes, everything looks good. There is small chance that esp32 is broken and radio does not work.
My advice is to run SimpleWiFiServer, i know you need ble but wifi and ble are using the same radio though, and if you have other bluetooth device that can you use as server or peripheral then try to run BLE_client on esp32. 
This shoul have answer question if  there is issue with ble.

lbalzs, 2018-03-22:
Thanks, that helped me to solve my problem. The problem was with my phone, the ESP32 was working fine, thanks for help!

lbalzs, 2018-03-22:
(and i actually did scan with hcitool instead of lescan, just writing if some1 else messes this up)

closed 2018-03-22 by lbalzs