nkolban/esp32-snippets #447: BLEEddystoneTLM.cpp doesn't compile without Arduino.h

thomasvdwege, 2018-03-22:
BLEEddystoneTLM.cpp doesn't compile when you don't have an Arduino environment on your pc. Could this be fixed? @pcbreflux made the file, I don't think the cpp-utils are supposed to break when you don't have Arduino installed.

For anyone encountering this issue, simply commenting the whole BLEEddystoneTLM.cpp file works.

chegewara, 2018-03-22:
Or just deleting it

nkolban, 2018-03-29:
I have pulled these files and moved them into the cpp_utils/onhold directory.   Thank you mr @thomasvdwege for reporting and sorry for the issue.  We should always have a clean build of cpp_utils without any trace of Arduino.   I also didn't find any documentation in the source to explain these source files.  I'm a huge fan of mr @pcbreflux  work and I hope he will come back and clarify the files.  If they are samples, then we can include them in some samples folder as opposed to being in the core build path of the BLE and cpp_utils library.

commit 5f0a1685 (2018-08-07): fixes #447
commit d04db847 (2018-08-07): Merge pull request #622 from Donderda/feature_eddystone

Fixes #447
closed 2018-08-07 by nkolban