nkolban/esp32-snippets #449: Make BLEClient::connect (BLEAddress address) to be a non blocking call

ihmgsm, 2018-03-26:
Hello,
it's a great library so I want to thank You, and I have a little proposal.

Make BLEClient::connect (BLEAddress address) to be a non-blocking call or at least provide a way to set a timeout. 

Thanks
Ivan

nkolban, 2018-03-27:
Studying ... the core function behind BLEClient::connect() is the API called esp_ble_gattc_open.  Checking  to see what its timeout properties may be.

ihmgsm, 2018-03-27:
Yes, if in time frame between scanning and connecting to the server, it disappears, or stop responding, or change it's address and then we try to connect to it, entire our app will 'hang', waiting for server to be back online. I know it's a rare situation, but it happened to me ,while developing and experimenting.

ihmgsm, 2018-03-27:
Hello again,
I think I found an answer to my question in following example explanation:
[https://github.com/espressif/esp-idf/blob/5ee7233b5bc09d7446c14cdda74868b491495cd2/examples/bluetooth/gatt_client/tutorial/gatt_client_example_walkthrough.md](url)

> Finally if the remote device name is the same as we have defined above, the local device stops scanning and tries to open a connection to the remote device using the esp_ble_gattc_open() function. This function takes as parameters the Application Profile GATT interface, the remote server address and a boolean value. The boolean value is used to indicate if the connection is done directly or if it’s done in the background (auto-connection), at the moment this boolean value must be set to true in order to establish the connection....

Thanks for help and if You find a way to solve this problem please let us know.

Ivan

harbaum, 2019-11-22:
Is there any progress or update on this?

This would be great especially in context with the esp32's ability to connect multiple devices. The first device can then already be used while the connection to the second on is in progress.

Is it e.g. possible to run the connect in a separate esp32 process or similar? Or implement a custom version of connect?

chegewara, 2019-11-22:
> Is it e.g. possible to run the connect in a separate esp32 process or similar? Or implement a custom version of connect?

You can try. When you create new BLEClient object it will have its own set of semaphores, thia may work as long as there is no limitation in hardware or bluedroid stack.

evanthomas, 2019-12-08:
I want to use an ESP32 as a BLE->WiFi gateway to handle multiple servers where the servers come and go a lot. I'd really like to see a non-blocking API.

rhoggarth, 2020-07-06:
esp32 emulating a WeMo smart switch.  Sending BLE reads and writes when on/off commands are received by the Wemo code.  The Wemo becomes unresponsive when the BLE code blocks.  Even when the BLE server is in range,  it can take 1050mS for connect to comeback.