nkolban/esp32-snippets #450: Simple task.h question

rweller59, 2018-03-26:
can I call task.stop() from within the task? Do I need to pass handle to task in order to do this? Do I even need to do this or can I just return from the task?

chegewara, 2018-03-26:
Yes, you can. I would say you even can call `vTaskDelete(NULL)` from inside that task, but this can cause memory leak, so myTask->stop() is enough.

rweller59, 2018-03-26:
thanks. so I need to pass the handle to the task?

chegewara, 2018-03-26:
No, all its required is handled by Task class. Here m_handleTask is filed with data:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/Task.cpp#L75
so if you start task before you try to stop it then m_handleTask wont be nullptr and such this line will be executed:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/Task.cpp#L90

So, all you have to do somhere in your code is:
- create task (Task* myTask = Task())
- start task (myTask->start())
- stop task when its no longer needed (myTask->stop())
- again start task if required (myTask->start())

rweller59, 2018-03-26:
yes, get that. What I am not clear on is whether or not the task has a handle to itself?

What I'm doing is starting the task, task executes code and when it has completed its routine, deletes itself - the code that started the task doesn't care about the task after task->start().

chegewara, 2018-03-26:
myTask has handle to itself, and like i said its this line:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/Task.cpp#L75

to delete task you need to execute myTask->stop() somewhere or myTask will stay in loop

rweller59, 2018-03-26:
got it. thanks

closed 2018-03-29 by rweller59