nkolban/esp32-snippets #451: How to Indicate BLE 'Service Changed' (0x2A05)

mainr, 2018-03-26:
Under BLE standard 'Generic Attribute' (0x1801) there is a Characteristic 'Service Changed' (0x2A05) with 'indicate' property, which is supposed to be used to indicate that a range of handles in the BLE device have changed, so that they can be 're-discovered', updating any cached information. 

How would I go about triggering this? Does the client need to 'subscribe' to get the indication, or is this automatic (there is no 0x2902 descriptor to write to)?

I can't find this information in the ESP-IDF documentation, either.

chegewara, 2018-03-26:
Hi,
i also could not find any info about Generic attributes 0x1800 and 0x1801 in esp-idf documentation and source files. Those attributes with its characteristics are created in lower level API. My suggestion is to ask this question on esp-idf repo.

mainr, 2018-03-26:
Opened issue [#1777](https://github.com/espressif/esp-idf/issues/1777) with esp-idf.

Thanks @chegewara .

closed 2018-03-26 by mainr

Sladerix, 2023-03-05:
> Opened issue [#1777](https://github.com/espressif/esp-idf/issues/1777) with esp-idf.
> 
> Thanks @chegewara .

Hi @mainr did you solved your issue?
Do you still have the code in Arduino Framework (not esp-idf)?