nkolban/esp32-snippets #454: JSON assertion error

rweller59, 2018-03-29:
I'm using json.cpp in code moving config parameters received via BLE into structure variable. Tonight I updated cpp_utils, espidf and toolchain. After compiling and loading code, nvs parameters were messed up. My routine for updating parameters (have changed in weeks) is now failing on what seems like very simple code...
* logs correct data as expected
    	ESP_LOGI(TAG, "transformJSON SET_NETWORK 1 %d, %s",obj.getString("hubIP").length(), obj.getString("hubIP").c_str());
* when uncommented, I get the error list below
    	//config::network.hubIP = obj.getString("hubIP");
* when uncommented, code is executed and then fails after the next log. this continues with each object
    	//config::network.hubIP = "xx.xx.xx.x";
    	ESP_LOGI(TAG, "transformJSON SET_NETWORK 2");
    	config::network.hubPwd = obj.getString("hubPwd");

CONSOLE ERROR WHEN SETTING STRUCT VAR TO JSON OBJECT.
I (49073) transformJSON: transformJSON SET_NETWORK 1 10, xx.xx.xx.x
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/home/rick/esp/esp-idf/components/heap/./heap_caps.c", line 261, function: heap_caps_free
abort() was called at PC 0x400df34b on core 0
0x400df34b: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x40090f44:0x3ffe7730 0x4009111b:0x3ffe7750 0x400df34b:0x3ffe7770 0x400829e3:0x3ffe77a0 0x40082dd9:0x3ffe77c0 0x4000bec7:0x3ffe77e0 0x400f3d95:0x3ffe7800 0x400da22a:0x3ffe7820 0x400d6d65:0x3ffe7f30 0x400d6f37:0x3ffe7fe0 0x400dd66d:0x3ffe8020 0x400dd76b:0x3ffe80e0 0x400d693f:0x3ffe8120 0x400d465c:0x3ffe8140 0x4015ffca:0x3ffe8160
0x40090f44: invoke_abort at /home/rick/esp/esp-idf/components/esp32/./panic.c:648

0x4009111b: abort at /home/rick/esp/esp-idf/components/esp32/./panic.c:648

0x400df34b: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x400829e3: heap_caps_free at /home/rick/esp/esp-idf/components/heap/./heap_caps.c:123

0x40082dd9: _free_r at /home/rick/esp/esp-idf/components/newlib/./syscalls.c:42

0x400f3d95: operator delete(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc:46

0x400da22a: __gnu_cxx::new_allocator<char>::deallocate(char*, unsigned int) at /home/rick/eclipse-test/OldOrb/main/./transformJSON.cpp:382
 (inlined by) std::allocator_traits<std::allocator<char> >::deallocate(std::allocator<char>&, char*, unsigned int) at /home/rick/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/alloc_traits.h:386
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_destroy(unsigned int) at /home/rick/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/basic_string.h:185
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_dispose() at /home/rick/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/basic_string.h:180
 (inlined by) std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string() at /home/rick/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/basic_string.h:544
 (inlined by) transformJSON::setSettings(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /home/rick/eclipse-test/OldOrb/main/./transformJSON.cpp:196

nkolban, 2018-03-29:
Can you show me the source statements at the failing line which appears to be "transformJSON.cpp" line 196?  Make sure that you identify which source line is which.  Ideally attach the whole source file that is failing indicating the error for that exact source file.

rweller59, 2018-03-30:
Here is a portion of the file - its a fairly long file but repetitive. case SET_NETWORK: is the portion of the code I'm trying to troubleshoot but the issue seems consistent  in similar portions of the code. As you can see,  I've commented out the lines where I am using obj.getString(""); When I do this, code doesn't fail. When I uncomment any of these lines, I get the failure. Below the code is the failure log I'm getting whenever I uncommented 'setup.network.webPwd = obj.getString("webPwd");'

// Update configuration settings with payload received from network
bool transformJSON::setSettings(int topic_code, const std::string& payload) {
	ESP_LOGI(TAG, "transformJSON::setSettings: %d, %s",topic_code, payload.c_str());
	JSON json;
	JsonObject obj = json.createObject();
	obj = json.parseObject(std::string(payload));
	if(!obj.isValid()) {
		json.deleteObject(obj);
		return false; }
	std::string str;
	bool x;

    switch(topic_code) {
    case SET_DEVICE:
    	setup.device.id = obj.getString("id");
    	setup.device.name = obj.getString("name");
    	setup.device.pwd = obj.getString("pwd");
    	setup.device.network = obj.getBoolean("network");
    	setup.device.hubMonitor = obj.getBoolean("hubMonitor");
    	setup.device.webMonitor = obj.getBoolean("webMonitor");
    	setup.device.hubUpdate = obj.getBoolean("hubUpdate");
    	setup.device.webUpdate = obj.getBoolean("webUpdate");
    	setup.device.timeLogger = obj.getBoolean("timeLogger");
    	setup.device.activeLogger = obj.getBoolean("activeLogger");
    	break;

    case SET_NETWORK:
    	ESP_LOGI(TAG, "transformJSON::setSettings: %d", heap_caps_check_integrity_all(true));
    	//setup.network.hubIP = obj.getString("hubIP");
    	ESP_LOGI(TAG, "transformJSON::setSettings 1");
    	//setup.network.hubPwd = obj.getString("hubPwd");
    	ESP_LOGI(TAG, "transformJSON::setSettings 2");
    	setup.network.hubPort = obj.getInt("hubPort");
    	ESP_LOGI(TAG, "transformJSON::setSettings 3");
    	//setup.network.webIP = obj.getString("webIP");
    	ESP_LOGI(TAG, "transformJSON::setSettings 4");
    	//setup.network.webPwd = obj.getString("webPwd");
    	ESP_LOGI(TAG, "transformJSON::setSettings 5");
    	setup.network.webPort = obj.getInt("webPort");
    	ESP_LOGI(TAG, "transformJSON::setSettings 6");
    	for(size_t i=0; i<NUM_WAP; i++) {
        	if(i < obj.getArray("wifi").size()) {
        		setup.network.wifi[i].ssid = obj.getArray("wifi").getObject(i).getString("ssid");
        		setup.network.wifi[i].pwd = obj.getArray("wifi").getObject(i).getString("pwd");
        	}
        }
        break;

CONSOLE OUTPUT
I (51091) transformJSON: transformJSON::setSettings: 22, {"hubIP":"fgggg","hubPort":3011,"hubPwd":"hggg","webIP":"","webPort":1883,"wifi":[]}
I (51104) transformJSON: transformJSON::setSettings: 1
I (51108) transformJSON: transformJSON::setSettings 1
I (51113) transformJSON: transformJSON::setSettings 2
I (51118) transformJSON: transformJSON::setSettings 3
I (51123) transformJSON: transformJSON::setSettings 4
Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x401612ac  PS      : 0x00060930  A0      : 0x800da2ed  A1      : 0x3fff18b0  
0x401612ac: JsonObject::getString(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) at /home/rick/eclipse-test/OldOrb/components/cpp_utils/./JSON.cpp:287

A2      : 0x3fff1d48  A3      : 0x3fff1ec8  A4      : 0x3fff1d30  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x00000005  A8      : 0x801612ac  A9      : 0x3fff1890  
A10     : 0x00000000  A11     : 0x3fff1d38  A12     : 0x00000006  A13     : 0x3fff1d3e  
A14     : 0x00006477  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000010  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x401612ac:0x3fff18b0 0x400da2ea:0x3fff18d0 0x400d6db9:0x3fff1f40 0x400d6f8b:0x3fff1ff0 0x400dd561:0x3fff2030 0x400dd65f:0x3fff20f0 0x400d6993:0x3fff2130 0x400d46c8:0x3fff2150 0x4015fed2:0x3fff2170
0x401612ac: JsonObject::getString(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) at /home/rick/eclipse-test/OldOrb/components/cpp_utils/./JSON.cpp:287

0x400da2ea: transformJSON::setSettings(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /home/rick/eclipse-test/OldOrb/main/./transformJSON.cpp:392

0x400d6db9: manageTopic(int, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /home/rick/eclipse-test/OldOrb/main/./CommService.cpp:204

0x400d6f8b: commService::receivedBle(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) at /home/rick/eclipse-test/OldOrb/main/./CommService.cpp:109

0x400dd561: bleReceive(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) at /home/rick/eclipse-test/OldOrb/main/./bleService.cpp:209 (discriminator 1)

0x400dd65f: bleService::processRequest() at /home/rick/eclipse-test/OldOrb/main/./bleService.cpp:86

0x400d6993: commService::bleProcessRequest() at /home/rick/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/ext/new_allocator.h:110

0x400d46c8: Task_comm_monitor::run(void*) at /home/rick/eclipse-test/OldOrb/main/./main.cpp:124

0x4015fed2: Task::runTask(void*) at /home/rick/eclipse-test/OldOrb/components/cpp_utils/./Task.cpp:110

rweller59, 2018-03-30:
fyi - I opened an issue on the esp32 forum and am basically having the same conversation with Angus

nkolban, 2018-03-30:
I examined the code for the JSON method `getString()` and found that it would create a null reference exception if the member we were trying to get was not present.

For example, for the JSON:

```
{
   "a": "aValue";
}
```

From the above,

```
myObj.getString("a");
```
would work but

```
myObj.getString("b");
```

would result in a null pointer reference exception.

I have placed a guard on it now so that it returns an empty string on an attempt to get a field that doesn't exist.  I'm not sure I like that semantic and want to gauge whether we should throw an exception instead?

We also have the method `hasItem()` which could be called before accessing a member to determine if the member exists ... for example:

```
if (myObj.hasItem("b")) {
   myB = myObj.getString("b");
}
```

I'm not saying that is your problem but it would do no harm to test this out.  Consider running the "toString()" method on the JSON object to validate that the items you expect to be in the JSON object are in fact present.

rweller59, 2018-03-30:
put in checks as suggested. code runs until setting the struct variable to obj.getString then fails...
CODE:
    case SET_NETWORK:
        ESP_LOGI(TAG, "getFreeHeapSize %d", System::getFreeHeapSize());
    	ESP_LOGI(TAG, "SET_NETWORK %d", heap_caps_check_integrity_all(true));
    	ESP_LOGI(TAG, "SET_NETWORK hubIP - %s", obj.getString("hubIP").c_str());
    	ESP_LOGI(TAG, "SET_NETWORK asItem - %d", obj.hasItem("hubIP"));
    	ESP_LOGI(TAG, "SET_NETWORK toString - %s", obj.toString().c_str());
    	str = obj.getString("hubIP");
    	setup.network.hubIP = "this is a test IP";
    	ESP_LOGI(TAG, "transformJSON::setSettings 0");
    	setup.network.hubIP = obj.getString("hubIP");

CONSOLE OUTPUT:
I (44088) transformJSON: getFreeHeapSize 51116
I (40116) transformJSON: SET_NETWORK 1
I (40118) transformJSON: SET_NETWORK hubIP - ffggg
I (40123) transformJSON: SET_NETWORK asItem - 1
I (40127) transformJSON: SET_NETWORK toString - {
	"hubIP":	"ffggg",
	"hubPort":	3011,
	"hubPwd":	"hh",
	"webIP":	"",
	"webPort":	1883,
	"wifi":	[]
}
I (40141) transformJSON: transformJSON::setSettings 0
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/home/rick/esp/esp-idf/components/heap/./heap_caps.c", line 261, function: heap_caps_free

nkolban, 2018-03-30:
The heap management of the ESP32 has some excellent debugging built into it.  Unfortunately it requires preparation of the application as well as injection of test points into your code.  This can be found here:

http://esp-idf.readthedocs.io/en/latest/api-reference/system/heap_debug.html

If it were me, an initial test I would run would be to hard code a string at the start of the function that contains an example of the JSON you are passing in.  That way we would have eliminated the notion that (perhaps) the data being passed in is being "changed" underneath the JSON processing.

rweller59, 2018-03-30:
Couple of observations. I had my setup structure defined in another cpp (stores.cpp) and declared extern in TransformJSON.cpp - this was the situation when getting the crash. I defined a new structure variable in transformJSON.cpp and did not get the failure.

I am loading the setup struct from nvs, checking for saved version number and variable length. Both of these checks were saying I was retrieving valid nvs data, however all of the struct's string variables were being set to arrays of '\0' (str length = length of originally saved data) in the nvs read although str.length() was returning >0.

I changed the settings version # so that my nvs read routine would not load from nvs, and everything works again.

Tried to mimic this by setting my struct str = "\x00\x00\x00"; then setting it to the json objString where the failure was occurring but that did not cause any failure.

I'm guessing this doesn't have anything to do with json and the heap corruption is initiated when I load struct from nvs.

nkolban, 2018-03-30:
Howdy sir.  I'm afraid you lost me in the weeds of your own application.  However that is ok ... and when I'm describing my own code I too go deep and get excited.  Is the summation of the story that you have now found the problem and the issue is closed?  If not, let me know and we'll circle back ... but I think you are saying that for reasons beknownst to your own app, you might have been passing in "odd" data into the JSON processor?

rweller59, 2018-03-30:
Sorry for the 'weeds'. In summary, flashing new code to the chip seems to corrupt parameters saved to nvs. Somehow, setting one of these corrupted variables equal to the json obj.toString() variable causes a heap issue. I will set up a generic scenario outside of my code and if the same pattern repeats, I will let you know. Let's leave it open for a couple of days until I have time to set that up.

rweller59, 2018-04-11:
Neil,
I attempted to re-create this without luck. The issue has something to do with corrupting nvs data when uploading a new program. I tried to write a simple program to re-create but can't re-create the nvs corruption. If I see it again and can capture the issue on a repeatable basis, I will re-open the issue.

closed 2018-04-11 by rweller59