nkolban/esp32-snippets #456: BLE Service not found

Brunez3BD, 2018-03-30:
I'm using the [BLE_client example](https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/examples/BLE_client/BLE_client.ino) to connect to a BLE server with multiple services, but with no success. 
I understood that it search for a server that has as first service the one specified in `static BLEUUID serviceUUID("mySercieUUID");`, so I tried to slightly modify the code. The problem is that a call to `advertisedDevice.isAdvertisingService(serviceUUID)` return false every time, but I'm sure that the service that I'm looking for exist because with the nRF Connect app I can see it. I've modified the `onResult()` function like this:
```
void onResult(BLEAdvertisedDevice advertisedDevice) {
    Serial.print("BLE Advertised Device found: ");
    Serial.println(advertisedDevice.toString().c_str());

    if(advertisedDevice.isAdvertisingService(serviceUUID)){
        Serial.println("my Service found!");
    } else {
        Serial.println("maybe next time, loser");
    }

    // We have found a device, let us now see if it contains the service we are looking for.
    if (advertisedDevice.haveServiceUUID() && advertisedDevice.getServiceUUID().equals(serviceUUID)) {

      // 
      Serial.print("Found our device!  address: "); 
      advertisedDevice.getScan()->stop();

      pServerAddress = new BLEAddress(advertisedDevice.getAddress());
      doConnect = true;

    } // Found our server
  } // onResult
```

without success. Someone can help me? Thank you very much

chegewara, 2018-03-30:
Hi,
when you scan with nRF connect you should have to see something similar to this screen:
https://ibin.co/3lNX93QvWwn3.png
In your case is not important Service data: UUID, but what is above:
`Incomplete List of 16-bit Service UUID-s`
You can find Incomplete or complete List of 16, 32 or 128-bits UUID. This mean that service is advertised. If you dont have any of them this mean service is not advertised and its just created when you connect to peripheral/server.

Brunez3BD, 2018-03-30:
Thank you so much! The problem was right with a Flower Care sensor :) 
I was able to connect to the service with the sensor data but I was not finding it with `advertisedDevice.isAdvertisingService(serviceUUID)`. 
I've some other problems with flower care sensor, can I ask a few question here or I have to open another issue?

chegewara, 2018-03-30:
Read this and you can ask in this issue, even if its closed now:
https://github.com/nkolban/esp32-snippets/issues/274

closed 2018-03-30 by Brunez3BD