nkolban/esp32-snippets #459: What are setName and setShortName supposed to do?

jrickard, 2018-04-01:
BLEDevice::init("Jack's Server4"); 
BLEServer* pJackServer = BLEDevice::createServer();
BLEAdvertising* pJackAdvertising = pJackServer->getAdvertising();
BLEAdvertisementData jacksAdData;
jacksAdData.setFlags(ESP_BLE_ADV_FLAG_GEN_DISC | ESP_BLE_ADV_FLAG_BREDR_NOT_SPT);
jacksAdData.setName("2nd Server");
jacksAdData.setShortName("4th Server");
pJackAdvertising->setAdvertisementData(jacksAdData);
pJackAdvertising->start(); 

Here I initialize a device as "Jack's Server4"

Later, I use advertisement data to set the name and shortname in the advertising data.  What are these supposed to accomplish????

No matter what I enter, the device always shows up as Jack's Server4

So what are they for?

S-March, 2018-04-02:
My guess is due to your total advertising packet being longer than 32 bytes. Try reducing the overall size of your advert packet and use different name/short name and see what that does.

jrickard, 2018-04-03:
I got it. Names, short or long in advertisement packet data units have nothing to do with the name of the device.  It is just advertising data.

closed 2018-04-03 by jrickard