nkolban/esp32-snippets #463: Issues building 0.4.10 on Platformio

nsamala, 2018-04-05:
I get the following errors when trying to compile on Platformio

```
.piolibdeps\ESP32 BLE Arduino_ID1841\src\BLEClient.cpp: In member function 'bool BLEClient::connect(BLEAddress)':
.piolibdeps\ESP32 BLE Arduino_ID1841\src\BLEClient.cpp:115:2: error: too many arguments to function 'esp_err_t esp_ble_gattc_open(esp_gatt_if_t, uint8_t*, bool)'
);
^
In file included from .piolibdeps\ESP32 BLE Arduino_ID1841\src\BLEClient.cpp:13:0:
C:\Users\Nishanth\.platformio\packages\framework-arduinoespressif32\tools\sdk\include\bluedroid/esp_gattc_api.h:293:11: note: declared here
esp_err_t esp_ble_gattc_open(esp_gatt_if_t gattc_if, esp_bd_addr_t remote_bda, bool is_direct);
^
*** [.pioenvs\featheresp32\lib084\ESP32 BLE Arduino_ID1841\BLEClient.cpp.o] Error 1
.piolibdeps\ESP32 BLE Arduino_ID1841\src\BLEDescriptor.cpp: In member function 'void BLEDescriptor::handleGATTServerEvent(esp_gatts_cb_event_t, esp_gatt_if_t, esp_ble_gatts_cb_param_t*)':
.piolibdeps\ESP32 BLE Arduino_ID1841\src\BLEDescriptor.cpp:158:53: error: 'struct esp_ble_gatts_cb_param_t::gatts_add_char_descr_evt_param' has no member named 'descr_uuid'
m_bleUUID.equals(BLEUUID(param->add_char_descr.descr_uuid)) &&
^
*** [.pioenvs\featheresp32\lib084\ESP32 BLE Arduino_ID1841\BLEDescriptor.cpp.o] Error 1
```

My environment is configured as:
```
[env:featheresp32]
platform = espressif32
board = featheresp32
framework = arduino
build_flags = -fexceptions
lib_deps = 
  ESP32 BLE Arduino@0.4.10
```

nsamala, 2018-04-05:
The first issue appears to be an API change with the Espressif32 platform.

The second was fixed as follows. Change
```c++
if (m_pCharacteristic != nullptr &&
					m_bleUUID.equals(BLEUUID(param->add_char_descr.descr_uuid)) &&
					m_pCharacteristic->getService()->getHandle() == param->add_char_descr.service_handle &&
					m_pCharacteristic == m_pCharacteristic->getService()->getLastCreatedCharacteristic()) {
				setHandle(param->add_char_descr.attr_handle);
				m_semaphoreCreateEvt.give();
			}
```

to 

```c++
if (m_pCharacteristic != nullptr &&
					m_bleUUID.equals(BLEUUID(param->add_char_descr.char_uuid)) &&
					m_pCharacteristic->getService()->getHandle() == param->add_char_descr.service_handle &&
					m_pCharacteristic == m_pCharacteristic->getService()->getLastCreatedCharacteristic()) {
				setHandle(param->add_char_descr.attr_handle);
				m_semaphoreCreateEvt.give();
			}
```

chegewara, 2018-04-05:
Looks like both errors are caused by esp API changes that were introduced lately but looks like are not in latest platformio library.

https://github.com/espressif/esp-idf/blob/aca000869406e35eba0d5fd7064304ba9e5acead/components/bt/bluedroid/api/include/esp_gattc_api.h#L294

https://github.com/espressif/esp-idf/blob/cd54a35cae290df7def0207db7031644d9729ce4/components/bt/bluedroid/api/include/esp_gatts_api.h#L161-L166

My suggestion is to use older cpp_utils library, 0.4.9 or 0.4.8.

closed 2024-06-27 by outlandnish