nkolban/esp32-snippets #465: Is it possible to re-set notifications on (re-)connect on 2902?

posics, 2018-04-07:
I have noticed that my nrfConnect sends stuff like gat.setCharacteristicNotification(6e..., true) whenever it connects to my esp32 based on whether I "subscribed" to notifications on particular (2902) characteristic or not.
This actually **works IF I connect, subscribe to some notifications, disconnect, then reconnect** (*) and it does indeed re-subscribe to all notifications and everything is ok, it starts receiving them from esp32.

BUT!

It does **NOT work if I re-upload (new) firmware and/or reboot esp32 and it re-initialises everything** including BLE stuff. Then when I (for me, but not for esp32) RE-connect to it, nothing, nada. No notifications. Not until I go into nrfConnect, UNsubscribe and REsubscribe to each of the descriptors I wish to be notified of.
But then looking at nrfConnect, it duly sends all subscription information when connecting. How can we extend the BLEDescriptorCallbacks or BLECharacteristicCallbacks (I presume this is where it would be done?) to accommodate this "initial"/"reconnect" behaviour, i.e. notification of subscribe/unsubscribe events to a particular 2902? I don't see it as a game-changer but for completeness it would be nice to be able to figure out which notifications the client wishes to subscribe to on connect first time and every time.

Thanks!

*took me a LOOOONG time to figure this one out, esp32 would just not respond to subsequent reconnection attempts, it had to do with my tasks. Picky.

chegewara, 2018-04-07:
Subscribing to notifications/indications is client job, so you can write your own android client app that will subscribe to notifications every time it connect. With android is one issue that not always disconnect message is sent, thats why it works this way for you, esp32 just does not know nRF was disconnected.

posics, 2018-04-07:
Can you please re-read the question again, slowly this time. What I am describing is the exact opposite. nrfConnect behaves propertly. esp32 also behaves properly, but not on FIRST connect. Only on subsequent re-connects.
nrfConnect presents all information to esp32 on what it wants but esp32 doesn't react the first time, only later.

chegewara, 2018-04-07:
In my case all works fine. I have HID keyboard example, it will works properly only if client during connection/pairing subscribe to 1(in real 2 characteristics notifications, because battery also) characteristic notifications. 
No matter what client i am using every time on first and every connection/pairing notifications are subscribed properly.

If you dont like my answers, im fine with that and i dont need to re-read your questions. I did not have any issue with notifications till now.