nkolban/esp32-snippets #469: Is it possible to connect to multiple BLE servers - what are the limitations

johnhmacleod, 2018-04-10:
I have created a gateway that connects to a TI SensorTag & forwards the data to an MQTT Broker (IBM Watson IoT Platform).  It works nicely, provided I find my BLE device before attempting to connect to WiFi!

My question is, is it possible & supported to connect to multiple BLE devices with this library (and its dependencies)?  What are the considerations & limitations?

chegewara, 2018-04-11:
Im not sure i understand. You want to create few esp32 clients that will read from one TI sensor?

chegewara, 2018-04-11:
Ok, in theory it can be done. You need to try to connect esp32 with 2 devices by creating 1 BLEClient object for each device, if you want to connect to it simultanously or connect to device, disconnect, connect to next one, disconnect and so on.

arsab, 2018-04-23:
I tried with Heart Rate (service 0x180D ,  characteristic 0x2A37) + 
         Cycling Cadence (service 0x1816 ,  characteristic 0x2A5B)

Can connect both 
->registerForNotify (...)  both
ask notification request (0x2902) to both.
Unfortunately  receive notification only from Cadence or none.
If one of client is not present to connect the other works smoothly.

According to following exchange it was a limitation which should have been resolved with v3.0
but something missing on Arduino?
https://esp32.com/viewtopic.php?t=954

Anything possible Neil?

Thanks for your help

nsamala, 2018-05-24:
@johnhmacleod @arsab I created this PR that gives you support for multiple servers. I've tested it with two TI SensorTag's simultaneously.

https://github.com/nkolban/ESP32_BLE_Arduino/pull/13

arsab, 2018-05-31:
Thanks nsamala, look very promising,  now that this ble implementation is the reference for ESP32 Arduino: https://github.com/espressif/arduino-esp32/tree/master/libraries
We might need to clarify few points for Neil Kolban:
- Backward compatibility for whom did develop unique Server: e.g. can we still call createClient();  without parameter, or always have a parameter, update code in order to recompile.
- Have a simple example e.g. create "client"  for each server then refer to it, register each,... What are the rules for appId, 0-n?

arsab, 2018-05-31:
You also made nice refactoring with BLENotifier which should also go in coding rules ;-)

nsamala, 2018-06-04:
@arsab Definitely, in this example, I'm going to show how I use the `BLENotifier` class to receive accelerometer data from the TI SensorTag

```cpp
#include <BLEDevice.h>

static const uint16_t appId = 3;  // arbitratrily chosen app ID. client default is 1 and server default is 2
static const BLEUUID serviceUUID("f000aa80-0451-4000-b000-000000000000");
static const BLEUUID accelDataUUID("f000aa81-0451-4000-b000-000000000000");

class AccelerometerNotificationReceiver : public BLENotifier {
  public:
    AccelerometerNotificationReceiver(BLEAddress address) {
      BLEClient client = BLEDevice::createClient(appId);
      client->connect(address);

      // get service from TI SensorTag
      BLEService service = client->getService();
      if (service == nullptr) {
        Serial.println("Could not find service UUID");
        return;
      }
      Serial.println("Found service");

      // get the data characteristic
      BLERemoteCharacteristic *dataCharacteristic = service->getCharacteristic(accelDataUUID);
      if (dataCharacteristic == nullptr) {
        Serial.println("Could not find data characteristic");
        return;
      }
      logInfo(LABEL, "Got data characteristic");

      // register for notifications
      dataCharacteristic->registerForNotify(this);

      // enable notifications
      const uint8_t v[] = { 0x1, 0x0 };
      dataCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)v, 2, false);
    }

    void onNotify(BLERemoteCharacteristic* pBLERemoteCharacteristic, uint8_t* pData, size_t length, bool isNotify) {
      // notification received
      Serial.println("Received accelerometer data");
    }
}
```

arsab, 2018-06-06:
Thanks,
Maybe I am not doing all right but have difficulties to compile:
Had to make few changes:
in BLEClient.
  public:
        BLEClient(uint16_t appId);    // replace by
//	BLEClient();                            // removed

in BLEDevice.cpp
//  BLEClient* BLEDevice::m_pClient = nullptr;    // needed to comment out in line 42 since removed in BLEDevice.h

in one hand we had 
in BLERemoteCharacteristic.cpp
   toNotify->onData
  here we declared "void onNotify"     realigned onData vs onNotify   ;-)

Also I need to have two different kind of servers:

after having cleaned elements above I continue to have  compiler error 
   error: expected class-name before '{' token 
class myHRBLENotificationReceiver : public BLENotifier {
                                                                                               ^

nsamala, 2018-06-13:
Sorry about that! I'll double check this evening. I may have had some issues when I copied it over from my repo for ESP32_BLE_Arduino that I missed.

gcharles81, 2018-07-06:
hello is there a simple example code that can compile using this appid , I have 2 servers that I would like to connect to with one client

gcharles81, 2018-08-14:
@arsab hello have you managed to fix the compile error and connect to two servers thanks

nsamala, 2018-09-11:
@gcharles81 I've gone ahead and fixed the compiler errors and pulled the latest code from the upstream. Give it a shot now

sagarvegg, 2019-06-08:
I have face problem with BLE server multi connect
only able to connect up to four device to single esp32 wrover 
I change sdkconfig "CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN" = 9 as well "CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN__EFF" = 9 
and also changed "CONFIG_BT_ACL_CONNECTIONS" = 7

but don't connect more than 4 device

after resetting above sdkconfig parameters only connect 3 device to esp32 wrover

if any one have solution than respond ...

davejel, 2019-10-27:
hi @outlandnish 
i was trying to use your example for multiconnect with BLENotifier but i get the error mentioned above:  expected class-name before '{' token
does this mean my library is out of date, or do i need to edit the library, if so can you tell me how.
i have checked for updates but there doesnt seem to be any
thanks in advance
Dave

sousajoao07, 2020-06-29:
@outlandnish 
I noticed that in your repository your Device.cpp file is updated, allowing you to connect more than one server. But when I replace the file that exists in the Arduino library it gives an error. I would like to be able to connect 2 servers to 1 client using Arduino IDE. It's possible?

outlandnish, 2020-06-30:
@sousajoao07 it's possible to connect to multiple servers using a single client. The error you're getting is the result of merging in upstream changes to my code without finishing the merge properly. Since my needs changed, I no longer have the time to support that fork. You're more than welcome to fix it and submit a PR!

That said, have you considered the NimBLE additions to the ESP32? It should give you support for multiple servers without hacking together old code on the Bluedroid implementation.