nkolban/esp32-snippets #470: BLE save known devices an last seen

lauwie007, 2018-04-11:
I want to scan Bluetooth and differentiate every device in categories (new, last seen <1 week, last seen <2 weeks, last seen <4 weeks). So, the results should be the number of devices per category (and this will be send over LoRa). Categories can (and probably will change, but this will be my starting point)

I never had lessons programming and I’m googling everything together so sorry if my code is a bit ‘dirty’. My code is pasted below, but I think I'll need to start over for this concept.

pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks()) saves every device address (for as far I know). I want to search known device addresses. If the device is know I want to add the day it was seen, if the device address is unkown I want to add it to the list. This will be used to category’s the devices. Where are the devices saved and how could I make a sort of table of device addresses with last seen in days?

I’m lost in all the code and curious if anyone could help me. I deleted a lot of unnecessary code (LoRa code) for this question to keep it simple. All help is welcome.

`/*
version:        2.3
Last updated:   Working on
What does it do?
Sends LoRa msg
Scans BLE 3 times
Calculates average BLE devices
Sends Runtime + count average BLE devices + count last average BLE devices over LoRa 
repeat
*/
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>
// ===================================Settings==================================================
unsigned int Data1 = 15; //Send LoRa and measure BLE every x minutes
// =============================================================================================
//BLE
int scanTime = (Data1*60)/3; //BLE scanTime in seconds 
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
    }
};
//timers
#include <elapsedMillis.h>
elapsedMillis timer1; //pauze na BLE scan
elapsedMillis timer2; //run time
long Rday = 86400000;
long Rhour = 3600000;
long Rminute = 60000;
// =============================================================================================
//
//
// Variables
float BLE_Total_scan1 = 0;
float BLE_Total_scan2 = 0;
float BLE_Total_scan3 = 0;
float BLE_Total_Average = 0;
unsigned int BLE_Total = 0;
unsigned int BLE_Total_last = 0;
unsigned int BLE_Scan_count = 0;
// =============================================================================================
void setup() {

BLEDevice::init("");
timer2 = 0; //runtime
timer1 = 0; //Loop timer
}

// =============================================================================================
void loop() {
LoRa_send(); 
if (timer1 > 10000){
BLE_scan(); 
BLE_scan(); 
BLE_scan(); 
BLE_Total_Average = (BLE_Total_scan1 + BLE_Total_scan2 + BLE_Total_scan3) / 3;
BLE_Total = round(BLE_Total_Average);
int Rdays = timer2 / Rday ;                                
int Rhours = (timer2 % Rday) / Rhour;                       
int Rminutes = ((timer2 % Rday) % Rhour) / Rminute ; //calculates runtime (Rdays, Rhours, Rminutes)
msg1[0] = Rdays;
msg1[1] = Rhours;
msg1[2] = Rminutes;  
msg1[3] = BLE_Total; 
msg1[4] = BLE_Total_last;
msg1[5] = 0;
BLE_Total_last = BLE_Total;
timer1 = 0;
Serial.print("[BLE]    BLE_Total_Average: ") && Serial.println(BLE_Total_Average);
Serial.print("[BLE]    BLE_Total: ") && Serial.println(BLE_Total);
Serial.println("------------------------End of timer loop------------------------");
Serial.flush();
  }
}

void LoRa_send(){  
}

void BLE_scan(){
  Serial.println("[BLE]    BLE_scan");
  BLEScan* pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks()); //Causes ram leak? Every new adress is saved
//    MyAdvertisedDeviceCallbacks myCallbacks;
//    pBLEScan->setAdvertisedDeviceCallbacks(&myCallbacks);
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  BLEScanResults foundDevices = pBLEScan->start(scanTime);
  if (BLE_Scan_count == 0){
     BLE_Total_scan1 = foundDevices.getCount();
     BLE_Scan_count = 1;
     Serial.print("[BLE]    BLE_scan1 Devices found: ");
     Serial.println(BLE_Total_scan1);
  }
  else if (BLE_Scan_count == 1){
     BLE_Total_scan2 = foundDevices.getCount();
     BLE_Scan_count = 2;
     Serial.print("[BLE]    BLE_scan2 Devices found: ");
     Serial.println(BLE_Total_scan2);
  }
  else if (BLE_Scan_count == 2){
     BLE_Total_scan3 = foundDevices.getCount();
     BLE_Scan_count = 0;
     Serial.print("[BLE]    BLE_scan3 Devices found: ");
     Serial.println(BLE_Total_scan3);
  }
}
`

chegewara, 2018-04-11:
You have to consider few things. 
- how many devices you are predicting to keep tracking
- you want to store tracked devices data in NVS or flash, or maybe just keep it in RAM
- how do you want to track devices with private MAC address
- maybe instead store data about devices just scan with esp32 and send data to "central" device with more storage and processing power that will store, filter and do other things with data

Here you have useful code snippet:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEScan.cpp#L95-L104

lauwie007, 2018-04-11:
Thank you for your response. For the first test it will be places around a small park. 

- I'm predicting a max of ±30 devices on one moment and max 80 known devices.
- Power is not a issue so RAM would be sufficient for now.
- I have not thought about devices with a private MAC address (I am aware of this 'issue'). I'll have to think about a solution at a later moment
- The data can not be send since LoRa does not offer sufficient bandwidth for large amounts of data. Also the ESP32 is a very powerful chip, right?

Thank you for that snippet! I will check it out

chegewara, 2018-04-11:
In this case i would think about ESP32-WROVER, with 4-16MB of PSRAM. Thats why:
- you could store more data in RAM
- i dont know how, but im pretty sure its possible to to recognize device with private MAC even if MAC will change over time; to get more info you will have to find someone with deeper knowledge about bluetooth
- i think that you will need to store more than just MAC address to track devices with private MAC, so with extra RAM you could store whole device structure and not just address

This may be beyond my knowledge, but this is how i would start with it:
- here you can see usage of vector https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEScan.cpp#L120 , i would create another class(structure) that will have advertised device and another field with timestamp (or just date)
- my understanding is that push_back will add data to end of vector, then maybe there is pop() function that will pop first record only if its older than (week, 2 weeks and so on)  and then push it to another vector
- about storing data in RAM, power supply is one thing that can cause device to reset, but there may be much more causes that you can even not think about: this library bug, esp-idf bugyour code bug, and a lot more, so storing data (at least some part of it, to not lose all) should be considered at some point (just saying)

lauwie007, 2018-04-12:
If it is beyond your knowledge it seems almost impossible for me. I'm a student urban planning.

- this is a small project within one larger project (this is my first collaboration and largest project I've ever done). The setup now gives mee (just) 600KB free memory to work with (from the total of 4MB). 
- I only have to get the bluetooth part working and the others will write that to NVM. I do know I will be running into memory issues, but that will be something to worry about at a later time.

thanks for that vector. It looks/sounds like that would be a great way to save the addresses (I do have current time on the esp32 so I already have the timestamp). I am kinda busy so I can't program now. I would have to import and edit the BLEScan.cpp file right?

Wouldn't I be better of using a already existing product? Or maybe forget about saving know devices and just count the amount of devices?

chegewara, 2018-04-12:
Its all depends on your project requirements. BLEScan has additional paramete, that lets you ignore or not already scanned ble device. Again this is that part of code:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEScan.cpp#L95-L104

You can for example create new BLEScan object every 24h with or without addidtional parameter (depend on needs), this way internal vector will be clean (i dont know how about destroying old blescan object to avoid memory leak, you need to do some research or tests). Now in BLEScan callback->onResult() you can do whatever you want, store seen device data or just count it etc.
All depend on project needs/reuirements. 

In my opinion 600kB should be onough for your needs, maybe not to track and store devices in RAM for weeks but to build nice project.

lauwie007, 2018-04-16:
You have been of great help chegewara.

I want to capture the device address, preferable by multiple ints (than store it in a msg to send over LoRa). To do that would I need to change 'BLEScan.cpp' or could I capture the device address in my normal void loop?

chegewara, 2018-04-16:
If you will follow code you can see `BLEAddress::getNative()` function which returns esp_bd_addr_t* and `esp_bd_addr_t ` is an array of 6 uint8_t values.

lauwie007, 2018-04-18:
How could I use this? I keep ketting a error `cannot call member function 'uint8_t (* BLEAddress::getNatice())[6]' without object`. I tried these lines in my loop: `BLEAddress::getNative();`, `BLEScan* pBLEScan = BLEAddress::getNative();`, `pBLEScan->BLEUUID::getNative();` and `pBLEScan->BLEAdvertising::getNative();`

defining the uint8_t above my setup as  `uint8_t esp_bd_addr_t[6];` gives a error: redeclared as different kind of symbol

chegewara, 2018-04-18:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAddress.cpp#L70

https://github.com/nkolban/esp32-snippets/blob/ff9368469868c090cb1172aad61e985f2e587e5a/cpp_utils/tests/BLETests/SampleClientWithWiFi.cpp#L99

AdrianViverosL, 2020-09-29:
May I ask you how I could store the native address into an array of integers or bytes? 

I was trying something like this at the onResult from MyAdvertisedDeviceCallbacks Class
`
int addrs[6] = {advertisedDevice.getAddress().getNative()};
`
But it doesn't work, I get the following error: 
`error: cannot convert 'uint8_t (*)[6] {aka unsigned char (*)[6]}' to 'uint8_t* {aka unsigned char*}' in initialization
         uint8_t* addrs[6] = {advertisedDevice.getAddress().getNative()};`

I already tried changing the variable type of "addrs", I tested with char*, uint8_t however it keeps not working and only changes the variable type at the log error. 
What am I doing wrong? 

Thanks in advance!

chegewara, 2020-09-30:
Did you try to use `memcpy`?

AdrianViverosL, 2020-09-30:
That was all, it works! Thank you so much, you are such a genius!