nkolban/esp32-snippets #471: BLE client don't see service from ESP32 BLE server

SalvaPedraza, 2018-04-11:
In my ESP32 project I have issues with BLE server. Changing some order of code or adding some code in other functions (nothing to do with BLE) caused that a BLE client (I used nRF Connect and BLE Scanner android applications) don't see the BLE service. I tested in 2 different android mobiles.
After lot of simplification to reproduce the error with minimum code I manage to reproduce the error with small variation of: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleServer.cpp example.

The example as provided works perfectly as expected. You can see client screenshot:
[image]

But just changing line 24:
BLEService* pService = pServer->createService("91bad492-b950-4226-aa2b-4ede9fa42f59");

by:
BLEService* pService = pServer->createService(BLEUUID("91bad492-b950-4226-aa2b-4ede9fa42f59"));

(I assume this is completely equivalent, right?)
Caused that the client do not see the service when connected to ESP32 BLE server:
[image]

The raw data obtained by client previous to connect is exactly the same in both cases:
[image]
The difference is when client connect to server.
But the problem disappear if for instance in line 23 I introduce:
BLEUUID aux("91bad492-b950-4226-aa2b-4ede9fa42f59");
aux variable is not used in the code, but just creating it make the problem to disappear.
It seems to be a problem with memory conflict somewhere.
I have been reviewing the BLE modules but I can not find any explanation for this fail.
Any idea?

chegewara, 2018-04-11:
I can only ask 2 questions: did you update this library lately? and could you try with settings like in this post:
https://github.com/nkolban/esp32-snippets/issues/264#issuecomment-357353186

SalvaPedraza, 2018-04-16:
Thanks very much for advise chegeware. Certainly, with update of toolchain, esp-idf and esp32-snipset to the latest, everything works now as expected. No needs to apply #264.

hansmbakker, 2018-12-15:
@SalvaPedraza it seems your issue is solved, could you please close it to keep the issue list clean?
cc @chegewara

closed 2018-12-30 by chegewara