nkolban/esp32-snippets #472: "Too few arguments to function 'esp_err_t esp_ble_gattc_open(esp_gatt_if_t, uint8_t*, esp_ble_addr_type_t, bool)'" when trying to run any of the examples

pgrlopes, 2018-04-12:
Hey guys.

I've been looking to get my esp32 ble component to send some information to my smartphone for the past couple days and I'm now trying to use this repository as a base for my code. However, none of the examples for the server or client work because of the error in the title.

I've googled it and found some answers pointing to some errors introduced with new updates to the libraries and that I should revert to a previous version but I'm new here and I just don't know how to do it.

Any help would be appreciated.
Thank you.

chegewara, 2018-04-12:
Arduino-ide or esp-idf?

pgrlopes, 2018-04-12:
Sorry, i'm using arduino IDE

chegewara, 2018-04-12:
If you are working on windows then go to folder:
`C:\Users\USER\Documents\Arduino\hardware\espressif\esp32`
execute command:
`git status`

if you get result other than:
```
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
```
then execute:
```
git pull
git submodule update --init
```

Maybe there is something i am missing because i did not check before update, but with updated stack first example i tested (BLEServer) works.

pgrlopes, 2018-04-12:
Yeah I had tried this solution previously as it was suggested in another thread and it didn't work. I tried it now and it's working now. My issue is that when I installed the esp32 modules it wasn't on the c:\users directory but instead in the arduino ide directory itself (C:\Program and files\Arduino...) and the libraries were in conflict.

Could you just tell me what is the directory that I should be using for the esp32 folder? Is it the C:\users one?

Thank you very much for your help.

MajicCat, 2018-04-25:
Just hit the same issue. I did not use Git, just the regular install. Any thoughts on how to correct?

Arduino: 1.8.5 (Windows 10), Board: "Nano32, 80MHz, 921600"

C:\Users\xxx\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\src\BLEClient.cpp: In member function 'bool BLEClient::connect(BLEAddress)':

C:\Users\xxx\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\src\BLEClient.cpp:115:2: error: too many arguments to function 'esp_err_t esp_ble_gattc_open(esp_gatt_if_t, uint8_t*, bool)'

  );

  ^

In file included from C:\Users\Ames\Documents\Arduino\libraries\ESP32_BLE_Arduino-master\src\BLEClient.cpp:13:0:

C:\Users\Ames\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid/esp_gattc_api.h:293:11: note: declared here

 esp_err_t esp_ble_gattc_open(esp_gatt_if_t gattc_if, esp_bd_addr_t remote_bda, bool is_direct);

           ^

exit status 1
Error compiling for board Nano32.

MajicCat, 2018-04-26:
Got it, was outdates ESP lib.

The demo consume a huge amount of memory. Is that expected

Sketch uses 1232701 bytes (94%) of program storage space. Maximum is 1310720 bytes.
Global variables use 46624 bytes (15%) of dynamic memory, leaving 248288 bytes for local variables. Maximum is 294912 bytes.

chegewara, 2018-04-26:
Yes, i dont know why, it was about 900kB some time ago, but now its a lot more.

MajicCat, 2018-04-26:
Is there a stable build with the smaller profile  that can be pulled and compared?

pana216, 2019-06-12:
> Arduino-ide or esp-idf?

I'm working with the esp-idf and am getting that error - any suggestions?

chegewara, 2019-06-12:
> I'm working with the esp-idf and am getting that error - any suggestions?

Yes, after a year its so many changes in esp-idf so i suggest to open new issue and provide esp-idf version you are using and logs.

VijitSingh97, 2019-06-12:
@pana216 I fixed by changing lines in BLEDevice.cpp with:

line 494: 	esp_err_t errRc = esp_ble_gap_update_whitelist(true, *address.getNative(), BLE_WL_ADDR_TYPE_PUBLIC);  // True to add an entry.

line 508: 	esp_err_t errRc = esp_ble_gap_update_whitelist(false, *address.getNative(), BLE_WL_ADDR_TYPE_PUBLIC);  // False to remove an entry.

pana216, 2019-06-12:
> @pana216 @chegewara I fixed by changing lines in BLEDevice.cpp with:
> 
> line 494: esp_err_t errRc = esp_ble_gap_update_whitelist(true, *address.getNative(), BLE_WL_ADDR_TYPE_PUBLIC); // True to add an entry.
> 
> line 508: esp_err_t errRc = esp_ble_gap_update_whitelist(false, *address.getNative(), BLE_WL_ADDR_TYPE_PUBLIC); // False to remove an entry.
> 
> that last argument can be public or random, idk what it does though
> 
> https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/bluetooth/esp_bt_defs.html#_CPPv422esp_ble_wl_addr_type_t

This resolved to some other errors and warning, but these were easily to handle, just googled and fixed them. Thank you very much I really appreciate it!!