nkolban/esp32-snippets #480: unable to import c_includes.xml

wdouglas1999, 2018-04-17:
Good day. 
I managed to compile boot WiFi with no issues , but when I try to include files I need to do them one at a time since I changed the file to . Cpp extension. 
I am unable to import includes file c_includes.xml.
after giving path. and finish nothing happens.
Can anyone assist? Thanks

chegewara, 2018-04-17:
https://github.com/nkolban/esp32-snippets/blob/master/eclipse/c_includes.xml

wdouglas1999, 2018-04-17:
Hi, thank you for the reply.
I have this file already and I use it to try to import the includes, but the includes are never added and the list remains blank. I can only add it one by one? What could I be doing wrong? I’m on Windows 10.

chegewara, 2018-04-17:
I may be wrong but this file may be too long, ive got my own include.xml prepared on ubuntu and with many includes it cant be loaded. You dont need assembler part, so you can remove it from this file:
https://github.com/nkolban/esp32-snippets/blob/master/eclipse/c_includes.xml#L7-L37

Or just prepare your own includes in one project, then export it to any_includes.xml.

wdouglas1999, 2018-04-17:
Hi, I shortened the file but still not able to import the includes.
Thank you for you help

Gnorkus, 2018-04-22:
Hi,

I think I have a weird solution to the problem, wdouglas.  I experimented by exporting entries, and noticed, that although the entries are labelled GNU C and GNU C++, it really wants "C Source File" and "C++ Source File".  Must be a bug in eclipse, I suppose.  Mine even loads the Assembly Files.  Can you tell me if this works for you???  It's an XML file stuffed in the following zip.  Just store it in the same folder where c_includes.xml is found...

I'm running a Windows 10 machine also, and it might be that the current version is different than the Unix in some minor way.  I can't confirm or deny it...

[Test_includes_GNU.zip](https://github.com/nkolban/esp32-snippets/files/1935125/Test_includes_GNU.zip)

Lukykl1, 2019-01-07:
Hi, I can confirm that @Gnorkus xml works. I couldn't import with original.

Lukykl1, 2019-01-07:
Hi, I can confirm that, @Gnorkus version is working. Original don't work for me.