nkolban/esp32-snippets #481: #Exception was unhandled

viky1997, 2018-04-17:
[image]

we tried to communicate between two ESP 32 boards using BLE . it was successfully done by editing the example sketches of uart and client  now when we  tried to connect the two esp 32 and communicate between them by using  a new library the crash occurs . even when we put a line of our code inside a new library  crash occurs.kindly help us

CPP file -> "Bleserial.cpp"

#include "Bleserial.h"
#include "BLEDevice.h"
#include "BLEUtils.h"
#include "BLEServer.h"
#include "BLEScan.h"
#include "BLERemoteCharacteristic.h"
#include "BLECharacteristic.h"
//#include "BLE2902.h"

#include "sdkconfig.h"
#if defined(CONFIG_BT_ENABLED)
#include <esp_log.h>
#include <esp_bt.h>
#include <esp_bt_main.h>
#include <esp_gap_ble_api.h>
#include <esp_gattc_api.h>
#include "BLEClient.h"
#include "BLEUtils.h"
#include "BLEExceptions.h"
#include "BLEService.h"
#include "GeneralUtils.h"
#include <string>
#include <sstream>
#include <unordered_set>
#ifdef ARDUINO_ARCH_ESP32
#include "esp32-hal-log.h"
#endif
#endif

/*#include "sdkconfig.h"
#if defined(CONFIG_BT_ENABLED)

#include <esp_gattc_api.h>
#include <esp_log.h>
#include <esp_err.h>

#include <sstream>
#include "BLEExceptions.h"
#include "BLEUtils.h"
#include "GeneralUtils.h"
#include "BLERemoteDescriptor.h"
#ifdef ARDUINO_ARCH_ESP32
#include "esp32-hal-log.h"
#endif*/

int flag=0;
std::string x;
std::string val;
static bool doConnect=false;
static bool connected=false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAddress *pServerAddress;
static BLECharacteristic *pCharacteristic;
BLEUUID serviceUUID;
BLEUUID charUUID;
BLEScan* pBLEScan;

void onWrite(BLECharacteristic *pCharacteristic);
bool connectToServer(BLEAddress pAddress);
void scan();

/*class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
 
  // Called for each advertising BLE server.
   
void onResult(BLEAdvertisedDevice advertisedDevice) {
   // Serial.print("BLE Advertised Device found: ");
   // Serial.println(advertisedDevice.toString().c_str());

    // We have found a device, let us now see if it contains the service we are looking for.
   // if (advertisedDevice.haveServiceUUID() && advertisedDevice.getServiceUUID().equals(serviceUUID)) {

      // 
      //Serial.print("Found our device!  address: "); 
      advertisedDevice.getScan()->stop();

      pServerAddress = new BLEAddress(advertisedDevice.getAddress());
     doConnect = true;

    } // Found our server// onResult
};// MyAdvertisedDeviceCallbacks*/

Bleserial::Bleserial()
{

  //Serial.begin(115200);
 // BLEDevice::init("UART_RX");
  

  BLEScan* pBLEScan = BLEDevice::getScan();
  pBLEScan->setActiveScan(true);
  pBLEScan->start(30);
 
  
 // pCharacteristic->setCallbacks(new MyCallbacks());

  //pCharacteristic->setValue("Hello World");
  
}

 

std::string Bleserial::get_data()
{
 
//pCharacteristic->setCallbacks(new MyCallbacks());	
 //pCharacteristic->addDescriptor(new BLE2902());
class MyCallbacks: public BLECharacteristicCallbacks {
	
  void onWrite(BLECharacteristic *pCharacteristic) {
      std::string value = pCharacteristic->getValue();
      val=value;
      }   
  };
  
  //return(val);

}

void Bleserial::start()
{
	
  BLEDevice::init(""); 	
 
  BLEServer *pServer = BLEDevice::createServer();
  
  BLEService *pService = pServer->createService(serviceUUID);

  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         charUUID,
                                         BLECharacteristic::PROPERTY_READ 
                                         //BLECharacteristic::PROPERTY_WRITE
                                       );
                                       
  pService->start();
  //pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());	
  BLEAdvertising *pAdvertising = pServer->getAdvertising();
  pAdvertising->start();
                                     
                                       

}
//class MyCallbacks: public BLECharacteristicCallbacks {

       // };

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {

  void onResult(BLEAdvertisedDevice advertisedDevice) {
   // Serial.print("BLE Advertised Device found: ");
   // Serial.println(advertisedDevice.toString().c_str());
   // Serial.println(advertisedDevice.getUuids().c_str());
    advertisedDevice.getScan()->stop();
    pServerAddress = new BLEAddress(advertisedDevice.getAddress());
    doConnect = true;
 

    } // Found our server
     // onResult
}; // MyAdvertisedDeviceCallbacks*/

void Bleserial::push(std::string x)
{  
    pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());

   //std::string x=x.toString();
   if (doConnect == true) {
     if (connectToServer(*pServerAddress)) {
     // Serial.println("We are now connected to the BLE Server.");
      connected = true;
    } 

/*	else {
     // Serial.println("We have failed to connect to the server; there is nothin more we will do.");
    }*/
     doConnect = false;
 }
 

  // If we are connected to a peer BLE Server, update the characteristic each time we are reached
  // with the current time since boot.
 // connected=true;
  if (connected) {
    pRemoteCharacteristic->writeValue(x,x.length());
  }

}

bool connectToServer(BLEAddress pAddress) {
    /*Serial.print("Forming a connection to ");
    Serial.println(pAddress.toString().c_str());*/
    
    BLEClient*  pClient  = BLEDevice::createClient();
    //Serial.println(" - Created client");

    // Connect to the remove BLE Server.
    pClient->connect(pAddress);
   // Serial.println(" - Connected to server");

    // Obtain a reference to the service we are after in the remote BLE server.
   BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService == nullptr) {
      //Serial.print("Failed to find our service UUID: ");
     // Serial.println(serviceUUID.toString().c_str());
      return false;
    }
   // Serial.println(" - Found our service");
    
   pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
    if (pRemoteCharacteristic == nullptr) {
    /*   Serial.print("Failed to find our characteristic UUID: ");
      Serial.println(charUUID.toString().c_str());*/
      return false;
    }
   // Serial.println(" - Found our characteristic");

    // Read the value of the characteristic.
    std::string value = pRemoteCharacteristic->readValue();
   /* Serial.print("The characteristic value was: ");
    Serial.println(value.c_str());*/
    //return true;
}

Header file->"bleserial.h"

#ifndef Bleserial_h
#define Bleserial_h

#include "Bleserial.h"
#include "BLEDevice.h"
#include "BLEUtils.h"
#include "BLEServer.h"
#include "BLEScan.h"
#include "BLERemoteCharacteristic.h"
#include "BLECharacteristic.h"
//#include "BLE2902.h"

#include "sdkconfig.h"
#if defined(CONFIG_BT_ENABLED)
#include <esp_log.h>
#include <esp_bt.h>
#include <esp_bt_main.h>
#include <esp_gap_ble_api.h>
#include <esp_gattc_api.h>
#include "BLEClient.h"
#include "BLEUtils.h"
#include "BLEExceptions.h"
#include "BLEService.h"
#include "GeneralUtils.h"
#include <string>
#include <sstream>
#include <unordered_set>
#ifdef ARDUINO_ARCH_ESP32
#include "esp32-hal-log.h"
#endif
#endif

class Bleserial
{
	public:
		Bleserial();                        /*constructor*/
		void start();                      /*initaliztion function*/
		std::string get_data();            /*function to read*/
		void push(std::string);
	
};

#endif

chegewara, 2018-04-26:
Hi, i would like to help but this code is hard to read without proper formatting here. There is one thing i see. Why do you include Bleserial.h in Bleserial.h file?

```
Header file->"bleserial.h"

#ifndef Bleserial_h
#define Bleserial_h

#include "Bleserial.h"
```

viky1997, 2018-04-26:
Hi .. bleserial is the  name of the library we are going to make for
communicating between two esp boards .. we have done. Communication between
two boards by editing the uart and client examples. We tried to import the
same edited code into a library

chegewara, 2018-04-26:
Yes, but you are including header file to itself. Probably its not the case, because there is guard:
```
#ifndef Bleserial_h
#define Bleserial_h
```