nkolban/esp32-snippets #482: querying status of congestion before sending

johnty, 2018-04-17:
i seem to be hitting the maximum bandwidth of the BLE connection, and am getting the following:

>E (2202025) BT: L2CAP - CID: 0x0004 cannot send, already congested             xmit_hold_q.count: 200 buff_quota: 200
>E (2202026) BT: ATT   failed to pass msg:0x1b to L2CAP

is it possible to query the status of the send queue so that i can programmatically back off the send rate if we get close to this point?

johnty, 2018-04-20:
quick update:

I've started experimenting with this, and one first attempt was to look at issuing a callback from the [BLEServer's handleGATTServerEvent](https://github.com/johnty/ESP32_BLE_Arduino/blob/28dcf9c32a35c75d74c46f5699d658193db6c9bb/src/BLEServer.cpp#L224) for the ESP_GATTS_CONGEST_EVT. I am able to receive the callback right when the above error happens. However, I realize that this is actually too late because by the time we see this, the send queue is already full and packets are being dropped already.

Ideally, it would be nice to poll the current size of the queue (xmit_hold_q.count?) once in a while, and then react preemptively before the queue reaches a certain size. But I haven't managed to navigate through the layers of APIs to see how this can be done.

Any pointers would be greatly appreciated!

chegewara, 2018-04-20:
This is what i think, when you get ESP_GATTS_CONGEST_EVT means your packets are sending too fast and last packet has not been send. So you should have ie to resend it after some small delay from onCongest callback, or set some flag that will let know the rest of your code whats going on. 
Im not 100% sure, but as far as i know congest is raised when you send packets too fast, and you can try to optimise it by sending bigger packets if its possible.

johnty, 2018-04-20:
Thanks for the comment!

Yes, because our data rate is around or possibly more than the available BLE bandwidth*, our approach is indeed to slow things down when we hit this stage. But wasn't sure if it is possible to try and detect this _before_ we hit the wall and seeing that congestion event. Poking at the size of the transmit queue (if possible) seems like one preemptive method. Otherwise may also have to just keep track of the current sending rates based on data we try to push out and then making a guess at a better (slower) rate... Will poke around further... :)

*for an idea: we're seeing a max of around 5-6kBps throughput going to Mac OSX, which seems to be in the ballpark based on [this article](https://punchthrough.com/blog/posts/maximizing-ble-throughput-on-ios-and-android), when using connection interval of 11.25ms. In this case we also want to have low latency, so its like the worst of both worlds... :P