nkolban/esp32-snippets #483: Use .cpp snippets in .c or visa Versa.

wdouglas1999, 2018-04-17:
Good day,
Is it at all possible to use these .cpp snippets and include them in a .c file or visa Versa?
Are there any examples of where this is done?
Thank you.

nkolban, 2018-04-18:
Howdy,
Code written in C++ (that uses C++ features) needs to be compiled with a C++ compiler.  One can mix code written in C with code written in C++ but each must be in its own source file.  For example, exclusively C code should be in a file with suffix `.c` and exclusively C++ code should be in a file with suffix `.cpp`.  These are then compiled with their respective compilers.  At linkage time, the resulting object files can be carefully linked together.

wdouglas1999, 2018-04-18:
Howdy from Houston,
Thank you for the great explanation. Sounds like things can get complicated mixing these languages, however the reward can be great.
Is there any reading recommendations you have where I can learn more about the topic? Like how to link the files in eclipse? Seems interesting.

chegewara, 2018-04-18:
Easiest way is to study esp-idf code, which is C code, and see how its mixed with C++ in this library.

nkolban, 2018-04-18:
C++ is mostly a superset of C ... I'm pretty sure some google searches will show up recipes for mixed language support.  The good news is that mixing C and C++ in ESP32 is almost identical to how it is performed in Linux and other operating systems ... so nothing special need really be done for the ESP32.

For compilation, I use the ESP-IDF supplied make/build system and don't recommend attempting to link from Eclipse.

wdouglas1999, 2018-04-18:
Thank you all for the advice.

closed 2018-05-15 by nkolban