nkolban/esp32-snippets #484: Arduino Librarys ease of use

Project-Aurora, 2018-04-18:
I posted on the esp32 Arduino Forum earlier, this is a Continuation of that Thread.
I'm more of a Hardware guy, I can compile and flash the examples from this Repo just fine, however modifying it to suit my needs is too hard for me, given that the Documentation is more suited for People who already know what they are doing.

In contrast, it took me 10 Minutes to get a working Gamepad going using this Arduino Library:
https://github.com/MHeironimus/ArduinoJoystickLibrary

Are there any Plans to make the Arduino Library easier to use?
I specifically mean things like replacing the state Controll with optional functions like "onConnnect()" that can be omitted if you're not doing anything special with them.
Another thing that raises the Barrier of Entry is having to Fiddle around with Descriptors, other Arduino Librarys usually have a Function like "Keyboard.begin()" that takes care of these things.

I know this is a lot to ask, but this would open up the ESP32 to a lot of People.

chegewara, 2018-04-18:
First of all we have two different libraries. The one from this repo is general HID or to be precise its HID over GATT library, the one from you link is a Joystick library which is typicaly prepared to build joysticks/gamepads.
```
In contrast, it took me 10 Minutes to get a working Gamepad going using this Arduino Library:
https://github.com/MHeironimus/ArduinoJoystickLibrary
```

Second of all, i understand its not easy to use something that you dont understand, it took me about month to study and understand USB, HID and HID over GATT and then this class. 
Here are links that may help you understanding few basic things:
http://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/
https://github.com/jpbrucker/BLE_HID/blob/master/doc/HID.md
https://github.com/jpbrucker/BLE_HID/blob/master/doc/HIDService.md

You are saying its not easy to use, and i am saying its as much easy as it can be. HID structure is not constant, you can have 1 to n input, output and feature reports. Each report is a single ble characteristic with descriptors and I cant predict how many reports you are gonna to use (need). But i know that each report (characteristic) requires some descriptors and this library is creating those descriptors for you in background. All required services and characteristics are created for you and you dont have code it or to worry about it. Sure, i can write gamepad example, but it will be ready to use gamepad code, and it is not hard to do it. You just need to write:
- report map, that will describe it is gamepad with buttons and sticks structure,
- write interrupt handler, that will read buttons state when something is changed, 
- write buttons state to proper characteristic and send notify.

chegewara, 2018-04-18:
Yes, its hard to tell what you need to change. To simplify all this is part of code that you should reuse in your project, its minimum base (without optional security):
```
#include "BLEDevice.h"
#include "BLEServer.h"
#include "BLEUtils.h"
#include "BLE2902.h"
#include "BLEHIDDevice.h"
// #include "SampleKeyboardTypes.h"  <-- HID keyboard only
#include <esp_log.h>
#include <string>
#include <Task.h>

#include "sdkconfig.h"

static char LOG_TAG[] = "SampleHIDDevice";

static BLEHIDDevice* hid;

/*
* here is example how to handle and send InputReport message. For each device input report will be build in different way. In this example its keyboard, and according to specs input report is 8 byte message with 1st byte is a flag, 2nd is reserver (0x00) and bytes 3 to 8 are key stroke encoded values.
*
*/
/*
class MyTask : public Task {
	void run(void*){
    	vTaskDelay(10000);
    	const char* hello = "Hello world from esp32 hid keyboard!!!";
		while(*hello){
			KEYMAP map = keymap[(uint8_t)*hello];
			uint8_t a[] = {map.modifier, 0x0, map.usage, 0x0,0x0,0x0,0x0,0x0};
			hid->inputReport(NULL)->setValue(a,sizeof(a));
			hid->inputReport(NULL)->notify();

			hello++;
		}
			uint8_t v[] = {0x0, 0x0, 0x0, 0x0,0x0,0x0,0x0,0x0};
			hid->inputReport(NULL)->setValue(v, sizeof(v));
			hid->inputReport(NULL)->notify();
		vTaskDelete(NULL);
	}
};
*/
  MyTask *task;

  class MyCallbacks : public BLEServerCallbacks {
    void onConnect(BLEServer* pServer){
    	task->start();
    }

    void onDisconnect(BLEServer* pServer){

    }
  };
class MainBLEServer: public Task {
	void run(void *data) {
		ESP_LOGD(LOG_TAG, "Starting BLE work!");

		task = new MyTask();
		BLEDevice::init("ESP32");
		BLEServer *pServer = BLEDevice::createServer();
		pServer->setCallbacks(new MyCallbacks());

		/*
		 * Instantiate hid device
		 */
		hid = new BLEHIDDevice(pServer);

		/*
		 * Set manufacturer name
		 * https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.manufacturer_name_string.xml
		 */
		std::string name = "esp-community";
		hid->manufacturer()->setValue(name);

		/*
		 * Set pnp parameters
		 * https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.pnp_id.xml
		 */
		const uint8_t pnp[] = {0x01,0x02,0xe5,0xab,0xcd,0x01,0x10};
		hid->pnp()->setValue((uint8_t*)pnp, sizeof(pnp));

		/*
		 * Set hid informations
		 * https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.hid_information.xml
		 */
		const uint8_t val1[] = {0x01,0x11,0x00,0x03};
		hid->hidInfo()->setValue((uint8_t*)val1, 4);

		/*
		 * Report map
		 */
		const uint8_t reportMap[] = {
               // HERE YOU HAVE TO INSERT REPORT MAP THAT WILL DESCRIBE YOUR DEVICE CAPABILITIES AND LET CLIENT DRIVER TO UNDERSTAND WHAT DEVICE IT IS
		};
		/*
		 * Set report map (here is initialized device driver on client side)
		 * https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.report_map.xml
		 */
		hid->setReportMap((uint8_t*)reportMap, sizeof(reportMap));

		/*
		 * We are prepared to start hid device services. Before this point we can change all values and/or set parameters we need.
		 * Also before we start, if we want to provide battery info, we need to prepare battery service.
		 * We can setup characteristics authorization
		 */
		hid->startServices();

		/*
		 * Its good to setup advertising by providing appearance and advertised service. This will let clients find our device by type
		 */
		BLEAdvertising *pAdvertising = pServer->getAdvertising();
		pAdvertising->setAppearance(HID_GAMEPAD);    // <-- optional
		pAdvertising->addServiceUUID(hid->hidService()->getUUID());
		pAdvertising->start();

		ESP_LOGD(LOG_TAG, "Advertising started!");
		delay(1000000);
	}
};

void SampleHID(void)
{
	//esp_log_level_set("*", ESP_LOG_DEBUG);
	MainBLEServer* pMainBleServer = new MainBLEServer();
	pMainBleServer->setStackSize(20000);
	pMainBleServer->start();

} // app_main
```

chegewara, 2018-04-18:
Now report map (its a heart of hid device). For me most useful was this blog:
http://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/

You have to describe in HID language how many buttons your device have, how many sticks etc and how each of them will be encoded in input report. Standard way is to encode each single button as a bit value but you can easy send each button as a single byte.
Understanding report map structure is hardest part, but you can find many joysticks and gamepads examples and copy report map from it. Then i will try to help you to encode it and prepare to make it ready to send.

chegewara, 2018-04-18:
I just did quick look at this code and i believe that with minimum effort you can combine it with this library to have nice and easy joystick:
https://github.com/MHeironimus/ArduinoJoystickLibrary/blob/master/Joystick/src/Joystick.cpp

In my opinion its enough to add and change just a few lines to make it works.

Project-Aurora, 2018-04-18:
I'm feeling pretty stupid right now.
I don't even know if the code you posted is C++ or for the Arduino IDE.
It fails to build in both cases, please tell me which log to provide.

chegewara, 2018-04-18:
You are right. My bad, sorry for that. Those examples are outdated and i forgot to change it to match changes in class. I will fix this code for you and post it here so you can start with arduino.

chegewara, 2018-04-18:
One more time sorry for old and broken code. Here you have something you can start on arduino:
(i am also available on IRC channel #esp32)
```
#include "BLEDevice.h"
#include "BLEServer.h"
#include "BLEUtils.h"
#include "BLE2902.h"
#include "BLEHIDDevice.h"
#include <esp_log.h>

//#include "sdkconfig.h"

static char LOG_TAG[] = "SampleHIDDevice";

static BLEHIDDevice* hid;

//EXAMPLE
struct gamepad_report_t
{
    uint16_t buttons;
    int8_t left_x;
    int8_t left_y;
    int8_t right_x;
    int8_t right_y;
};

void setup() {
  // put your setup code here, to run once:
    ESP_LOGD(LOG_TAG, "Starting BLE work!");

    BLEDevice::init("ESP32");
    BLEServer *pServer = BLEDevice::createServer();
//    pServer->setCallbacks(new MyCallbacks());

    /*
     * Instantiate hid device
     */
    hid = new BLEHIDDevice(pServer);

    /*
     * Set manufacturer name
     * https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.manufacturer_name_string.xml
     */
    std::string name = "esp-community";
    hid->manufacturer()->setValue(name);

    /*
     * Set pnp parameters
     * https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.pnp_id.xml
     */
    hid->pnp(0x01,0x02e5,0xabcd,0x0110);

    /*
     * Set hid informations
     * https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.hid_information.xml
     */;
    hid->hidInfo(0x00,0x03);

    /*
     * Report map
     */
    const uint8_t reportMap[] = {
               // HERE YOU HAVE TO INSERT REPORT MAP THAT WILL DESCRIBE YOUR DEVICE CAPABILITIES AND LET CLIENT DRIVER TO UNDERSTAND WHAT DEVICE IT IS
          // http://www.usb.org/developers/hidpage/Hut1_12v2.pdf
          // in this document you have examples of gamepad and joystick report map (Appendix A)
          //
                
                USAGE_PAGE(1),       0x01,
                USAGE(1),         0x02,
                 COLLECTION(1),     0x01,
                 USAGE(1),        0x01,

//          .....

              END_COLLECTION(0),
            END_COLLECTION(0)         
    };
    /*
     * Set report map (here is initialized device driver on client side)
     * https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.report_map.xml
     */
    hid->reportMap((uint8_t*)reportMap, sizeof(reportMap));

    /*
     * We are prepared to start hid device services. Before this point we can change all values and/or set parameters we need.
     * Also before we start, if we want to provide battery info, we need to prepare battery service.
     * We can setup characteristics authorization
     */
    hid->startServices();

    /*
     * Its good to setup advertising by providing appearance and advertised service. This will let clients find our device by type
     */
    BLEAdvertising *pAdvertising = pServer->getAdvertising();
    pAdvertising->setAppearance(HID_GAMEPAD);    // <-- optional
    pAdvertising->addServiceUUID(hid->hidService()->getUUID());
    pAdvertising->start();

}

gamepad_report_t oldValue, newValue;

void loop() {
  // put your main code here, to run repeatedly:

// pseudocode 
/*
 * newValue = read all buttons state and set all bits according to button value
 * or you may have setup interrupts procedure that will be called when any button change state 
 * and set newValue in it
 * 
 * if(oldValue != newValue)
 * {
 * 
 * reportCharacteristic->setValue(value)
 * reportCharacteristic->notify()
 * oldValue = newValue;
 * }
 * 
 * 
 */
}
```

Project-Aurora, 2018-05-02:
I'll work on this on my own time.
So this Ticket doesn't have a Purpose anymore.

closed 2018-05-02 by Project-Aurora