nkolban/esp32-snippets #486: The GATT role and a READ property usage example

pgrlopes, 2018-04-19:
Hey guys.

So I've been reading a lot about GATT as I want to implement a simple sensor node transmitting data through BLE and so far I haven't quite understood what the point of a specific service UUID is. Whats the difference between writing the UUID as a general purpose service or a specific one like the heart rate monitor? Does that have any impact at all on the devices connecting to the service? What would be the consequences of having an environment sensing service defined as a heart rate monitor to the system?

Another question that I would like to ask is if anyone knows of any READ property examples. The service that I want to use has this property as mandatory but I can't quite understand how it actually works and how to broadcast whatever is read through BLE to the device connected.

Thank you

chegewara, 2018-04-19:
Its answer on second question. Characteristic needs to have `BLECharacteristic::PROPERTY_READ`:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleServer.cpp#L28

First question, what is the purpose and consequences:
lets think about UUID like phone number: general purpose UUID is like ordinary phone number, anyone can have one and only the client who wants to connect needs to knows it; now lets think about specific UUID like special phone number, lets say 112 is heart rate monitor; every client may knows it and knows exactly what to expect when connects to this number, and every client app knows exactly what data expects when connects and reads data from service/characteristic with this UUID, its because all of this is described in bluetooth specs.

Only difference is that you can use heart rate monitor UUID in your weather station app, but all heart rate monitor applications on android or iPhone may being fooled and try to connect, also your custom android/iPhone client app may be confused and try to read weather station data from real heart rate monitor peripheral.

pgrlopes, 2018-04-19:
Thanks chegewara.

As for the first question I understand that I have to setup my READ property. I've already done that, i'm trying to create an environmental sensing hardware node that has a BLE component to it. I'm not sure what exactly im supposed to read though. If I go to the bluetooth website, in the GATT section I find the environmental sensing service is composed of several characteristics, one of them being the temperature (which is what I'm trying to read now). That characteristic has it's READ property as mandatory. What does this mean though? Sure I set it up but then what do I do with it? How do I check if my server is notifying the temperature value that was written to it to all the ble devices connected?

As for the second question I understand it now, it's what I thought at the beginning but because I don't know how those apps work I was kinda sceptical about it.

pgrlopes, 2018-04-20:
Thank you, that was indeed quite helpful.

However when I got to the part where he notifies whenever a value changes, I got lost. Where exactly does he do it? I'm using the arduino IDE so I cannot do it in the loop function as the characteristics are not defined in that scope, but he does not declare any callbacks so it's also not done in a callback. This essentially brought me back to my first question which is: where exactly is this done? Where is the read request received and where can I send the information?

Thanks once again!

chegewara, 2018-04-20:
You can do it in loop. If your characteristic is not defined in this scope then you need to create BLECharacteristic outside setup function, somewhere above it. Then you have global variable accessed from setup, loop and every function you have created.

To receive any requests, read or write, you have to create class derived from BLECharacteristicCallbacks. Here you have example with onWrite() callback:

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_uart/BLE_uart.ino#L51-L65

You can replace onWrite or add onRead to this class. Set callbacks, and you are ready:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_uart/BLE_uart.ino#L94

but to read characteristic value you dont need this callback, its for other purpose, like ie if you want to keep tracking how many times characteristic has been read or something.

chegewara, 2018-04-23:
uint8_t* is a pointer to variable, ie array of uint8_t(char) values. You can create variable like this:
`uint8_t variable[5] = {0};`
Now you have variable type uint8_t with size_t equal 5. Now you can pass this variable to function like this (array is passed by first element address):
`tempCharacteristics.setValue(variable,5);` 

in your case you i assuming you have variable `uint8_t tempOut;` so you can do it like this:
`tempCharacteristics.setValue(&tempOut, 1);`

pgrlopes, 2018-04-23:
Yeah I did that and I can see in the phone app the notifying function working. The values however are not increasing by 1. At the end of my loop function I add

`tempOut++;`

But it's increasing in 0.01 figures, not 1 by 1. So I guess i'm not fully understanding the conversion from between what I have defined in my server and what is showing up on my phone. It's also not helping that I can't read the value in the serial monitor, it's showing CharReadEvt when I'm trying to print the tempOut value like this:

`Serial.println(tempOut + "\n");`

chegewara, 2018-04-23:
`Serial.println((int)tempOut + "\n");`

To fix this `But it's increasing in 0.01 figures, not 1 by 1` you have 2 options:
- change value before setValue():` tempOut * 100`,
- study descriptors, because (im assuming) for environmental temperature characteristic nRF connect has setup to read value with 0.01 precision, you can change it with one of descriptors (dont remember which one but i think its 0x2904)
https://www.bluetooth.com/specifications/gatt/descriptors
https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.characteristic_presentation_format.xml

pgrlopes, 2018-04-26:
Sorry for not answering before, but thanks chegewara, the descriptor explanation is just what I was looking for, I knew it had to be something related to the implementation of the protocol.  Do you know where I can find specifics about the descriptors? I was searching for it but I can't seem to find what values I should modify to get the increase rate to change in the smartphone app.

chegewara, 2018-04-26:
There is no need to appologise. You answer when you can or whenever you are ready, no rush.

This is descriptor you need:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLE2904.cpp

All you need to use it is in this issue:
https://github.com/nkolban/esp32-snippets/issues/314

This is example to display integer value to show battery level as percent:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEHIDDevice.cpp#L38-L41