nkolban/esp32-snippets #487: Minimum Heap to run BLE SPP

PhilColbert, 2018-04-19:
Wondering what the minimum heap needed to run BLE SPP is ?

I have now 80k heap, which drops to 10k when BLE service is started.

Then it crashes.

Thanks

chegewara, 2018-04-21:
Try to use this command in app_main() (it could be first command):

https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEDevice.cpp#L348

PhilColbert, 2018-04-23:
Thanks, already have that line in to give me the 80k! :)

think I have grabbed as much heap back as possible so far.

chegewara, 2018-04-23:
You can search esp32.com forum to see how to get back even more heap by editing menuconfig.

PhilColbert, 2018-05-04:
Seems you need over 85K free heap before starting BLE to get it working, just in case anyone has this issue.

Anything less and it crashes out.

Bit mental we need more heap than the esp8266 has got just to run BLE !

closed 2018-05-04 by PhilColbert