nkolban/esp32-snippets #489: Two I2C OLED Problem

trad0v, 2018-04-20:
Hi Mr. @nkolban ,
I write on the ssd1306 oled using the HAL you provided and it works fine !(https://github.com/nkolban/esp32-snippets/blob/master/hardware/displays/U8G2/test_SSD1306_i2c.c)
But when i try to connect two OLEDs to the I2c bus, each one having a different address, the ESP32 fails to install i2c driver the output is:
**Output:**

E (940) i2c: i2c driver install error
E (940) err: esp_err_t = -1
assertion "0 && "i2c_driver_install(I2C_MASTER_NUM, conf.mode, I2C_MASTER_RX_BUF_DISABLE, I2C_MASTER_TX_BUF_DISABLE, 0)"" failed: file "C:/Users/trad0v/Issue_OLED/components/U8G2/u8g2_esp32_hal.c", line 130, function: u8g2_esp32_i2c_byte_cb

**Code:**
```C
#include <driver/gpio.h>
#include <driver/spi_master.h>
#include <esp_log.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <stdio.h>
#include <string.h>
#include "esp_wifi.h"
#include "u8g2.h"

#include "sdkconfig.h"
#include "u8g2_esp32_hal.h"

// SDA - GPIO21
#define PIN_SDA 13

// SCL - GPIO22
#define PIN_SCL 15

u8g2_t lcd1; // a structure which will contain all the data for one display
u8g2_t lcd2; // a structure which will contain all the data for one display

void task_test_SSD1306i2c(void *ignore) {
	u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;
		u8g2_esp32_hal.sda   = PIN_SDA;
		u8g2_esp32_hal.scl  = PIN_SCL;
		u8g2_esp32_hal_init(u8g2_esp32_hal);

		u8g2_Setup_ssd1306_i2c_128x32_univision_f(&lcd1, U8G2_R0, u8g2_esp32_i2c_byte_cb, u8g2_esp32_gpio_and_delay_cb);
		u8g2_Setup_ssd1306_i2c_128x32_univision_f(&lcd2, U8G2_R0, u8g2_esp32_i2c_byte_cb, u8g2_esp32_gpio_and_delay_cb);

		u8x8_SetI2CAddress(&lcd2.u8x8,0x7A);
		u8g2_InitDisplay(&lcd2);
		u8g2_SetPowerSave(&lcd2, 0);
		u8g2_ClearBuffer(&lcd2);
		u8g2_SetFont(&lcd2, u8g2_font_ncenB14_tr);
		u8g2_DrawStr(&lcd2, 0,30,"Test");
		u8g2_SendBuffer(&lcd2);

		u8x8_SetI2CAddress(&lcd1.u8x8,0x78);
		u8g2_InitDisplay(&lcd1);
		u8g2_SetPowerSave(&lcd1, 0);
		u8g2_ClearBuffer(&lcd1);
		u8g2_SetFont(&lcd1, u8g2_font_ncenB14_tr);
		u8g2_DrawStr(&lcd1, 2,17,"Hello");
		u8g2_SendBuffer(&lcd1);

		vTaskDelete(NULL);

}

void app_main(void)
{
	xTaskCreate(task_test_SSD1306i2c, "task_test_SSD1306i2c", 2*4096, NULL, 10, NULL);

}
```

trad0v, 2018-04-21:
Adding i2c_driver_delete(I2C_NUM_1); between the two solved the problem !

**Code:**
```C
#include <driver/gpio.h>
#include <driver/spi_master.h>
#include <esp_log.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <stdio.h>
#include <string.h>
#include "esp_wifi.h"
#include "u8g2.h"

#include "sdkconfig.h"
#include "u8g2_esp32_hal.h"

// SDA - GPIO21
#define PIN_SDA 13

// SCL - GPIO22
#define PIN_SCL 15

u8g2_t lcd1; // a structure which will contain all the data for one display
u8g2_t lcd2; // a structure which will contain all the data for one display

void task_test_SSD1306i2c(void *ignore) {
	u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;
		u8g2_esp32_hal.sda   = PIN_SDA;
		u8g2_esp32_hal.scl  = PIN_SCL;

		u8g2_esp32_hal_init(u8g2_esp32_hal);

		u8g2_Setup_ssd1306_i2c_128x32_univision_f(&lcd1, U8G2_R0, u8g2_esp32_i2c_byte_cb, u8g2_esp32_gpio_and_delay_cb);
		u8g2_Setup_ssd1306_i2c_128x32_univision_f(&lcd2, U8G2_R0, u8g2_esp32_i2c_byte_cb, u8g2_esp32_gpio_and_delay_cb);

		u8x8_SetI2CAddress(&lcd2.u8x8,0x7A);
		u8g2_InitDisplay(&lcd2);
		u8g2_SetPowerSave(&lcd2, 0);
		u8g2_ClearBuffer(&lcd2);
		u8g2_SetFont(&lcd2, u8g2_font_ncenB14_tr);
		u8g2_DrawStr(&lcd2, 0,30,"Test");
		u8g2_SendBuffer(&lcd2);

		i2c_driver_delete(I2C_NUM_1);

		u8x8_SetI2CAddress(&lcd1.u8x8,0x78);
		u8g2_InitDisplay(&lcd1);
		u8g2_SetPowerSave(&lcd1, 0);
		u8g2_ClearBuffer(&lcd1);
		u8g2_SetFont(&lcd1, u8g2_font_ncenB14_tr);
		u8g2_DrawStr(&lcd1, 2,17,"Hello");
		u8g2_SendBuffer(&lcd1);

		vTaskDelete(NULL);

}

void app_main(void)
{
	xTaskCreate(task_test_SSD1306i2c, "task_test_SSD1306i2c", 2*4096, NULL, 10, NULL);

}
```

closed 2018-04-22 by trad0v

trad0v, 2018-04-22:
Hello,
i2c_driver_delete(I2C_NUM_1) was a temporary solution,
the problem was completely solved when i added a boolean variable in the HAL it checks if the i2c already been initialized it doesn't let it iniatialize another time.

Thank you 

Code:
```C
#include "u8g2_esp32_hal.h"

#include <stdio.h>
#include <string.h>

#include "sdkconfig.h"
#include "esp_log.h"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

static const char *TAG = "u8g2_hal";
static const unsigned int I2C_TIMEOUT_MS = 1000;

static spi_device_handle_t handle_spi;      // SPI handle.
static i2c_cmd_handle_t    handle_i2c;      // I2C handle.
static u8g2_esp32_hal_t    u8g2_esp32_hal;  // HAL state data.
static bool initialized=false;				// this is what i added
#undef ESP_ERROR_CHECK
#define ESP_ERROR_CHECK(x)   do { esp_err_t rc = (x); if (rc != ESP_OK) { ESP_LOGE("err", "esp_err_t = %d", rc); assert(0 && #x);} } while(0);

/*
 * Initialze the ESP32 HAL.
 */
void u8g2_esp32_hal_init(u8g2_esp32_hal_t u8g2_esp32_hal_param) {
	u8g2_esp32_hal = u8g2_esp32_hal_param;
} // u8g2_esp32_hal_init

/*
 * HAL callback function as prescribed by the U8G2 library.  This callback is invoked
 * to handle SPI communications.
 */
uint8_t u8g2_esp32_spi_byte_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) {
	ESP_LOGD(TAG, "spi_byte_cb: Received a msg: %d, arg_int: %d, arg_ptr: %p", msg, arg_int, arg_ptr);
	switch(msg) {
		case U8X8_MSG_BYTE_SET_DC:
			if (u8g2_esp32_hal.dc != U8G2_ESP32_HAL_UNDEFINED) {
				gpio_set_level(u8g2_esp32_hal.dc, arg_int);
			}
			break;

		case U8X8_MSG_BYTE_INIT: {
			if (u8g2_esp32_hal.clk == U8G2_ESP32_HAL_UNDEFINED ||
					u8g2_esp32_hal.mosi == U8G2_ESP32_HAL_UNDEFINED ||
					u8g2_esp32_hal.cs == U8G2_ESP32_HAL_UNDEFINED) {
				break;
			}

		  spi_bus_config_t bus_config;
		  bus_config.sclk_io_num   = u8g2_esp32_hal.clk; // CLK
		  bus_config.mosi_io_num   = u8g2_esp32_hal.mosi; // MOSI
		  bus_config.miso_io_num   = -1; // MISO
		  bus_config.quadwp_io_num = -1; // Not used
		  bus_config.quadhd_io_num = -1; // Not used
		  //ESP_LOGI(TAG, "... Initializing bus.");
		  ESP_ERROR_CHECK(spi_bus_initialize(HSPI_HOST, &bus_config, 1));

		  spi_device_interface_config_t dev_config;
		  dev_config.address_bits     = 0;
		  dev_config.command_bits     = 0;
		  dev_config.dummy_bits       = 0;
		  dev_config.mode             = 0;
[... 152 lines cut ...]
			break;
	// Set the Software IÂ²C pin to the value passed in through arg_int.
		case U8X8_MSG_GPIO_I2C_CLOCK:
			if (u8g2_esp32_hal.scl != U8G2_ESP32_HAL_UNDEFINED) {
				gpio_set_level(u8g2_esp32_hal.scl, arg_int);
//				printf("%c",(arg_int==1?'C':'c'));
			}
			break;
	// Set the Software IÂ²C pin to the value passed in through arg_int.
		case U8X8_MSG_GPIO_I2C_DATA:
			if (u8g2_esp32_hal.sda != U8G2_ESP32_HAL_UNDEFINED) {
				gpio_set_level(u8g2_esp32_hal.sda, arg_int);
//				printf("%c",(arg_int==1?'D':'d'));
			}
			break;

	// Delay for the number of milliseconds passed in through arg_int.
		case U8X8_MSG_DELAY_MILLI:
			vTaskDelay(arg_int/portTICK_PERIOD_MS);
			break;
	}
	return 0;
} // u8g2_esp32_gpio_and_delay_cb

```

ShubhamS68, 2021-08-07:
Hello @jadtrad  I am still facing this issue
E (20) i2c: i2c_param_config(640): i2c clock choice is invalid, please check flag and frequency
E (20) err: esp_err_t = 258
assertion "0 && "i2c_param_config(I2C_MASTER_NUM, &conf)"" failed: file "IDF/ESP-IDF************************/main/u8g2_esp32_hal.c", line 129, function: u8g2_esp32_i2c_byte_cb

jadtrad, 2021-08-07:
Hello @ShubhamS68,

- can you share your code where you initialize the 2 OLED ?
- did you use the code above instead of the provided library ?

ShubhamS68, 2021-08-07:
Here is my code.
I have also tried the given code but that didn't work for me.
`
#include <driver/gpio.h>
#include <driver/spi_master.h>
#include <esp_log.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <stdio.h>
#include <string.h>
#include <u8g2.h>

#include "u8g2_esp32_hal.h"

// #define PIN_CLK 14
// #define PIN_MOSI 13
// #define PIN_RESET 26
// #define PIN_DC 27
// #define PIN_CS 15
#define PIN_SDA 13
#define PIN_SCL 15

static char tag[] = "test_SSD1306";

void app_main(void)
{
  	// u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;
	// u8g2_esp32_hal.clk   = PIN_CLK;
	// u8g2_esp32_hal.mosi  = PIN_MOSI;
	// u8g2_esp32_hal.cs    = PIN_CS;
	// u8g2_esp32_hal.dc    = PIN_DC;
	// u8g2_esp32_hal.reset = PIN_RESET;
	// u8g2_esp32_hal_init(u8g2_esp32_hal);

	u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;
	u8g2_esp32_hal.sda = PIN_SDA;
	u8g2_esp32_hal.scl = PIN_SCL;
	u8g2_esp32_hal_init(u8g2_esp32_hal);

	u8g2_t u8g2; // a structure which will contain all the data for one display
	u8g2_Setup_ssd1306_128x64_noname_f(
		&u8g2,
		U8G2_R0,
		u8g2_esp32_i2c_byte_cb,
		u8g2_esp32_gpio_and_delay_cb);  // init u8g2 structure

	u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,

	u8g2_SetPowerSave(&u8g2, 0); // wake up display
	u8g2_ClearBuffer(&u8g2);
	u8g2_DrawBox(&u8g2, 10,20, 20, 30);
	u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
	u8g2_DrawStr(&u8g2, 0,15,"Hello World!");
	u8g2_DrawCircle(&u8g2,30, 20, 10,U8G2_DRAW_ALL);
	u8g2_SendBuffer(&u8g2);

	ESP_LOGD(tag, "All done!");

	vTaskDelete(NULL);
}

`

jadtrad, 2021-08-07:
Are you using another i2c device with the OLED ?

ShubhamS68, 2021-08-07:
Here is my code.
I have tried the above codes but that didn't work for me.
Code: 

#include <driver/gpio.h>
#include <driver/spi_master.h>
#include <esp_log.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <stdio.h>
#include <string.h>
#include <u8g2.h>

#include "u8g2_esp32_hal.h"

/*
*#define PIN_CLK 14
*#define PIN_MOSI 13
*#define PIN_RESET 26
*#define PIN_DC 27
*#define PIN_CS 15
*#define PIN_SDA 13
*#define PIN_SCL 15
*/

static char tag[] = "test_SSD1306";

void app_main(void)
{
	/*
	*u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;
	*u8g2_esp32_hal.clk   = PIN_CLK;
	*u8g2_esp32_hal.mosi  = PIN_MOSI;
	*u8g2_esp32_hal.cs    = PIN_CS;
	*u8g2_esp32_hal.dc    = PIN_DC;
	*u8g2_esp32_hal.reset = PIN_RESET;
	*u8g2_esp32_hal_init(u8g2_esp32_hal);
	*/

	u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;
	u8g2_esp32_hal.sda = PIN_SDA;
	u8g2_esp32_hal.scl = PIN_SCL;
	u8g2_esp32_hal_init(u8g2_esp32_hal);

	u8g2_t u8g2; // a structure which will contain all the data for one display
	u8g2_Setup_ssd1306_128x64_noname_f(
		&u8g2,
		U8G2_R0,
		u8g2_esp32_i2c_byte_cb,
		u8g2_esp32_gpio_and_delay_cb);  // init u8g2 structure

	u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,

	u8g2_SetPowerSave(&u8g2, 0); // wake up display
	u8g2_ClearBuffer(&u8g2);
	u8g2_DrawBox(&u8g2, 10,20, 20, 30);
	u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
	u8g2_DrawStr(&u8g2, 0,15,"Hello World!");
	u8g2_DrawCircle(&u8g2,30, 20, 10,U8G2_DRAW_ALL);
	u8g2_SendBuffer(&u8g2);

	ESP_LOGD(tag, "All done!");

	vTaskDelete(NULL);
}

ShubhamS68, 2021-08-07:
Here is the code 

Code:

#include <driver/gpio.h>
#include <driver/spi_master.h>
#include <esp_log.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <stdio.h>
#include <string.h>
#include <u8g2.h>

#include "u8g2_esp32_hal.h"

// #define PIN_CLK 14
// #define PIN_MOSI 13
// #define PIN_RESET 26
// #define PIN_DC 27
// #define PIN_CS 15
#define PIN_SDA 13
#define PIN_SCL 15

static char tag[] = "test_SSD1306";

void app_main(void)
{
  	// u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;
	// u8g2_esp32_hal.clk   = PIN_CLK;
	// u8g2_esp32_hal.mosi  = PIN_MOSI;
	// u8g2_esp32_hal.cs    = PIN_CS;
	// u8g2_esp32_hal.dc    = PIN_DC;
	// u8g2_esp32_hal.reset = PIN_RESET;
	// u8g2_esp32_hal_init(u8g2_esp32_hal);

	u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;
	u8g2_esp32_hal.sda = PIN_SDA;
	u8g2_esp32_hal.scl = PIN_SCL;
	u8g2_esp32_hal_init(u8g2_esp32_hal);

	u8g2_t u8g2; // a structure which will contain all the data for one display
	u8g2_Setup_ssd1306_128x64_noname_f(
		&u8g2,
		U8G2_R0,
		u8g2_esp32_i2c_byte_cb,
		u8g2_esp32_gpio_and_delay_cb);  // init u8g2 structure

	u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,

	u8g2_SetPowerSave(&u8g2, 0); // wake up display
	u8g2_ClearBuffer(&u8g2);
	u8g2_DrawBox(&u8g2, 10,20, 20, 30);
	u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
	u8g2_DrawStr(&u8g2, 0,15,"Hello World!");
	u8g2_DrawCircle(&u8g2,30, 20, 10,U8G2_DRAW_ALL);
	u8g2_SendBuffer(&u8g2);

	ESP_LOGD(tag, "All done!");

	vTaskDelete(NULL);
}

ShubhamS68, 2021-08-07:
> Are you using another i2c device with the OLED ?

No

jadtrad, 2021-08-07:
Please try this code in : hardware/displays/U8G2/test_SSD1306_i2c.c
#include <driver/gpio.h>
#include <driver/spi_master.h>
#include <esp_log.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <stdio.h>
#include <string.h>
#include <u8g2.h>

#include "sdkconfig.h"
#include "u8g2_esp32_hal.h"

// SDA - GPIO21
#define PIN_SDA 21

// SCL - GPIO22
#define PIN_SCL 22

static const char *TAG = "ssd1306";

void task_test_SSD1306i2c(void *ignore) {
	u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;
	u8g2_esp32_hal.sda   = PIN_SDA;
	u8g2_esp32_hal.scl  = PIN_SCL;
	u8g2_esp32_hal_init(u8g2_esp32_hal);

	u8g2_t u8g2; // a structure which will contain all the data for one display
	u8g2_Setup_ssd1306_i2c_128x32_univision_f(
		&u8g2,
		U8G2_R0,
		//u8x8_byte_sw_i2c,
		u8g2_esp32_i2c_byte_cb,
		u8g2_esp32_gpio_and_delay_cb);  // init u8g2 structure
	u8x8_SetI2CAddress(&u8g2.u8x8,0x78);

	ESP_LOGI(TAG, "u8g2_InitDisplay");
	u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,

	ESP_LOGI(TAG, "u8g2_SetPowerSave");
	u8g2_SetPowerSave(&u8g2, 0); // wake up display
	ESP_LOGI(TAG, "u8g2_ClearBuffer");
	u8g2_ClearBuffer(&u8g2);
	ESP_LOGI(TAG, "u8g2_DrawBox");
	u8g2_DrawBox(&u8g2, 0, 26, 80,6);
	u8g2_DrawFrame(&u8g2, 0,26,100,6);

	ESP_LOGI(TAG, "u8g2_SetFont");
    u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
	ESP_LOGI(TAG, "u8g2_DrawStr");
    u8g2_DrawStr(&u8g2, 2,17,"Hi nkolban!");
	ESP_LOGI(TAG, "u8g2_SendBuffer");
	u8g2_SendBuffer(&u8g2);

	ESP_LOGI(TAG, "All done!");

	vTaskDelete(NULL);
}